openhwfoundation/core-v-verif #1472: remove references to core-v-verif/vendor_lib/imperas

eroom1966, 2022-11-04:
@MikeOpenHWGroup @silabs-hfegran @silabs-mateilga 

I have a blocking issue due to the incorporation of libraries into the running simulator executable
This is due to the previous generation ISS integration loading an out of date library
in order to move forward I need any references removing, could somebody take the lead on this ?

MikeOpenHWGroup, 2022-11-04:
@eroom1966 can you provide more information?

eroom1966, 2022-11-04:
I think @silabs-hfegran is working on this but ...
There is a shared object in the vendor_lib/imperas which gets loaded 
This conflicts with a newer version.

The requirement here, is to remove all references to the code in vendor_lib/imperas - this can then be deleted and the problem goes away

eroom1966, 2022-11-08:
Looks like this is now resolved thanks @silabs-hfegran

closed 2022-11-08 by eroom1966