openhwfoundation/core-v-verif #1488: Error when I run the core-v and compiliance test

Zhengshuhang, 2022-11-14:
Hi, I am a beginner of core-v-verif. I have successfully run through the core testbench and UVM testbench examples with USE_ISS=NO, but I have problems with other code:

First, when I generate, compile, and run tests with core-v:
make corev-dv
make gen_corev-dv test TEST=corev_rand_jump_stress_test CV_SIMULATOR=vsim USE_ISS=NO

and the error is :
[image]
~~~~~~~~~
Second, when I try RISC-V Compliance Test-suite and Regressions, It doesn't work properly:
make compliance RISCV_ISA=rv32i COMPLIANCE_PROG=I-ADD-01 CV_SIMULATOR=vsim USE_ISS=NO
 
and error:
<img width="554" alt="image" src="https://user-images.githubusercontent.com/77762994/201567252-c49efb9f-da75-485f-b785-c1ec8656c18d.png">

look forward to your kind help!

Zhengshuhang, 2022-11-14:
the second error is this , sorry
[image]

MikeOpenHWGroup, 2022-11-14:
Hi @Zhengshuhang, thanks for creating this issue.  Actually, this is two issues, so let's work on the first one.

Please attach the complete compile logfile to this issue - there may be some useful information that is not included in the snipit you have posted.

Your logfile indicates that `gcc` cannot parse the linker script, `link_ld`.  This will be at `/home/ubuntu20/workspace/core-v-verif/cv32e40p/sim/uvmt/vsim_results/default/corev_rand_jump_stress_test/0/test_program/link.ld` on your system.  The Makefile will copy this from `core-v-verif/cv32e40p/bsp/link_corev-dv.ld`, so please check that these two files exist.

The only other thing I notice is that you are using gcc 12.0.0.  This should be OK, but I use gcc 11.  Where did you get your gcc distribution from?

Zhengshuhang, 2022-11-14:
Thx for your reply!
Firstly,I can find the first link.ld :
[image]

and the second file exists too:
[image]

both are only one line: link.ld

I download corev gcc cross-compile toolchain from Embecosm, the newest version seems to to result in an error and I use a previous version.

and log:
[vsim_gen_corev-dv.log](https://github.com/openhwgroup/core-v-verif/files/10004029/vsim_gen_corev-dv.log)

Thank you again!I'm new to this and I'm having a lot of problems building environments with ubuntu. I wonder if there will be something like docker?

Zhengshuhang, 2022-11-14:
By the way, I find that I do not have a commercial license, so I can only set USE_ISS=NO, is there any way for me to get a commercial license?

MikeOpenHWGroup, 2022-11-14:
I do not see any issues in your logfile except for the `link.ld:0: syntax error` message, so I strongly suspect the issue is your gcc version.  I am running Ubuntu 22.04 and using `riscv32-embecosm-gcc-ubuntu2004-20211212` from the Embecosm website.

To answer your other questions:
> I wonder if there will be something like docker?

We currently have no plans for this.  We are a member-driven, open-source organization and would gladly consider a contribution!  :stuck_out_tongue_winking_eye: 

> Is there any way for me to get a commercial license?

Yes.  ImperasDV is a commerical product of Imperas Software.  You can reach them at https://www.imperas.com/.

pascalgouedo, 2022-11-14:
Hi,

gcc 12.0 is used for CV32E40Pv2 version of the GNU Compiler toolchain.
I switched to that version last week without any trouble.
But I am working on rhel7, not ubuntu.

Zhengshuhang, 2022-11-15:
Hi，thx for your help;
I finally run the corev-dv test successfully by 
`make gen_corev-dv test TEST=corev_rand_jump_stress_test CV_SIMULATOR=vsim USE_ISS=NO`
I solve the previous question by copy the content in `/home/ubuntu20/workspace/core-v-verif/cv32e40p/bsp/link_corev-dv.ld` to `/home/ubuntu20/workspace/core-v-verif/cv32e40p/bsp/link.ld`
It seems that at some point link.ld was changed, so it needs to be restored.

Also, I switched to Ubuntu22, I didn't find riscv32-embecosm-gcc-ubuntu2004-20211212 and I use riscv32-embecosm-gcc-ubuntu2204-20220911.tar. 
and when I ran hello-world I had the following error:
[image]
[uvm_hello-world_error.log](https://github.com/openhwgroup/core-v-verif/files/10013460/uvm_hello-world_error.log)
I encountered this error in Ubuntu16 and did not resolve it.
Thinking that I had run through some examples on Ubuntu20, I decided to go back to Ubuntu20 to fix my second error.

MikeOpenHWGroup, 2022-11-18:
I'm confused.  These work-around should not be necessary.  Are the [required environment variables](https://github.com/openhwgroup/core-v-verif/blob/master/mk/README.md#required-corev-environment-variables) set properly?

pascalgouedo, 2022-11-18:
This is master branch.
As we were told to use cv32e40p/dev branch (and not master) since the beginning, we always used following setup :
setenv COREV YES
setenv COREV_SW_TOOLCHAIN ...
because CV_SW_TOOLCHAIN setting never worked (don't remember why).
Lee experienced the same lately when trying to help debug interrupt stuff ...

Anyway apart from that slight different setup, we don't have to make specific things like linker script copy or anything else like that.

MikeOpenHWGroup, 2022-11-21:
Hi @Zhengshuhang, what is the status of this issue?  If you are using the master branch of CORE-V-VERIF, you should not have any any trouble.  If your TOOLCHAIN parameters are set like this:
```
CV_SW_TOOLCHAIN=/opt/riscv
CV_SW_PREFIX=riscv32-unknown-elf-
```
And your toolchain is installed in `$CV_SW_TOOLCHAIN/bin/$CV_SW_PREFIX` then the following should work for you:
```
$ git clone https://github.com/openhwgroup/core-v-verif
$ cd core-v-verif/cv32e40p/sim/uvmt
$ make hex TEST=hello-world
```
The "hex" target will compile the test-program (hello-world in this case), but will not attempt to compile any SystemVerilog code. 
 The results are placed a directory called `./${CV_SIMULATOR}_results/default/hello-world/0/test_program`. Attached is the output of the `make hex` command on my system:
[make_hex_hw.txt](https://github.com/openhwgroup/core-v-verif/files/10059371/make_hex_hw.txt)

MikeOpenHWGroup, 2022-11-22:
Happy to see you making progess @Zhengshuhang.

1. You should not set `RISCV_DEVICE` in your `make compliance` command line.  Are there instructions in the READMEs to do that?  If there are, please let me know and I will fix it.
2. You will need to contact Imperas Software if you want a functional ISS to use.

Zhengshuhang, 2022-11-24:
Hi @MikeOpenHWGroup ，Thank you for your kind and timely reply!
1.I should not set `RISCV_DEVICE` ,I just copied this part by mistake and I've corrected it now.
but when I run `make compliance RISCV_ISA=rv32i COMPLIANCE_PROG=I-ADD-01 CV_SIMULATOR=vsim USE_ISS=NO`
the same error still exists, I would like to know whether I can run the compliance test sample code of this project without LICENSE? Since I've run through most of the other sample commands in the documentation, I think I should be able to run the relevant RISCV compliance test example as well.
[make_error_compliance.log](https://github.com/openhwgroup/core-v-verif/files/10080129/make_error_compliance.log)
In addition, I found the following error message:
[image]
[image]
I wonder what causes these errors. Is it because I lack the LICENSE or because of other reasons that the HEX file is not generated correctly?
2.Thanks for your advice, I contacted the Imperas and received following reply:
[image]
So I'm going to do a little learning without a LICENSE.Thank you for your enthusiastic support.

Zhengshuhang, 2022-11-29:
Hi，Thanks for your kindly help!
I think this may be caused by my previous mistakes. I am going to try to solve this problem by myself. If I cannot solve it, I will come back to you for help. Now I'm going to close this comment. I am deeply touched by your kindness and patience. Wish you every success in your work!

closed 2022-11-29 by Zhengshuhang