openhwfoundation/core-v-verif #1502: Fencei VP Misbehavior

silabs-robin, 2022-11-22:
### Type
* Functionally incorrect behavior

### Steps to Reproduce
```
cd cv32e40s/sim/uvmt/
make test TEST=fencei USE_ISS=0
```

### Additional context
The line at https://github.com/openhwgroup/core-v-verif/blob/cv32e40s/dev/cv32e40s/env/uvme/vseq/uvme_cv32e40s_vp_fencei_tamper_seq.sv#L150 calls an ISS DPI function, but it doesn't first check that the ISS is enabled.
The ISS subsequently complaints about a function being called before initialization.
These should be an if-statement to check that the ISS is enabled before trying that call.

MikeOpenHWGroup, 2022-11-22:
I'm confused.  Shouldn't the ENV guarantee that the Reference Model is initialised before any virtual sequences are launched?

silabs-robin, 2022-11-22:
> Shouldn't the ENV guarantee that the Reference Model is initialised before any virtual sequences are launched?

Good point. I changed the description above to ask for more investigation on what the best solution actually is.

The problem here is that if you run with USE_ISS=0, then there is deliberately no initialization of the ISS. In that case, the VP should not boldly make a DPI call to the ISS.

(Note, this is not a big issue. It is almost a non-issue. The tests using the fencei VP should preferably be run with the ISS enabled, it is just that _if_ one for some reason runs the fencei tests without the ISS then it will currently fail.)

silabs-robin, 2022-11-22:
Actually... there is also a slightly bigger issue.

> if one for some reason runs the fencei tests without the ISS then it will currently fail

This is not strictly true.
There will be printed an "error", but the test will say "passed".
This can be dangerous, if the ISS is printing some error messages and regressions just pass.

MikeOpenHWGroup, 2022-11-22:
> This can be dangerous, if the ISS is printing some error messages and regressions just pass.

This is a separate and very dangerous situation.  Please open another issue for this.  Upon detection of an error the Reference Model _must_ either invoke `uvm_error()` or provide a signal to the ENV that will result in an unambiguous invocation of `uvm_error()`.

silabs-hfegran, 2023-01-20:
For this specific case I think the sequence needs a small update, but I also think that this warrants a change in how the ISS reports errors to the environment. I have commented further on that in the linked #1503 issue.

eroom1966, 2023-01-26:
Note, this breaks the alternate case when the ISS is enabled, try running 
```
cd cv32e40s/sim/uvmt/
make test TEST=fencei USE_ISS=0
```
suggested fix is to use $test$plusargs
```
function void uvme_cv32e40s_vp_fencei_tamper_seq_c::write_iss_mem();
  if ($test$plusargs("USE_ISS")) begin
    rvviRefMemoryWrite(0, addr, data, 4);
  end
endfunction : write_iss_mem
```

silabs-robin, 2023-12-11:
Previously fixed using the `$test$plusargs("USE_ISS")` suggestion.

closed 2023-12-11 by silabs-robin