openhwfoundation/core-v-verif #1506: Script variable problem

guojiazhuxi, 2022-11-24:
## Bug Title
When verifying the Core-V-Verif Quick Start Guide article，Discover Error: UnrecoGnized OPCode `CSRW MTVEC, A0 ', Extension` ZICSR' Required error，Discovery after review，the $CFLAGS variable in ~/core-v-verif/mk/Makefile is not passed to ~/core-v-verif/cv32e40p/bsp/Makefile

是a中的p变量没有传递到c

### Type
Indicate whether the type of problem you found:
* Compile error (hopefully nobody has committed anything that doesn't compile!)

### Steps to Reproduce
Please provide:
[image]
[image]

MikeOpenHWGroup, 2022-11-28:
Hi @guojiazhuxi, thanks for your issue.  I do not see this problem.  Could you provide me with your _exact_ commands starting from `git clone` and ending with `make...`?  It would also be helpful if you could attach anything written to stdout.

Thanks!

pascalgouedo, 2022-11-28:
Hi @MikeOpenHWGroup,

This is all related to the last Embecosm toolchain (20221031) which now requires _zicsr in all march definitions.

Hi @guojiazhuxi 
CFLAGS value transmission between Common.mk and bsp Makefile seems to be incorrect.
The update you need right now is just to replace rv32imc by rv32imc_zicsr in bsp makefile.
We will manage to update the repo once another bunch of updates will be finished and pushed.

Regards.

pascalgouedo, 2022-11-30:
Looks good as long as you stay in master branch ...

guojiazhuxi, 2022-12-02:
> 

yes，the value CFLAGS should use export prefix

MikeOpenHWGroup, 2023-07-26:
This issue was resolved in #1946

closed 2023-07-26 by MikeOpenHWGroup