openhwfoundation/core-v-verif #1508: CV32E40X simulation

aledolme, 2022-11-25:
Hello!

I'm having some troubles in compiling CV32E40X uvm simulation (starting from /sim/uvmt folder) that I've not encountered with CV32E40P. 
In particular, using the same simulation enviroment and the same simulator, with CV32E40X I got this error, at Running vlog step:

*******************************************************************************************
* Running vlog in $HOME/core-v-verif/cv32e40p/sim/uvmt/vsim_results/default
* Log: $HOME/core-v-verif/cv32e40p/sim/uvmt/vsim_results/default/vlog.log
*******************************************************************************************
cd $HOME/core-v-verif/cv32e40p/sim/uvmt/vsim_results/default && \
	 vlog \
		-work work \
		-l vlog.log \
		-suppress 2577 -suppress 2583 -suppress 13185 -suppress 13314 -suppress 13288 -suppress 2181 -suppress 13262 -timescale "1ns/1ps" -sv -mfcu +acc=rb -quiet -writetoplevels  uvmt_cv32e40x_tb  "+define+CV32E40X_TRACE_EXECUTION" "+define+UVM" \
		+define+ZBA_ZBB_ZBC_ZBS \
..

And then the simulation stops because of some errors related to the simulator like:
- ** Error: (vlog-13069) $HOME/core-v-verif/core-v-cores/cv32e40x/rtl/if_c_obi.sv(27): near ";": syntax error, unexpected ';', expecting ::.
This version of the compiler is incompatible with library file.
This version of the compiler is incompatible with library file.
** Error: ** while parsing file included at $HOME/core-v-verif/core-v-cores/cv32e40x/rtl/../bhv/cv32e40x_wrapper.sv(34)
** at $HOME//core-v-verif/core-v-cores/cv32e40x/rtl/../sva/cv32e40x_rvfi_sva.sv(29): (vlog-13006) Could not find the package (cv32e40x_rvfi_pkg).  Design read will continue, but expect a cascade of errors after this failure.  Furthermore if you experience a vopt-7 error immediately before this error then please check the package names or the library search paths on the command line.
This version of the compiler is incompatible with library file.
This version of the compiler is incompatible with library file.
etc

However, the simulator is the same used for CV32E40P simulation, and there it works fine. Any ideas of which can be the problem?
CFG_COMPILE_FLAGS is different (in vsim.mk), but I have not found where it is instantiated.

Thanks you!

MikeOpenHWGroup, 2022-11-28:
Hi @aledolme, thanks for your issue.  Up to last week I did not have access to Questa tools and AFAIK, none of the teams involved in the E40X are using Questa, so it is possible that the Makefiles for running the E40X on Questa have suffered bit-rot.

The good news is that I now have access to Questa and I will look into your issue as soon as possible (probably not today).

MikeOpenHWGroup, 2022-11-28:
Pull-request #1511 _should_ restore Questa for E40X.

aledolme, 2022-11-29:
Thanks you! However, despite the changes in the vsim.mk, there are still some problems which I not encounter with CV32E40P version, in the compiling phase of BSP in Common.mk file. 

***************************************************************************************
* Compiling the BSP
***************************************************************************************
mkdir -p  $HOME/core-v-verif/cv32e40x/sim/uvmt/vsim_results/default/hello-world/0/test_program/bsp
cp  $HOME/core-v-verif/cv32e40x/bsp/Makefile  $HOME/core-v-verif/cv32e40x/sim/uvmt/vsim_results/default/hello-world/0/test_program/bsp
make -C  $HOME/core-v-verif/cv32e40x/sim/uvmt/vsim_results/default/hello-world/0/test_program/bsp \
    VPATH= $HOME/core-v-verif/cv32e40x/bsp \
    RISCV= $HOME/opt/riscv \
    RISCV_PREFIX=riscv32-unknown-elf- \
    RISCV_EXE_PREFIX= $HOME/opt/riscv/bin/riscv32-unknown-elf- \
    RISCV_MARCH=rv32imc \
    RISCV_CC=gcc \
    RISCV_CFLAGS="" \
    all
make[2]: Entering directory ' $HOME/core-v-verif/cv32e40x/sim/uvmt/vsim_results/default/hello-world/0/test_program/bsp'
 $HOME/opt/riscv/bin/riscv32-unknown-elf-gcc -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem  $HOME/miniconda3/include -c  $HOME/core-v-verif/cv32e40x/bsp/crt0.S -o crt0.o
Assembler messages:
Fatal error: -march=nocona: ISA string must begin with rv32 or rv64
cc1: error: '-march=nocona': ISA string must begin with rv32 or rv64
cc1: error: unknown cpu 'haswell' for '-mtune'
make[2]: *** [Makefile:21: crt0.o] Error 2
make[2]: Leaving directory ' $HOME/core-v-verif/cv32e40x/sim/uvmt/vsim_results/default/hello-world/0/test_program/bsp'

This error does not occur with the other core. I have control all the file present in the /CV32E40X/BSP folder, yet I have not found where "nocona" or "haswell" are set, and still the simulation is not termined as it should. 
Have you have any idea about which bugs there can still be between cv32e40x and cv32e40p for Questasim simulation?

MikeOpenHWGroup, 2022-11-29:
This is very odd indeed.  It is not at all clear where is command-line is being issued:
> $HOME/opt/riscv/bin/riscv32-unknown-elf-gcc -march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem $HOME/miniconda3/include -c $HOME/core-v-verif/cv32e40x/bsp/crt0.S -o crt0.o

That is not a command that is issued by any of the CORE-V-VERIF Makefiles and/or scripts.  It looks to me like `$HOME/core-v-verif/cv32e40x/sim/uvmt/vsim_results/default/hello-world/0/test_program/bsp/Makefile` was somehow corrupted. 

Try `make clean_all` and try compiling/running again.

MikeOpenHWGroup, 2022-12-01:
Hi @aledolme, where are we on this issue.  Pull-request #1511 has been merged in, so the Makefile support for Questa should be in good shape.  I am not sure I can help you with the `-march=nocona` issue.

LMK.

aledolme, 2022-12-02:
Thanks you! 
The problem occurs when compiling Makefile of CV32E40X/bsp folder. Despite the fact that I'm using the same toolchain, the same CFLAGS of CV32E40P simulations and that the bsp files generated in hello-world/bsp folder are the same, in CV32E40X simulations CFLAGS changes, leading to the error mentioned above. 
I will try to solve this issue. 

Thank you for your help with Questasim!
AD

aledolme, 2022-12-13:
Hi Mike! 
I found out that depending on the test and the core that are compiled, the flag  **-march=nocona -mtune=haswell -ftree-vectorize -fPIC -fstack-protector-strong -fno-plt -O2 -ffunction-sections -pipe -isystem $HOME/miniconda3/include** is overwritten above the one defined during runtime. 
This problems occurs also for some tests of CV32E40P core (like Fibonacci one): here, in order to solve it, I have simply substitute the flag $(CFLAGS) using **-0s -g -statis -mabi=ilp32 -march=rv32imc -Wall -pedantic** and it works. 

In the case of CV32E40X core, in addition to this modification, I need to write manually these flags also in "Compiling test-program" sections of Common.mk, and even is the simulation goes on, it now stops at "running vlog" section founding some error in "uvma_isacov_pkg.sv".
Also "vsim.mk" has been modified, substituting $(CFG_COMPILE_FLAGS) for the same reason explained abov with "+define+NO_PULP".

Any idea on how to solve the problem? Are you able to properly compile hello-world test with Questa and CV32E40x?