openhwfoundation/core-v-verif #1525: Radom example by riscv-dv cannot run in CV32E UVM DV environment

Zhengshuhang, 2022-12-06:
Hi, I am a new learner of riscv DV. I have run the command successfully in Ubuntu20,Questrasim2021,CV32E40P UVM DV:
`make gen_corev-dv test TEST=corev_rand_arithmetic_base_test CV_SIMULATOR=vsim USE_ISS=NO`
Cause I do not have license for ISS, I tried other ISS as an alternative,I have run riscv-dv+spike successfully, I can generate radom test and output trace log by spike by `run`;
Then I try to carry random tests generated by riscv-dv  to CV32E40P as directional tests, 
I found some suspected configuration file and I put it in riscv-dv as follows:
`core-v-verif/cv32e40p/env/corev-dv/cv32e40p_csr_template.yaml` -> `workspace/riscv-dv/yaml/csr_template.yaml`
`core-v-verif/cv32e40p/env/corev-dv/target/cv32e40p/riscv_core_setting.sv`-> `workspace/riscv-dv/target/rv32imc/riscv_core_setting.sv`
Then I type `run` in riscv-dv,random test .S file and .log file by spike generated successfully; 

for example I put the generated random test .S in `/home/ubuntu20/workspace/core-v-verif/cv32e40p/tests/programs/custom/zsh_riscv_arithmetic_basic_test_0`
and type: `make test TEST=zsh_riscv_arithmetic_basic_test_0 CV_SIMULATOR=vsim USE_ISS=NO`
However, the run always gets stuck on this line:
[image]
Then I check the `trace_core_00000000.log`,I found that the core trace looked right at first, but when it ran more than about 10,000 lines, it entered an endless loop and a kind of repetition, so it couldn't continue; 
Following the spike log:
[riscv_arithmetic_basic_test.0.log](https://github.com/openhwgroup/core-v-verif/files/10160334/riscv_arithmetic_basic_test.0.log)
Following the core trace log:
[trace_core_00000000.log](https://github.com/openhwgroup/core-v-verif/files/10160336/trace_core_00000000.log)
Following the random test .S file generated by single riscv-dv package:
[zsh_riscvdv_arithmetic_basic_test_0_S.log](https://github.com/openhwgroup/core-v-verif/files/10160367/zsh_riscvdv_arithmetic_basic_test_0_S.log)

Can you help me and tell me what else I neglected in generating.S that resulted in this result? I wish I could use spike as a reference model, allowing me to compare trace logs without a license?

MikeOpenHWGroup, 2022-12-06:
Hi @Zhengshuhang, thanks for your issue.  There are lots of things to consider here:
1. You cannot run test-programs generated by riscv-dv in the core-v-verif environment.  In core-v-verif, we extend riscv-dv to make it compatible with our environment.  We call these extensions [corev-dv](https://github.com/openhwgroup/core-v-verif/tree/master/mk#corev-dv-generated-tests).  Here is a [presentation](https://docs.google.com/presentation/d/18IHmCRNLtPp16tTtUlwtVCwZwDvlgtmu/edit#slide=id.g11adfdf45a7_0_53) that explains writting tests for core-v-verif in detail (have a look at slide 22).
2. The old trace_core files are no longer used and we do not support them.  I recommend you look at the output of the RVFI Agent (`uvm_test_top.env.rvfi_agent.trn.log` in the *_results directory).
3. If you want to use Spike, you are strongly encouraged to follow the [step-and-compare](https://docs.openhwgroup.org/projects/core-v-verif/en/latest/cv32_env.html#step-and-compare-2-0) method, rather than post-processing logfiles.

Zhengshuhang, 2022-12-07:
Hi，Thanks for your advice, I'll try using the step-and-compare method later.

Zhengshuhang, 2022-12-09:
Hi， I reopen this issue to ask an additional question:
I found such a page in your PPT：
[image]
Maybe my  relevant knowledge is too shallow, but I did not find the relevant target or command to achieve regression test through `cv32e40p_full_covg_no_pulp.yaml`,I would venture to ask, is there a way in the project that I can implement a more complete regression test directly from the YAML file?

MikeOpenHWGroup, 2022-12-10:
There is a set of scripts for running regressions in the `bin` directory and there is an associated README.  The script you want is [cv-regress](https://github.com/openhwgroup/core-v-verif/tree/master/bin#cv_regress).

I agree this is hard to find.  :stuck_out_tongue_winking_eye:

Zhengshuhang, 2022-12-12:
Thx for your timely reply, which has solved my previous problem. I will close this problem again.

closed 2022-12-12 by Zhengshuhang