openhwfoundation/core-v-verif #1527: unrecognized opcode 'csrw' with cv32e40p

yangguangguang, 2022-12-07:
### Is there an existing core-v-verif task for this?

- [X] I have searched the existing task issues

### Task Description

Hi：

  I'm check out the CORE-V-VERIF Quick Start Guide in the [OpenHW Group CORE-V Verification Strategy](https://docs.openhwgroup.org/projects/core-v-verif/en/latest/intro.html) and running it step by step. When I compile the RTL and the core testbench using Verilator(version:4.102) then get the following ERROR:

----
make[3]: Entering directory '/home/verif/core-v-verif/cv32e40p/sim/core/simulation_results/hello-world/0/test_program/bsp'
/opt/riscv/bin/riscv32-unknown-elf-gcc -Os -g -static -mabi=ilp32 -march=rv32imc -Wall -pedantic -c /home/verif/core-v-verif/cv32e40p/bsp/crt0.S -o crt0.o
/home/verif/core-v-verif/cv32e40p/bsp/crt0.S: Assembler messages:
/home/verif/core-v-verif/cv32e40p/bsp/crt0.S:33: Error: unrecognized opcode `csrw mtvec,a0', extension `zicsr' required
make[3]: *** [Makefile:20: crt0.o] Error 1
----

  what could be the cause of the problem？I use CORE-V Toolchain on my system(Platform:Ubuntu 22.04), which from Embecosm.

### Description of Done

done

pascalgouedo, 2022-12-07:
Hi,
The correction for that is on-going PR https://github.com/openhwgroup/core-v-verif/pull/1513

Cheers,
Pascal.

yangguangguang, 2022-12-07:
sorry, my fault

MikeOpenHWGroup, 2022-12-07:
Thanks for raising this issue @yangguangguang.  If you consider this issue resolved, please close this issue.

yangguangguang, 2022-12-07:
The problem is solved! thanks a lot!