openhwfoundation/core-v-verif #1556: VCS scripts do not exist

abdullahyildiz, 2022-12-28:
## VCS tool scripts missing
Repository (including the releases) does not fully support VCS.

### Type
* Compile error

### Steps to Reproduce
```
export CV_SW_PREFIX=riscv32-unknown-elf-
export CV_SW_TOOLCHAIN=/eda/riscv/riscv-gnu-toolchain-rv32g
export CORE_V_VERIF=`realpath .`/core-v-verif

git clone -b cv32e40p/release https://github.com/openhwgroup/core-v-verif $CORE_V_VERIF
cd $CORE_V_VERIF

git apply cv32e40p_vcs_flow.patch
make -C cv32e40p/sim/core/ vcs-hello-world CV_SIMULATOR=vcs
```

Error message:
`do: couldn't open ../tools/vcs/vcs_batch.tcl : no such file or directory`

### Additional context
[cv32e40p_vcs_flow.patch](https://github.com/openhwgroup/core-v-verif/files/10314588/cv32e40p_vcs_flow.patch)

MikeOpenHWGroup, 2022-12-30:
Thanks for your issue @abdullahyildiz.  I am not surprised by this as no current member of OpenHW working on CV32E40P is an active VCS user, so the Makefiles for that simulator may have suffered from "bit rot".  Once we have this issue resolved for you I hope you will consider making a pull-request to fix the Makefiles.  As we say, Contributions are always welcome!  :wink:

Thanks also for your detailed report - it really helps!  A couple of things to look at:
1. Run `make` from the `cv32e40p/sim/core` directory.  It appears that the VCS "do file" relies on relative, not absolute paths.
2. Unless you have created a new test-program called `vcs-hello-world` and placed it in `cv32e40p/tests/programs/custom/vcs-hello-world/vcs-hello-world.c`, your command-line will not work.

Try this:
```
cd cv32e40p/sim/core
make sanity CV_SIMULATOR=vcs
```

abdullahyildiz, 2023-01-02:
@MikeOpenHWGroup,

Thanks for your support. Actually the default Makefile throws path errors when it has also been run inside `cv32e40p/sim/core`.

PS We would like to support VCS flow. I will try to make a pull request in a couple of days.

closed 2023-01-02 by abdullahyildiz