openhwfoundation/core-v-verif #1561: SP makefile CFLAGS value is not changed with user defined flags

alpkoyun, 2023-01-03:
## Bug Title
BSP makefile CFLAGS value does not change with user defined flags

### Type
* Compile error (

### Steps to Reproduce
follow the steps in the "https://docs.openhwgroup.org/projects/core-v-verif/en/latest/quick_start.html" link and try to simulate hello world with verilator
(if you have a compiled bsp, use make clean to clean it)

### Additional context
Hello I noticed that BSP makefile has CFLAGS value but the value is not changed with user defined RISCV_CFLAGS,
The BSP makefile below uses CFLAGS for compilation as shown below but these flags are default and does not take argument from anywhere. 
https://github.com/openhwgroup/core-v-verif/blob/6113f4f179645eb05c4178b6ec25be316a9a1aa8/cv32e40p/bsp/Makefile#L9-L20

Common.mk file which calls the bsp makefile gives arguments shown below
https://github.com/openhwgroup/core-v-verif/blob/6113f4f179645eb05c4178b6ec25be316a9a1aa8/mk/Common.mk#L509-L523

As we can see there is no argument for CFLAG, and the value of the CFLAG is stuck at its default value which produces a compile error.
In the Common.mk file shown above, changing RISCV_CFLAG to CFLAG, or changing the CFLAG to RISCV_CFLAG in the bsp makefile would solve the issue.

Note:Also Verilated.mk which is a file of verilator not core-v-verif, has -Werror command which causes compilation error during rtl compilation, deleting the command solves the issue

MikeOpenHWGroup, 2023-01-03:
Hi @alpkoyun, thanks for the issue.  CFLAGS for the BSP compile are static by design.  The BSP is common code that _everyone_ uses and we wanted it to be the same for all test-programs, regardless of the CFLAGS used for the test-program itself.

Is there a reason you need to compile the BSP with user-defined CFLAGS?

> Note:Also Verilated.mk which is a file of verilator not core-v-verif, has -Werror command which causes compilation error during rtl compilation, deleting the command solves the issue

Please introduce this as a new issue (we try to avoid confusion by discussing one-and-only-one bug report per GitHub issue).  Also, RTL issues should be reported on the RTL repo, not this one.

alpkoyun, 2023-01-03:
Hi,
I just wanted to run the hello world program given in quick start guide and the program did not compile

MikeOpenHWGroup, 2023-01-03:
> I think a quick start guide should work on the first try.

AFAIK, it does, and it is _very_ important to us that it should always "just work".  I have just tried it and it worked for me.  Please follow the instructions in the QSG exactly.  If it does not work for you, please provide the exact commands you used and attach your logfile to this issue.   Thanks!

> I assumed that Bsp makefile was meant to be generic and changeable by users

That is almost true.  The BSP is meant to be generic.  That is, it supports all test-programs that can run on the CV32E40P under core-v-verif.  It is not meant to be changeable by users.  Of course, you are free to do so, but we cannot guarantee that will work.

alpkoyun, 2023-01-03:
I get a compilation when I run "make" in cv32e40p/sim/core
`/opt/riscv/riscv32-embecosm-gcc-ubuntu2004-20221030/bin/riscv32-unknown-elf-gcc -Os -g -static -mabi=ilp32 -march=rv32imc -Wall -pedantic -c /opt/riscv/core-v-verif/cv32e40p/bsp/crt0.S -o crt0.o
/opt/riscv/core-v-verif/cv32e40p/bsp/crt0.S: Assembler messages:
/opt/riscv/core-v-verif/cv32e40p/bsp/crt0.S:33: Error: unrecognized opcode `csrw mtvec,a0', extension `zicsr' required
make[3]: *** [Makefile:20: crt0.o] Error 1
`
If the startup configuration file crt0.s includes an instruction that requires zicsr extension, is not it expected that the error would be present since the default CFLAGS in the bsp makefile does not include zicsr extention.

MikeOpenHWGroup, 2023-01-03:
Ah!  Its the toolchain.  By unhappy coincidence we have learned that the newer versions of the toolchain will not compile the CSR instructions without `zicsr`. I am running an older toolchain which does not have this problem.  We discovered this issue just before the holidays and I had forgotten about it.

We have just merged a pull-request to address this [link](https://github.com/openhwgroup/core-v-verif/pull/1564).   Can you give that a try?  Note that the PR was to the `cv32e40p/dev` branch only (it will be fully tested there before it is merged up to the master branch).

alpkoyun, 2023-01-03:
Thank you all for the quick response and solution, the update worked.

closed 2023-01-03 by alpkoyun

MikeOpenHWGroup, 2023-01-03:
Great!   The fix will be propogated to the master branch in the coming days.  (We will want to soak it on the `cv32e40p/dev` branch first.)