openhwfoundation/core-v-verif #1568: UVM simulation support for xsim (Vivado)

MikeDoingThings, 2023-01-08:
## UVM simulation support for xsim (Vivado)
Hello everyone! When looking into the core-v projects, it seems to me that currently there is no support for a non-cost simulator that runs UVM tests. Since I do not see myself buying an expensive license, I am checking if I can implement support for Vivado's internal simulator. This would also be a nice practice in setting up UVM and hopefully I get a better grasp of the core-v projects before I start to tinker with the processors. However, the main motivation is to reduce the barrier for contributions and productive work for people without an expensive company or academic infrastructure. 

### Open questions
- Is this actually useful and desirable? Are there existing, non-cost options for running UVM tests that I have not yet discovered?
- So far, I believe that xsim should be capable of running UVM tests, but it is suspicious to me that no one has implemented support by now. If there are any known problems with e. g. Vivado's UVM library, I appreciate any heads-up. :) 
- below I listed some files that I assume have to be created/changed. The list is based on my still-not-that-good knowledge of core-v-verif, so again, I appreciate any tips or insights.

### Task Outcome
Implement makefile support for UVM testbenches using Vivado Simulator (xsim). In a best case scenario, the UVM tests for all processors will run as it does with any other simulator.

### Location Information
 - create new file xsim.mk in mk/uvmt/
 - add support for xsim to processor specific UVM makefiles, e. g. cv32e40p/sim/uvmt/Makefile

### Completion Criteria
UVM tests should work as good in Vivado as they do with other simulators.

### Additional context
Upon a quick search I read that the UVM library is precompiled when installing Vivado. I believe that for UVM tests in xsim it might be obligatory to download the UVM source code and point UVM_HOME to it.

MikeOpenHWGroup, 2023-01-09:
Hi @MikeDoingThings, thanks for your interest in CORE-V-VERIF.  Lots to unpack here, let me see if I can answer your questions:

> it seems to me that currently there is no support for a non-cost simulator that runs UVM tests

That is correct.  Our philosophy on this topic is [here](https://docs.openhwgroup.org/projects/core-v-verif/en/latest/intro.html#a-note-about-eda-tools).

> I am checking if I can implement support for Vivado's internal simulator.

At least one member of the OpenHW Group has attempted this in the past. As of 2021.1 Vivado is not able to compile/elaborate/run the CORE-V-VERIF environment.  We are "all in" with SystemVerilog and UVM so you'll need a complete implementation of the SV LRM.  To date I am aware of five commercial tools that can run CORE-V-VERIF:

- Aldec Rivera-Pro
- Cadence Xcelium
- Metrics DSim
- Siemens Questa
- Synopsys VCS

> the main motivation is to reduce the barrier for contributions and productive work

We absolutely support this goal!  If you are successful getting Vivado to work and are willing/able to contribute your updates to this repository, that would be most welcome.  You seem to have a good idea of where to start.  Let me know if I can be of assistance.

---mike (yes, yet another "mike")

MikeOpenHWGroup, 2024-04-05:
Hi @MikeDoingThings, since we haven't heard from you in some time, and AMD/Xilinx has not shown much interest in supporting UVM environments, I am closing this issue.

closed 2024-04-05 by MikeOpenHWGroup