openhwfoundation/core-v-verif #1569: Use this verification environment to verify the problem of CV32E40P core found

guojiazhuxi, 2023-01-09:
<br>**Environment**: OS：VMware Ubuntu 20.04.5 LTS  VCS:O-2018.09-SP2-3

## Bug 1：
core-v-verif/lib/uvm_agents/uvma_obi_memory/src/comps/uvma_obi_memory_cov_model.sv Line 51-68 of the file, use VCS to compile and report an error. The error is shown in the figure below.
[image]
In order to run smoothly, I first commented out lines 51-68
## Bug 2：
In core-v-verif/mk/Common.mk file. The CFLAGS variable in line 316 of the file should be export, Because this variable is transmitted to the Makefile file behind.
## Bug 3：
Line 31 in file core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p.flist, It should not be $(DV_SVLIB_PATH)，It should be ${DV_SVLIB_PATH}
## Bug 4：In file core-v-verif/mk/vcs.mk.
    1.Line 33, "SIMV             = $(CV_TOOL_PREFIX) simv -licwait 20" is error , It should be "SIMV             = $(CV_TOOL_PREFIX)simv -licwait 20" ,No space before simv
    2.Line 189, "cd $(SIM_CFG_RESULTS)/$(CFG) && $(VCS) $(VCS_COMP) -top uvmt_$(CV_CORE_LC)_tb" is error ,It should be "cd $(SIM_CFG_RESULTS) && $(VCS) $(VCS_COMP) -top uvmt_$(CV_CORE_LC)_tb"
   3.Line 223,VCS_RESULTS is no define var, It should be use SIM_RESULTS

##problem:
After modifying the above problems and verification software, I took the following steps. May I ask, how can I join the License of OVPSIM?
[image]

MikeOpenHWGroup, 2023-01-09:
Hi @guojiazhuxi, thanks for your issue.  AFAIK, currently there are no OpenHW members using VCS for CV32E40P in CORE-V-VERIF, so the Makefiles have suffered from "bit rot".

We try to avoid discussing multiple topics in a signal issue, so we will use this issue to resolve **Bug 1** only.
- For **Bug 2** and **Bug 3** please create a pull-request to resolve these.  (With our thanks!)
- For **Bug 4** you will need a license from [Imperas Software](https://www.imperas.com/).

The remainder of this issue should focus on **Bug 1** only.  It seems that VCS does not like the _with_ clauses in the ignore_bins of [this](https://github.com/openhwgroup/core-v-verif/blob/6113f4f179645eb05c4178b6ec25be316a9a1aa8/lib/uvm_agents/uvma_obi_memory/src/comps/uvma_obi_memory_cov_model.sv#L51) code:
```
we: coverpoint (trn.access_type) {
      ignore_bins IGN_WRITE = {UVMA_OBI_MEMORY_ACCESS_WRITE} with (!write_enabled);
      ignore_bins IGN_READ =  {UVMA_OBI_MEMORY_ACCESS_READ} with (!read_enabled);
      bins WRITE = {UVMA_OBI_MEMORY_ACCESS_WRITE};
      bins READ = {UVMA_OBI_MEMORY_ACCESS_READ};
   }
```
AFAIK, this is legal SystemVerilog code and it works well with other simulators.  @silabs-robin, can you comment on this?

silabs-robin, 2023-01-10:
About "bug 1", [uvma_obi_memory_cov_model.sv#L52](https://github.com/openhwgroup/core-v-verif/blob/master/lib/uvm_agents/uvma_obi_memory/src/comps/uvma_obi_memory_cov_model.sv#L52).

This seems like the same style we have used much in isacov and is to my knowledge legal SystemVerilog.

One could read the grammar definitions of the LRM to be sure.
And one could start one of the other simulators with COV=1 etc to see if that exact code runs without warning there (I cannot remember ever having seen a warning about that).

guojiazhuxi, 2023-01-10:
@MikeOpenHWGroup 
Thank you! I want to try creating a pull-request

guojiazhuxi, 2023-01-10:
There are also some warnings. Please check whether they are reasonable
```
Warning-[CNST-PPRW] Constraint randomize NULL object warning
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/../../tests/uvmt/base-tests/uvmt_cv32e40p_base_test.sv, 389
  Member object 'test_randvars' was found to be NULL during randomization.
  Please make sure all random variables/arrays/function calls being randomized
  are allocated fully and properly.

Warning-[CNST-PPRW] Constraint randomize NULL object warning
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/../../tests/uvmt/base-tests/uvmt_cv32e40p_base_test.sv, 389
  Member object 'env_cntxt' was found to be NULL during randomization.
  Please make sure all random variables/arrays/function calls being randomized
  are allocated fully and properly.

Warning-[LCA_FEATURES_ENABLED] Usage warning
  LCA features enabled by '-lca' argument on the command line.  For more 
  information regarding list of LCA features please refer to Chapter "LCA 
  features" in the VCS/VCS-MX Release Notes

Warning-[UNK_COMP_ARG] Unknown compile time plus argument used
  Unknown compile time plus argument 'UVM_VERBOSITY=UVM_MEDIUM' is ignored.

 + use `+plusarg_save' to bind in runtime plusargs;
 + use `+plusarg_ignore' to suppress this message.

Warning-[UNK_COMP_ARG] Unknown compile time plus argument used
  Unknown compile time plus argument 'UVM_VERBOSITY=UVM_MEDIUM' is ignored.

Warning-[BMAOSIN] Bind module allows only simple instance name.
/home/tang/Desktop/core-v-verif/core-v-cores/cv32e40p/rtl/../bhv/cv32e40p_wrapper.sv, 108
  In a bind module with instance name list, only instance names that are 
  simple identifiers are applied design wide to instances of the module. Other
  instance names are treated like bind instance.
  Source info: bind cv32e40p_prefetch_controller:  
  core_i.if_stage_i.prefetch_buffer_i.prefetch_controller_i 
  cv32e40p_prefetch_controller_sva #(.DEPTH(DEPTH), .PULP_XPULP(PU ...

Warning-[UII-L] Interface not instantiated
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb_ifs.sv, 30
  Interface 'uvmt_cv32e40p_clk_gen_if' defined in logic library 'work' is 
  never instantiated in design. It will be ignored.

Warning-[SV-CEBS] Code execution before super.new
/home/tang/Desktop/core-v-verif/lib/uvm_libs/uvml_logs/uvml_logs_reg_logger_cbs.sv, 92
uvml_logs_pkg
  Call to 'super.new' must be the first executable statement in constructor of
  class 'uvml_logs_reg_logger_cbs_c'.
  Such cases may lead to wrong results as the object hierarchy is not fully 
  initialized yet. While initializing in declaration, only constant RHS is 
  allowed.

Warning-[IDTS] Ignoring dynamic type sensitivity
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_step_compare.sv, 444
"_vcs_step_compare_if_riscv_retire"
  Dynamic types used in always_comb, always_latch, always (@*) will be ignored
  for the inferred sensitivity list. 
  
  Declared at: 
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb_ifs.sv, 
  289
  

Warning-[IDTS] Ignoring dynamic type sensitivity
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_step_compare.sv, 449
"_vcs_step_compare_if_riscv_trap"
  Dynamic types used in always_comb, always_latch, always (@*) will be ignored
  for the inferred sensitivity list. 
  
  Declared at: 
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb_ifs.sv, 
[... 59 lines cut ...]
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_step_compare.sv, 509
"ev_compare"
  Dynamic types used in always_comb, always_latch, always (@*) will be ignored
  for the inferred sensitivity list. 
  
  Declared at: 
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_step_compare.sv,
  410
  

Warning-[IDTS] Ignoring dynamic type sensitivity
/home/tang/Desktop/core-v-verif/lib/uvm_agents/uvma_obi_memory/src/uvma_obi_memory_assert.sv, 202
"contiguous_be"
  Dynamic types used in always_comb, always_latch, always (@*) will be ignored
  for the inferred sensitivity list. 
  
  Declared at: 
  /home/tang/Desktop/core-v-verif/lib/uvm_agents/uvma_obi_memory/src/uvma_obi_memory_assert.sv,
  188
  

Notice: Ports coerced to inout, use -notice for details

```
Thanks!

MikeOpenHWGroup, 2023-01-10:
Terrific!   Please read [CONTRIBUTING](https://github.com/openhwgroup/core-v-verif/blob/master/CONTRIBUTING.md) beforehand.  It will make the process easier.  :stuck_out_tongue_winking_eye:

abdullahyildiz, 2023-01-11:
@MikeOpenHWGroup,

Section 19.5.1.1 in SystemVerilog 1800-2017 LRM states the following about using _with_ clause in coverbins:

_... In the expression, the name item shall be used to represent the candidate value._

MikeOpenHWGroup, 2023-01-11:
> ... In the expression, the name item shall be used to represent the candidate value.

Sure.  In what way does `ignore_bins IGN_WRITE = {UVMA_OBI_MEMORY_ACCESS_WRITE} with (!write_enabled);` not conform to this statement in the LRM?

abdullahyildiz, 2023-01-11:
I understand from the LRM that it is mandatory to use the keyword `item` inside the expression as in the following example:

```
a: coverpoint x
  {
    bins mod3[] = {[0:255]} with (item % 3 == 0);
  }
```

That is why VCS complains about it.

MikeOpenHWGroup, 2023-01-12:
I can see why you think so, but it is not a requirement to use the keyword `item` in the "with_covergroup_expression".  In the 1800-2017 version of the LRM, section 19.5.1.1 states that
```
In the expression, the name item shall be used to represent the candidate value.
```
So it the "with_covergroup_expression" refers to the "candidate value", then `item` must be used.  However, in the core-v-verif code in question the "with_covergroup_expression" does not refer to the "candidate value".  In our case the "candidate value" is `trn.access_type`, but our "with_covergroup_expression" evaluates `write_enabled`.

I believe our code is LRM compliant.  I notice that you are using an older version of VCS.  Can you try a newer release?

guojiazhuxi, 2023-01-12:
@MikeOpenHWGroup 我将使用2022版本的vcs重新测试一次

abdullahyildiz, 2023-01-12:
@MikeOpenHWGroup,

Sure. I also submitted this issue to Synopsys VCS support team and will inform you of our findings.

guojiazhuxi, 2023-01-12:
@MikeOpenHWGroup 
I can find new versions of VCS, such as VCS of 2020, 20121, and 2022, to test whether it can

guojiazhuxi, 2023-01-13:
It is verified that the same error will be reported on the higher version of vcs. The current VCS version is ` R-2020.12-SP2`.
[image]

abdullahyildiz, 2023-01-13:
@MikeOpenHWGroup,

Here is a summary of the feedback I have received from Synopsys VCS support team:
 
_Although it is not 100% clear if the syntax is part of LRM or not, the semantics for `with` is strictly to be used to filter the values rather than just checking a signal such as `write_enabled`._

_So the error message is correct and what the user should or could use instead is:_
 
`ignore_bins IGN_WRITE = {UVMA_OBI_MEMORY_ACCESS_WRITE} iff (!write_enabled);`

In my humble opinion, these expressions can be left as is (until we are 100% sure that the syntax is not LRM compliant).

PS They are also planning to publish a release to support the aforementioned syntax in a couple of months.

MikeOpenHWGroup, 2023-01-16:
OK @abdullahyildiz, thanks for your understanding.  I agree that we should not (yet) change our code as I believe the problem is with VCS.

Having said that, I am not 100% sure.  I posted a [question](https://verificationacademy.com/forums/systemverilog/alternative-item-within-withcovergroupexpression#reply-110524) to the verification academy forum.  Let's see what Dave Rich says.

MikeOpenHWGroup, 2023-01-16:
[This thread](https://verificationacademy.com/forums/coverage/clause-error) in the verification academy forum supports the position that the code is compliant.  I would recommend that you [apply a patch](https://gist.github.com/nepsilon/22bc62a23f785716705c) until such time as a fix for this is released by your vendor.

MikeOpenHWGroup, 2023-06-23:
Update: pull-request #2006 should address all the problems raised in this Issue, including the FCIWCS error from VCS on the OBI coverage model.  See the attached file for details.

[FunctionalCoverageInvalidWithClauseSpecification.txt](https://github.com/openhwgroup/core-v-verif/files/11850528/FunctionalCoverageInvalidWithClauseSpecification.txt)

@guojiazhuxi, we have not heard from you so I will close this issue if #2006 gets merged in.

MikeOpenHWGroup, 2024-04-05:
Hi again @guojiazhuxi.  I have just installed VCS V-2023.12-1 locally and can confirm that the compilation issue with VCS has been resolved by Synopsys.  As such, and since #2006 has been merged in, I am closing this issue.  Thanks for your input and feel free to open issues with the OpenHW Group repos at any time.

closed 2024-04-05 by MikeOpenHWGroup