openhwfoundation/core-v-verif #1573: Warnings generated by compiling with VCS

guojiazhuxi, 2023-01-10:
Environment: OS：VMware Ubuntu 20.04.5 LTS VCS:O-2018.09-SP2-3

There are also some warnings. Please check whether they are reasonable
```
Warning-[CNST-PPRW] Constraint randomize NULL object warning
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/../../tests/uvmt/base-tests/uvmt_cv32e40p_base_test.sv, 389
  Member object 'test_randvars' was found to be NULL during randomization.
  Please make sure all random variables/arrays/function calls being randomized
  are allocated fully and properly.

Warning-[CNST-PPRW] Constraint randomize NULL object warning
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/../../tests/uvmt/base-tests/uvmt_cv32e40p_base_test.sv, 389
  Member object 'env_cntxt' was found to be NULL during randomization.
  Please make sure all random variables/arrays/function calls being randomized
  are allocated fully and properly.

Warning-[LCA_FEATURES_ENABLED] Usage warning
  LCA features enabled by '-lca' argument on the command line.  For more 
  information regarding list of LCA features please refer to Chapter "LCA 
  features" in the VCS/VCS-MX Release Notes

Warning-[UNK_COMP_ARG] Unknown compile time plus argument used
  Unknown compile time plus argument 'UVM_VERBOSITY=UVM_MEDIUM' is ignored.

 + use `+plusarg_save' to bind in runtime plusargs;
 + use `+plusarg_ignore' to suppress this message.

Warning-[UNK_COMP_ARG] Unknown compile time plus argument used
  Unknown compile time plus argument 'UVM_VERBOSITY=UVM_MEDIUM' is ignored.

Warning-[BMAOSIN] Bind module allows only simple instance name.
/home/tang/Desktop/core-v-verif/core-v-cores/cv32e40p/rtl/../bhv/cv32e40p_wrapper.sv, 108
  In a bind module with instance name list, only instance names that are 
  simple identifiers are applied design wide to instances of the module. Other
  instance names are treated like bind instance.
  Source info: bind cv32e40p_prefetch_controller:  
  core_i.if_stage_i.prefetch_buffer_i.prefetch_controller_i 
  cv32e40p_prefetch_controller_sva #(.DEPTH(DEPTH), .PULP_XPULP(PU ...

Warning-[UII-L] Interface not instantiated
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb_ifs.sv, 30
  Interface 'uvmt_cv32e40p_clk_gen_if' defined in logic library 'work' is 
  never instantiated in design. It will be ignored.

Warning-[SV-CEBS] Code execution before super.new
/home/tang/Desktop/core-v-verif/lib/uvm_libs/uvml_logs/uvml_logs_reg_logger_cbs.sv, 92
uvml_logs_pkg
  Call to 'super.new' must be the first executable statement in constructor of
  class 'uvml_logs_reg_logger_cbs_c'.
  Such cases may lead to wrong results as the object hierarchy is not fully 
  initialized yet. While initializing in declaration, only constant RHS is 
  allowed.

Warning-[IDTS] Ignoring dynamic type sensitivity
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_step_compare.sv, 444
"_vcs_step_compare_if_riscv_retire"
  Dynamic types used in always_comb, always_latch, always (@*) will be ignored
  for the inferred sensitivity list. 
  
  Declared at: 
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb_ifs.sv, 
  289
  

Warning-[IDTS] Ignoring dynamic type sensitivity
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_step_compare.sv, 449
"_vcs_step_compare_if_riscv_trap"
  Dynamic types used in always_comb, always_latch, always (@*) will be ignored
  for the inferred sensitivity list. 
  
  Declared at: 
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb_ifs.sv, 
  290
  

Warning-[IDTS] Ignoring dynamic type sensitivity
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_step_compare.sv, 453
"_vcs_step_compare_if_riscv_halt"
  Dynamic types used in always_comb, always_latch, always (@*) will be ignored
  for the inferred sensitivity list. 
  
  Declared at: 
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb_ifs.sv, 
  291
  

Warning-[IDTS] Ignoring dynamic type sensitivity
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_step_compare.sv, 478
"_vcs_step_compare_if_ovp_cpu_valid"
  Dynamic types used in always_comb, always_latch, always (@*) will be ignored
  for the inferred sensitivity list. 
  
  Declared at: 
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb_ifs.sv, 
  279
  

Warning-[IDTS] Ignoring dynamic type sensitivity
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_step_compare.sv, 479
"$root.uvmt_cv32e40p_tb.dut_wrap.cv32e40p_wrapper_i.tracer_i.ovp_retire"
  Dynamic types used in always_comb, always_latch, always (@*) will be ignored
  for the inferred sensitivity list. 
  
  Declared at: 
  /home/tang/Desktop/core-v-verif/core-v-cores/cv32e40p/rtl/../bhv/cv32e40p_tracer.sv,
  105
  

Warning-[IDTS] Ignoring dynamic type sensitivity
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_step_compare.sv, 483
"_vcs_step_compare_if_ovp_cpu_trap"
  Dynamic types used in always_comb, always_latch, always (@*) will be ignored
  for the inferred sensitivity list. 
  
  Declared at: 
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb_ifs.sv, 
  280
  

Warning-[IDTS] Ignoring dynamic type sensitivity
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_step_compare.sv, 487
"_vcs_step_compare_if_ovp_cpu_halt"
  Dynamic types used in always_comb, always_latch, always (@*) will be ignored
  for the inferred sensitivity list. 
  
  Declared at: 
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb_ifs.sv, 
  281
  

Warning-[IDTS] Ignoring dynamic type sensitivity
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_step_compare.sv, 509
"ev_compare"
  Dynamic types used in always_comb, always_latch, always (@*) will be ignored
  for the inferred sensitivity list. 
  
  Declared at: 
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_step_compare.sv,
  410
  

Warning-[IDTS] Ignoring dynamic type sensitivity
/home/tang/Desktop/core-v-verif/lib/uvm_agents/uvma_obi_memory/src/uvma_obi_memory_assert.sv, 202
"contiguous_be"
  Dynamic types used in always_comb, always_latch, always (@*) will be ignored
  for the inferred sensitivity list. 
  
  Declared at: 
  /home/tang/Desktop/core-v-verif/lib/uvm_agents/uvma_obi_memory/src/uvma_obi_memory_assert.sv,
  188
  

Notice: Ports coerced to inout, use -notice for details

```
Thanks!

MikeOpenHWGroup, 2023-01-10:
Thanks for your question @guojiazhuxi.  VCS was not heavily used in the CV32E40P verification project, so this kind of feedback is valuable.

Many of the warnings are unique to VCS, we do not see them with other simulators.  As such, I am not concerned about them.  Having said that, we would welcome a pull-request to resolve them.

There _is_ one warning that does worry me:
```
Warning-[IDTS] Ignoring dynamic type sensitivity
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_step_compare.sv, 453
"_vcs_step_compare_if_riscv_halt"
  Dynamic types used in always_comb, always_latch, always (@*) will be ignored
  for the inferred sensitivity list. 
  
  Declared at: 
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb_ifs.sv, 
  291
```
The warning is due to a SystemVerilog `event` type that VCS assumes should be in the sensitivity list (see comment in the code segment below). However, apparently VCS ignores events in dynamic sensitivity lists.  Again, other simulators do not emit warnings about this.  AFAIK, the always block in question need only be sensitive to the state variable, so perhaps this line could be changed to `always @(state) begin`.
```
   state_e state = INIT;
   initial state <= IDLE; // cause an event for always @*

   always @(*) begin   // line 433: might be able to change to always @(state) begin
     if (use_iss) begin
        case (state)
          IDLE: begin
              state <= RTL_STEP;
          end

          RTL_STEP: begin
              clknrst_if.start_clk();
              fork
                  begin
                      @step_compare_if.riscv_retire; // line 444: this is causing the warning
                      clknrst_if.stop_clk();
                      state <= RTL_VALID;
                  end
```

This code has been in production for many months and is stable, so I am reluctant to change it.  However, if this warning means that the code will not work under VCS, please try my suggestion and if that works, submit a pull-request to resolve.

Thanks!

guojiazhuxi, 2023-01-10:
thank you! I will try it

guojiazhuxi, 2023-01-12:
When I change to `always @(state) begin`, it will not report warning！

MikeOpenHWGroup, 2023-01-12:
Terrific.  Of course, the absence of a compile warning does not mean the code will work. :wink:  I tried it locally and it passed a CI regression, so if you would like to submit a pull-request for this I would be happy ask others to review it.

Something you should be aware of is that this module, `uvmt_cv32e40p_step_compare.sv`, will be deprecated in the near future as we are planning to transition the CV32E40P environment to use the ImperasDV reference model which does not require a "step-and-compare" testbench.

guojiazhuxi, 2023-01-16:
@MikeOpenHWGroup Last question：
When ./simv, Make will report ERROR
It will be output at the end `An unexpected termination has occurred in ./simv due to a signal: Segmentation fault`
[image]
I think it may be a problem with DPI

MikeOpenHWGroup, 2023-01-16:
Do you have an Imperas license?  It not, the UVM environment will not run.  You can disable the Imperas reference model with the `USE_ISS` Makefile variable:
```
make test TEST=hello-world USE_ISS=NO
```

guojiazhuxi, 2023-01-16:
This is the command I run:
make test TEST=hello-world CV_SIMULATOR=vcs USE_ISS=NO

guojiazhuxi, 2023-01-16:
`+DISABLE_OVPSIM` options are used in ./simv

pascalgouedo, 2023-01-16:
It seems $test$plusargs is missing around svimp_exit();

```
`ifndef UVM
    final begin
        if (!$test$plusargs("DISABLE_OVPSIM")) begin
            svimp_exit();
        end
    end
`endif
```

guojiazhuxi, 2023-01-17:
With pleasure!

guojiazhuxi, 2023-01-17:
When I switch the branch to CV32E40P/dev, run VCS, and the following errors have been made:
```
Warning-[TFIPC] Too few instance port connections
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb.sv, 147
cv32e40p_tb_wrapper, "uvma_rvfi_instr_if #(uvme_cv32e40p_pkg::ILEN, uvme_cv32e40p_pkg::XLEN) rvfi_instr_if_0_i( .clk (clk_i),  .reset_n (rst_ni),  .rvfi_valid (rvfi_i.rvfi_valid[0]),  .rvfi_order (rvfi_i.rvfi_order[(uvma_rvfi_pkg::ORDER_WL * 0)+:uvma_rvfi_pkg::ORDER_WL]),  .rvfi_insn (rvfi_i.rvfi_insn[(uvme_cv32e40p_pkg::ILEN * 0)+:uvme_cv32e40p_pkg::ILEN]),  .rvfi_trap (rvfi_i.rvfi_trap),  .rvfi_halt (rvfi_i.rvfi_halt[0]),  .rvfi_intr (rvfi_i.rvfi_intr.intr),  .rvfi_dbg (rvfi_i.rvfi_dbg),  .rvfi_dbg_mode (rvfi_i.rvfi_dbg_mode),  .rvfi_mode (rvfi_i.rvfi_mode[(uvma_rvfi_pkg::MODE_WL * 0)+:uvma_rvfi_pkg::MODE_WL]),  .rvfi_ixl (rvfi_i.rvfi_ixl[(uvma_rvfi_pkg::IXL_WL * 0)+:uvma_rvfi_pkg::IXL_WL]),  .rvfi_pc_rdata (rvfi_i.rvfi_pc_rdata[(uvme_cv32e40p_pkg::XLEN * 0)+:uvme_cv32e40p ... "
  The above instance has fewer port connections than the module definition.
  Please use '+lint=TFIPC-L' to print out detailed information of unconnected 
  ports.

Warning-[UNGB] Unnamed generate block
  Unnamed generate block at:
  "/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_interrupt_assert.sv",
  387
  VCS has encountered unnamed generate blocks. Accessing declarations in 
  unnamed generate blocks will result in cross module reference resolution 
  errors (XMRE).
  Next steps:
     1. Label all the unnamed generate blocks (preferred).
     2. Or, use -v2k_generate compile switch to enable cross module reference 
  for unnamed generate blocks. This switch will be obsolete in future VCS 
  versions.

Error-[XMRE] Cross-module reference resolution error
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb.sv, 150
  Error found while trying to resolve cross-module reference.
  token 'rvfi_i'.  Originating module 'cv32e40p_tb_wrapper'.
  Source info:  .rvfi_valid (rvfi_i.rvfi_valid[0])
  Instance stack trace:
      cv32e40p_tb_wrapper#(0, 0, 0, 0, 1)  
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_dut_wrap.sv, 
  143
      uvmt_cv32e40p_dut_wrap#(0, 0, 0, 0, 1, 32, 32, 22)  
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb.sv, 133
      uvmt_cv32e40p_tb  
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb.sv, 31
  

Error-[XMRE] Cross-module reference resolution error
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb.sv, 151
  Error found while trying to resolve cross-module reference.
  token 'rvfi_i'.  Originating module 'cv32e40p_tb_wrapper'.
  Source info:  .rvfi_order (rvfi_i.rvfi_order[(uvma_rvfi_pkg::ORDER_WL * 
  0)+:uvma_rvfi_pkg::ORDER_WL])
  Instance stack trace:
      cv32e40p_tb_wrapper#(0, 0, 0, 0, 1)  
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_dut_wrap.sv, 
  143
      uvmt_cv32e40p_dut_wrap#(0, 0, 0, 0, 1, 32, 32, 22)  
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb.sv, 133
      uvmt_cv32e40p_tb  
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb.sv, 31
  

Error-[XMRE] Cross-module reference resolution error
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb.sv, 152
  Error found while trying to resolve cross-module reference.
  token 'rvfi_i'.  Originating module 'cv32e40p_tb_wrapper'.
  Source info:  .rvfi_insn (rvfi_i.rvfi_insn[(uvme_cv32e40p_pkg::ILEN * 
  0)+:uvme_cv32e40p_pkg::ILEN])
  Instance stack trace:
      cv32e40p_tb_wrapper#(0, 0, 0, 0, 1)  
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_dut_wrap.sv, 
  143
      uvmt_cv32e40p_dut_wrap#(0, 0, 0, 0, 1, 32, 32, 22)  
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb.sv, 133
      uvmt_cv32e40p_tb  
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb.sv, 31
  

Error-[XMRE] Cross-module reference resolution error
[... 86 lines cut ...]
      uvmt_cv32e40p_tb  
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb.sv, 31
  

Error-[XMRE] Cross-module reference resolution error
/home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb.sv, 160
  Error found while trying to resolve cross-module reference.
  token 'rvfi_i'.  Originating module 'cv32e40p_tb_wrapper'.
  Source info:  .rvfi_ixl (rvfi_i.rvfi_ixl[(uvma_rvfi_pkg::IXL_WL * 
  0)+:uvma_rvfi_pkg::IXL_WL])
  Instance stack trace:
      cv32e40p_tb_wrapper#(0, 0, 0, 0, 1)  
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_dut_wrap.sv, 
  143
      uvmt_cv32e40p_dut_wrap#(0, 0, 0, 0, 1, 32, 32, 22)  
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb.sv, 133
      uvmt_cv32e40p_tb  
  /home/tang/Desktop/core-v-verif/cv32e40p/tb/uvmt/uvmt_cv32e40p_tb.sv, 31
  

Note-[MAX_ERROR_COUNT] Maximum error count reached
  Current number of errors has reached the default maximum error count (10).
  Please use +error+<count> to increase the limit.

```

pascalgouedo, 2023-01-17:
Hi @guojiazhuxi 

Did you add
VCS_USER_COMPILE_ARGS += +define+$(CV_CORE_UC)_RVFI
to your VCS scripts?

guojiazhuxi, 2023-01-17:
@pascalgouedo ,
no , the vcs scripts no the define!

guojiazhuxi, 2023-01-17:
It works normally!

MikeOpenHWGroup, 2023-01-18:
Can this issue be closed now @guojiazhuxi?

closed 2023-01-28 by guojiazhuxi