openhwfoundation/core-v-verif #1582: ISACOV Agent : wrong decode format

AyoubJalali, 2023-01-16:
Hello,
So I notice something suspicion about assign some C_format in the [file](https://github.com/openhwgroup/core-v-verif/blob/master/lib/uvm_agents/uvma_isacov/uvma_isacov_instr.sv) : 

1. line [336](https://github.com/openhwgroup/core-v-verif/blob/6113f4f179645eb05c4178b6ec25be316a9a1aa8/lib/uvm_agents/uvma_isacov/uvma_isacov_instr.sv#L336) to https://github.com/openhwgroup/core-v-verif/blob/6113f4f179645eb05c4178b6ec25be316a9a1aa8/lib/uvm_agents/uvma_isacov/uvma_isacov_instr.sv#L338, a duplicate IF, so should be removed.
2. https://github.com/openhwgroup/core-v-verif/blob/6113f4f179645eb05c4178b6ec25be316a9a1aa8/lib/uvm_agents/uvma_isacov/uvma_isacov_instr.sv#L395, I think it's a CL_TYPE for dasm_rvc_lw_imm(instr) not CS_TYPE because there is no function for CS_TYPE in dasm library.
3. https://github.com/openhwgroup/core-v-verif/blob/6113f4f179645eb05c4178b6ec25be316a9a1aa8/lib/uvm_agents/uvma_isacov/uvma_isacov_instr.sv#L436 to https://github.com/openhwgroup/core-v-verif/blob/6113f4f179645eb05c4178b6ec25be316a9a1aa8/lib/uvm_agents/uvma_isacov/uvma_isacov_instr.sv#L442, it's a rd function so why returning rs2 in [439](https://github.com/openhwgroup/core-v-verif/blob/6113f4f179645eb05c4178b6ec25be316a9a1aa8/lib/uvm_agents/uvma_isacov/uvma_isacov_instr.sv#L439), beside that i thing there is some miss assign types.

I think it should be like :  `return (itype inside {CA_TYPE, CL_TYPE, CIW_TYPE}) ? rd[2:0] : rd;`

MikeOpenHWGroup, 2023-01-16:
Hi @silabs-robin, I know it is unfair to ask as this particular code was written by Steve Richmond, but I think @AyoubJalali has a point.   Can you have a look?

@AyoubJalali, just FYI, Steve Richmond was one of the original Contributors to core-v-verif.  Steve is no longer a member of the OpenHW Group.

silabs-robin, 2023-01-23:
> 1. line [336](https://github.com/openhwgroup/core-v-verif/blob/6113f4f179645eb05c4178b6ec25be316a9a1aa8/lib/uvm_agents/uvma_isacov/uvma_isacov_instr.sv#L336) to [338](https://github.com/openhwgroup/core-v-verif/blob/6113f4f179645eb05c4178b6ec25be316a9a1aa8/lib/uvm_agents/uvma_isacov/uvma_isacov_instr.sv#L338), a duplicate IF, so should be removed.

I guess "331" and "336" is more accurate, but yes one of them can be removed.

[image]

silabs-robin, 2023-01-23:
> 2\. I think it's a CL_TYPE for dasm_rvc_lw_imm(instr) not CS_TYPE because there is no function for CS_TYPE in dasm library.

What do they look like in the tables of the spec? Are they compatible? Are they mapped incorrectly?

AyoubJalali, 2023-02-02:
> > 2. I think it's a CL_TYPE for dasm_rvc_lw_imm(instr) not CS_TYPE because there is no function for CS_TYPE in dasm library.
> 
> What do they look like in the tables of the spec? Are they compatible? Are they mapped incorrectly?

I don't think they have the same format

AyoubJalali, 2023-02-02:
> > 3. , it's a rd function so why returning rs2 in [439](https://github.com/openhwgroup/core-v-verif/blob/6113f4f179645eb05c4178b6ec25be316a9a1aa8/lib/uvm_agents/uvma_isacov/uvma_isacov_instr.sv#L439),
> 
> What does the relevant bitfields in the table of the spec look like? How does it compare to the code?
> 
> > I think it should be like : `return (itype inside {CA_TYPE, CL_TYPE, CIW_TYPE}) ? rd[2:0] : rd;`
> 
> Without `CB_TYPE`?

I think i should add CB_type because some instructions use rd'/rs1', you're right

AyoubJalali, 2023-02-02:
@MikeOpenHWGroup , @silabs-robin 
I open i PR to fix this issue, can you look at it ? (PR #1611)

Thanks

silabs-robin, 2023-02-03:
> https://github.com/openhwgroup/core-v-verif/blob/6113f4f179645eb05c4178b6ec25be316a9a1aa8/lib/uvm_agents/uvma_isacov/uvma_isacov_instr.sv#L395
> , I think it's a CL_TYPE for dasm_rvc_lw_imm(instr) not CS_TYPE because there is no function for CS_TYPE in dasm library.

The code in question ("uvma_isacov_instr_c::get_field_imm()") is about "when given a CS_TYPE instr, what is its immediate?".
If one changes the if-statement to "CL_TYPE", what will one get when one tries to fetch the immediate of a "CS_TYPE"?
Seems like it would just always return 0, regardless of what the actual immediate is.

However, it seems like the opposite case might be true today.
So maybe it should be something similar to `if (this.itype inside {CS_TYPE, CL_TYPE}) begin` (I haven't thought it all the way through).

_(If dasm doesn't have a dasm_rvc_sw_imm it might be because they realized they have the same format and just one of them could be re-used?)_

--------

>> What do they look like in the tables of the spec? Are they compatible? Are they mapped incorrectly?
>
> I don't think they have the same format

Are we talking about these?

[image]

[image]

[image]

silabs-robin, 2023-02-03:
> What does the relevant bitfields in the table of the spec look like? How does it compare to the code?

This is what it looks like in the spec:

[image]

(Note that CB doesn't say " rd' " in the table, but in effect it has one.)

[image]

--------

Comparing that with the original code:

[image]

Note that `uvma_isacov_instr_c::get_field_rd()` first check CA and CB.
They used the "primed" version of rd, only bits [2:0].

Then it checks CIW and CL.
They both have their " rd' " placed where rs2 is normally used, hence rs2[2:0].

Finally, the default is just the full 5-bit "rd" non-primed field.

With the new code I think CIW and CL would return their "imm" and " rs1' " as their "rd" value.

--------

I agree that it looks unclean that the code basically says `rd = rs2`, so I would be in favor of changing the naming of the class' fields to better reflect what these properties actually mean.

silabs-robin, 2023-02-03:
Side note:
Maybe we could link to https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/decode.h in the lib/dpi_dasm/ readme?
It is very handy, and always a tiny bit too difficult to find.

MikeOpenHWGroup, 2023-02-03:
> Maybe we could link to https://github.com/riscv-software-src/riscv-isa-sim/blob/master/riscv/decode.h in the lib/dpi_dasm/ readme?

Good idea!  Can you take care of that in a separate PR @silabs-robin?

AyoubJalali, 2023-02-06:
> > > 3. , it's a rd function so why returning rs2 in [439](https://github.com/openhwgroup/core-v-verif/blob/6113f4f179645eb05c4178b6ec25be316a9a1aa8/lib/uvm_agents/uvma_isacov/uvma_isacov_instr.sv#L439),
> > 
> > 
> > What does the relevant bitfields in the table of the spec look like? How does it compare to the code?
> 
> This is what it looks like in the spec:
> 
> [image]
> 
> (Note that CB doesn't say " rd' " in the table, but in effect it has one.)
> 
> [image]
> 
> Comparing that with the original code:
> 
> [image]
> 
> Note that `uvma_isacov_instr_c::get_field_rd()` first check CA and CB. They used the "primed" version of rd, only bits [2:0].
> 
> Then it checks CIW and CL. They both have their " rd' " placed where rs2 is normally used, hence rs2[2:0].
> 
> Finally, the default is just the full 5-bit "rd" non-primed field.
> 
> If we just did the `rd = rd` simplification I think CIW and CL would return their "imm" and " rs1' " as their "rd" value. But maybe it is all resolved when using `dasm_rvc_rd()`?
> 
> I agree that it looks unclean that the code basically says `rd = rs2`, so I would also be in favor of changing the naming of the class' fields to better reflect what these properties actually mean.

Yes understood now, It's just I didn't get this rd = rs2, now it's clear , thanks

AyoubJalali, 2023-02-06:
> If we just did the `rd = rd` simplification I think CIW and CL would return their "imm" and " rs1' " as their "rd" value. But maybe it is all resolved when using `dasm_rvc_rd()`?

I should look to the implementation of these functions in dasm, could you point where i can look on these function if you know ? thanks

silabs-robin, 2023-02-15:
> I should look to the implementation of these functions in dasm, could you point where i can look on these function if you know ? thanks

Hi. Maybe you found it already, but if you didn't I also added a link in the dpi_dasm readme here:
https://github.com/openhwgroup/core-v-verif/pull/1643/files

commit aa4f0cae (2023-03-27): ISACOV: FIX #1582
commit b737183d (2023-03-27): Merge pull request #1611 from AyoubJalali/pr/isacov

ISACOV: FIX #1582
closed 2023-05-26 by JeanRochCoulon