openhwfoundation/core-v-verif #1587: (0.8.0) ISS Debug triggers default to type 6, should be type 2

silabs-hfegran, 2023-01-17:
## ISS default debug trigger type incorrect
ISS currently expects trigger type 6, while 0.7.0/0.8.0 spec states that the default type should be type 2 which leads to mismatches if the CSR check for tdata1 and tinfo are reenabled in the wrapper.

### Type
* Functionally incorrect behavior

### Steps to Reproduce
Not available directly with published code, the following steps will illustrate the issue:
Requires latest RTL >= 0.8.0 and removed CSR-check disable from imperas_dv wrapper
cvverif 40s >= 3cb7442c634a348d38f2a0fd85246029a7714ce3
cvverif 40x >= 8b6470a517e6707a2d532af91bb32aa38a75d151
The above also require the most recent dev-version of the ISS binary with 0.8.0 spec changes, not publicly available.

### Additional context

Log output:
> Info (IDV)  2> CSR 7a1 (tdata1)
Info (IDV)   . dut:0x28001000 
Info (IDV)   . ref:0x68001000 type:6(mcontrol6) dmode:1 hit:0 select:0 timing:0 size:0(any) action:1(enter debug mode) chain:0 match:0 m:0 execute:0 store:0 load:0
Info (IDV)  3> CSR 7a4 (tinfo)
Info (IDV)   . dut:0x00008064 
Info (IDV)   . ref:0x00008060

eroom1966, 2023-01-17:
Hi @silabs-hfegran 
What was the version of the Imperas Ref model you were using to find this issue ?

eroom1966, 2023-01-17:
@silabs-hfegran 
I am a little confused about this issue you have reported
Did you try this test after taking the Pullrequest you accepted today
That PR should explicitly address the issue on tdata1, also you said

> ISS currently expects trigger type 6, while 0.7.0/0.8.0 spec states that the default type should be type 2 which leads to mismatches if the CSR check for tdata1 and tinfo are reenabled in the wrapper.

tdata1 is not disabled in the code you provided, although DCSR and TINFO are disabled for comparison

maybe we should have a shared desktop

silabs-hfegran, 2023-01-18:
@eroom1966 This is really odd, I am unable to reproduce the issue today on 40S (without changing any of the ISS files) - so I guess this a non issue. I will reopen this if I see this resurfacing.

Are these changes in for both cores or just the 40s?.

silabs-hfegran, 2023-01-18:
Simulator: hash `d1c7f33d5df4bec88df44265f0ce3e0840169bc1`  from your repo

>ImperasDVasync (64-Bit) v20230113.0 Open Virtual Platform simulator from www.IMPERAS.com.

I did reenable the TINFO and DCSR checks after this, but I didn't do extensive testing yet, just a quick hello-world and a couple of other tests as I saw this failing yesterday. 

Which tests are failing for you?

eroom1966, 2023-01-18:
I was running the 'sanity' test
I did have a mismatch in TINFO, this is related to the new mode that was added - my bad, I have a fix I can upload shortly

BUT for the DCSR, I think the REF model is doing the right thing, but the RTL is not, eg mprven should be hardwired to '1', yet the RTL is returning 0

```
Error (IDV) CSR register value mismatch (HartId:0, PC:0x00000080 _start+0):
Info (IDV)  0> CSR 7b0 (dcsr)
Info (IDV)   . dut:0x40000003 debugver:4 ebreakm:0 ebreaku:0 stepie:0 stopcount:0 stoptime:0 cause:0 mprven:0 nmip:0 prv:3(M-mode)
Info (IDV)   . ref:0x40000413 debugver:4 ebreakm:0 ebreaku:0 stepie:0 stopcount:1 stoptime:0 cause:0 mprven:1 nmip:0 prv:3(M-mode)
```

eroom1966, 2023-01-18:
TINFO does not match given the instructions above, TINFO is producing the value from RTL of 0x00008060, it should be 0x00008062

eroom1966, 2023-01-24:
correction the tinfo should be 0x8064

eroom1966, 2023-01-24:
@silabs-hfegran can you please try this again with the latest commit - this has the corrected setting for TINFO

eroom1966, 2023-01-27:
@silabs-hfegran can this issue be closed, I think it is working as expected now ?

closed 2023-01-31 by silabs-hfegran