openhwfoundation/core-v-verif #1592: ISS mismatch on debug trigger match

silabs-halfdan, 2023-01-18:
A debug trigger match causes ISS mismatches.

This seems to be because we are not able to communicate that a trigger match ocurred as rvfi signals this by setting the debug trap signal which is gated before being assigned to rvvi.

### Steps to Reproduce
1. checkout the following branch:
     https://github.com/silabs-halfdan/core-v-verif/tree/debug_test_trigger_iss_mismatch_example
2. `makeuvmt test TEST=debug_test_trigger`
3. Results in the following errors: 
Error (IDV) PC mismatch (HartId:0, PC:0x1a110800 _debugger_start+0):
Info (IDV)  0>
Info (IDV)   . dut:0x00000108 trigger_code+0
Info (IDV)   . ref:0x1a110800 _debugger_start+0
Error (IDV) Insn. bit pattern mismatch (HartId:0, PC:0x1a110800 _debugger_start+0):
Info (IDV)  1>
Info (IDV)   . dut:1141     c.addi  x2,x2,-16
Info (IDV)   . ref:b8f2     cm.push {x1,x8-x9,x18-x27},-64
Error (IDV) GPR register value mismatch (HartId:0, PC:0x1a110800 _debugger_start+0):
Info (IDV)  2> GPR x2
Info (IDV)   . dut:0x003fffd0 
Info (IDV)   . ref:0x003fff90

silabs-halfdan, 2023-01-19:
@eroom1966

eroom1966, 2023-01-19:
@silabs-halfdan @Silabs-ArjanB 

My environment is not equivalent to yours, in which case can I have a few more instructions please, firstly I have done the following

git clone https://github.com/silabs-halfdan/core-v-verif
export PATH=${PATH}:$(pwd)/core-v-verif/bin
cd core-v-verif
git checkout debug_test_trigger_iss_mismatch_example
export SIMULATOR=xrun
export CV_SW_TOOLCHAIN=/home/moore/riscv/riscv
export CV_CORE=cv32e40s
export COREV=YES
export CV_SW_PREFIX=riscv32-corev-elf-
export CV_SW_VENDOR=corev

makeuvmt test TEST=debug_test_trigger

I get a number of error messages during compilation
```
* Compiling test-program /home/moore/git/silabs/core-v-verif/cv32e40s/sim/uvmt/xrun_results/default/debug_test_trigger/0/test_program/debug_test_trigger.elf
*******************************************************************************************
/home/moore/riscv/riscv/bin/riscv32-corev-elf-gcc \
	 \
	-Os -g -static -mabi=ilp32 -march=rv32im_zba1p00_zbb1p00_zbc1p00_zbs1p00_zicsr_zca_zcmp_zifencei -Wall -pedantic  \
	 \
	-I ../../tests/asm \
	-I /home/moore/git/silabs/core-v-verif/cv32e40s/bsp \
	-o /home/moore/git/silabs/core-v-verif/cv32e40s/sim/uvmt/xrun_results/default/debug_test_trigger/0/test_program/debug_test_trigger.elf \
	-nostartfiles \
	/home/moore/git/silabs/core-v-verif/cv32e40s/tests/programs/custom/debug_test_trigger/debug_test_trigger.c \
	-T /home/moore/git/silabs/core-v-verif/cv32e40s/bsp/link.ld \
	 -L /home/moore/git/silabs/core-v-verif/cv32e40s/sim/uvmt/xrun_results/default/debug_test_trigger/0/test_program/bsp \
	-lcv-verif
/home/moore/riscv/corev-openhw-gcc-ubuntu1804-20221021/bin/../lib/gcc/riscv32-corev-elf/12.0.1/../../../../riscv32-corev-elf/bin/ld: Zcm* is not compatible with the `c' extension.
/home/moore/riscv/corev-openhw-gcc-ubuntu1804-20221021/bin/../lib/gcc/riscv32-corev-elf/12.0.1/../../../../riscv32-corev-elf/bin/ld: failed to merge target specific data of file /home/moore/riscv/corev-openhw-gcc-ubuntu1804-20221021/bin/../lib/gcc/riscv32-corev-elf/12.0.1/../../../../riscv32-corev-elf/lib/libc.a(libc_a-atexit.o)
/home/moore/riscv/corev-openhw-gcc-ubuntu1804-20221021/bin/../lib/gcc/riscv32-corev-elf/12.0.1/../../../../riscv32-corev-elf/bin/ld: rv32i2p1_m2p0_a2p1_c2p0: standard ISA extension `i' is not in canonical order

```

I suspect a compiler issue, the version of my compiler is 
riscv32-corev-elf-gcc ('corev-openhw-gcc-ubuntu1804-20221021') 12.0.1 20220422 (experimental)

Please advise
if this is wrong please provide a link to the correct compiler

silabs-halfdan, 2023-01-19:
Ah right, I'm using this compiler version: corev-openhw-gcc-centos7-20221213

eroom1966, 2023-01-19:
@silabs-halfdan can you provide a download link to the correct compiler
Thx
Lee

silabs-halfdan, 2023-01-19:
@eroom1966 it's this version directly from the embercosm jenkins build: https://buildbot.embecosm.com/job/corev-gcc-centos7/7/artifact/corev-openhw-gcc-centos7-20221213.tar.gz as we needed some features before they were properly released.

Using a later version from https://www.embecosm.com/resources/tool-chain-downloads/ should also work.

eroom1966, 2023-01-20:
@silabs-halfdan I have this working and produces this output with no errors

eroom1966, 2023-01-20:
The question is how to decode from your RVFI if an instruction traps orretires ?

silabs-halfdan, 2023-01-25:
Yes, rvfi_trap.trap is indeed intended as the way to decode whether the instruction traps of retires.

eroom1966, 2023-01-26:
@silabs-halfdan 
Is this still broken with the new ImperasDV release for 0.8.0 ?

silabs-halfdan, 2023-01-26:
For the 'ed14846' hash of imperasDV it is still broken. Is the 0.8.0 release newer?

eroom1966, 2023-01-27:
@silabs-halfdan 
Can you please try the following

1. merge the changes I have in this pull request https://github.com/openhwgroup/core-v-verif/pull/1599
2. use the 'main' branch of the ImperasDV reference model

There have been changes in the file uvmt_cv32e40s_imperas_dv_wrap.sv that need updating with the REF model

silabs-halfdan, 2023-01-30:
@eroom1966, perfect, this solves the issue.

silabs-halfdan, 2023-01-30:
Will close this issue when the PR is merged.

closed 2023-01-30 by silabs-halfdan