openhwfoundation/core-v-verif #1594: extension 'zicsr' required

gullahmed1251, 2023-01-22:
I am trying to run the core by following the [CORE-V-VERIF Verification Strategy (https://docs.openhwgroup.org/projects/core-v-verif/en/latest/quick_start.html) guide. I have followed all the mentioned steps but when i am finding an error that extension 'zicsr' not found.
I have got a solution from https://github.com/openhwgroup/cv32e40p/issues/757#issuecomment-1399456925 
, it worked but i am facing new issue.
[image]

Kindly check the attached screenshot and help me to solve the errors.

abukharmeh, 2023-01-22:
Where did you get the RISC-V toolchain from, is this from Ubuntu's official repository or is this from [embecosm](https://www.embecosm.com/resources/tool-chain-downloads/#corev) ? 

You can fix this error the same way you fixed the other one, *by explicitly including zicsr to march*. Searching the common makefile, you see that march can be overridden from the terminal by setting CV_SW_MARCH environment variable. 

Thus run `export CV_SW_MARCH=rv32imc_zicsr` and then rerun make again!

Version 2.2 of RISC-V ISA factored Zicsr out from the baseline. Previously, they were automatically included with integer instructions, but not at the moment.

MikeOpenHWGroup, 2023-01-22:
Hi @gullahmed1251. @abukharmeh is right that this will solve the issue.  A better approach is to simply follow the instructions in the [Quick Start Guide](https://docs.openhwgroup.org/projects/core-v-verif/en/latest/quick_start.html) and the associated READMEs ([link](https://github.com/openhwgroup/core-v-verif/blob/master/mk/README.md) and [link](https://github.com/openhwgroup/core-v-verif/blob/master/mk/TOOLCHAIN.md)) in CORE-V-VERIF and use the latest version of the Embecosm pre-built toolchain.

If there is anything missing (or just plain confusing) about these docs, please do raise an issue against them. At OpenHW we treat a documentation bug the same way we treat an RTL or TB bug.

gullahmed1251, 2023-01-22:
Thank you very much @abukharmeh and @MikeOpenHWGroup for your help. Core is  compiling the RTL and the core testbench using Verilator.