openhwfoundation/core-v-verif #1595: Compressed instructions are not logged properly in trace log

abdullahyildiz, 2023-01-24:
## Bug Title
Compressed instructions are not decoded and logged properly in trace log during RTL simulation.

### Type
Instruction decode error in trace log

### Steps to Reproduce
This situation has been observed in [cv32e40p/release](https://github.com/openhwgroup/core-v-verif/tree/cv32e40p/release) after running the following command:

`make sanity VCS_UVMHOME_ARG=$VCS_HOME/etc/uvm-1.2 CV_SIMULATOR=vcs USE_ISS=NO`

Here is the corresponding trace log and ELF dump respectively for the instruction `c.sub x12, x12, x10` at address `0xa8`:

trace_core_00000000.log:
```
Time	Cycle	PC	Instr	Decoded instruction	Register and memory contents
...
167.000 ns              50 000000a8 40a60633 c.sub            x12, x12, x10       x12=0000003c x12:00010648 x10:0001060c
...
```

hello-world.objdump:
```
...
a8:	8e09                	c.sub	x12,x10
...

```
### Additional context
N/A

MikeOpenHWGroup, 2023-01-24:
Thanks for the issue @abdullahyildiz.  Is this issue gating you?  I have assigned it to myself, but I won't be able to get to it for a while.

abdullahyildiz, 2023-01-25:
@MikeOpenHWGroup,

Thanks. We will also try to figure out the issue in the meantime.

pascalgouedo, 2023-01-25:
Hi,
This is done in the CV32E40P tracer.
I think it was done in purpose and personally I prefer to keep the tracer way of printing the compressed instruction because it expands it, read and written registers are explicitly shown together with their read and written values.
This is especially true for the one which is read and written (x12 in the example).
Tracing the compressed instruction like in objdump would give that

> c.sub            x12, x10       x12=0000003c x12:00010648 x10:0001060c

and could be a bit less understandable.

abdullahyildiz, 2023-01-25:
@pascalgouedo,

Actual problem here is about decoding the instruction word. The instruction word `0x40a60633` in trace log does not represent a `C.SUB` instruction according to RISC-V ISA Unprivileged ISA spec. It should rather be  `0x00008e09`.

pascalgouedo, 2023-01-25:
This is the decompressed 32 bit version of the compressed instruction.
As the tracer is taking the instruction from ID stage way after the decompressor which is in IF stage, it can only report 32-b instruction value.
In summary the tracer reports the fully expanded instruction except the mnemonic which keeps its compressed form to still figure out that the fetched instruction was a compressed one.

MikeOpenHWGroup, 2023-01-25:
This is interesting and something that was not clearly thought through.  The Tracer is not wrong to report the expanded instruction.  However, this has serious implications for functional coverage of the ISA.

@silabs-robin, @AyoubJalali, @eroom1966 what are your thoughts on this?

eroom1966, 2023-01-26:
Sounds wrong to me
All I can really say from our perspective is that the REF model, will display the correct bit pattern and mnemonic

pascalgouedo, 2023-01-26:
This was the "old" tracer that we kept until rvfi tracer works correctly.
At the end ISA coverage will take its info from new RVVI-TRACE in ImperasDV frmework.

murattokez, 2024-11-14:
Is there any update on this issue? I want to use the cv32e40p's tracer and not the RVVI-TRACE solution as I will be comparing the trace to the spike. The problem with the issue is giving me trouble.

MikeOpenHWGroup, 2024-11-14:
Hi @murattokez, my apologies, I should have marked this as "won't fix".  There are two reasons for this:
1. The issue being discussed here relates to code in the cv32e40p repo ([link](https://github.com/openhwgroup/cv32e40p/blob/360d272898d81806be3377193870dbf83a3ea79f/bhv/cv32e40p_tracer.sv#L29)), not this repo.
2. The "core tracer" originally developed for RI5CY has been deprecated in favour of the [RVFI](https://docs.openhwgroup.org/projects/cv32e40x-user-manual/en/latest/rvfi.html#risc-v-formal-interface-rvfi).

As an aside, we are currently working on a Spike-based reference module for use with RVFI.  You can see it [here](https://github.com/openhwgroup/core-v-verif/tree/master/lib/uvm_components/uvmc_rvfi_reference_model).  Note that it is still in active development.