openhwfoundation/core-v-verif #1607: Adopt CV_SW_ variables for toolchain control in CVA6

MikeOpenHWGroup, 2023-02-01:
This issue was motivated by pull-request #1602.

Update the scriptware used by the CVA6 adopt the common set of shell environment variables used by the E40 cores.  These are documented in [core-v-verif/mk/README.md](https://github.com/openhwgroup/core-v-verif/blob/master/mk/README.md#required-corev-environment-variables).

## Motivation
1. The E40 cores use a wide variety of toolchains and eventually the CVA6 will too.  It is important to provide a consistent, unified approach to toolchain control.  This is in keeping with the goals of the "Step 1" project.
2. The "two variable solution (one to point to the installation directory of all toolchains and another to point to the specific toolchain) has proven to be the most flexible and maintainable.
3. All cores verified in CORE-V-VERIF should use a common approach to toolchain selection and control.

I am assigning this issue to @JeanRochCoulon, so he can either kill it (with much protest from me) or assign it to an engineer who will implement it.

JeanRochCoulon, 2023-03-16:
@zchamski has already understand the issue, and maybe know how to solve it when he will be available.

PicoPET, 2023-03-21:
@MikeOpenHWGroup , I was able assign my Thales incarnation to https://github.com/openhwgroup/core-v-verif/issues/1731, but somehow is not possible on this issue (#1607).  Any clues why this may happen?

MikeOpenHWGroup, 2023-03-21:
> @MikeOpenHWGroup , I was able assign my Thales incarnation to #1731, but somehow is not possible on this issue (#1607). Any clues why this may happen?

It appears that @zchamski is not a member of the "core-v-contributors" group, but @PicoPET is.  I just send an invitation to @zchamski to join the group.  Should @PicoPET remain a member?

zchamski, 2023-03-21:
You can remove @PicoPET from the "core-v-contributors" group: the official account for my contributions is the Thales one (@zchamski).  This will simplify the issue assignment process.