openhwfoundation/core-v-verif #1621: CV32E40S simulation problem

weik56100, 2023-02-06:
Hello everyone, I have a problem when I try to run the hello-world program with Verilator on the cv32e40s processor.
[image]

### Steps to Reproduce
`sudo git clone https://github.com/openhwgroup/core-v-verif.git`
`cd /opt/core-v-verif/cv32e40s/sim/core`
`sudo make veri-test TEST=hello-world`

Branch: cv32e40s/
ExternalRepos.mk (default):
```
CV_CORE_REPO   ?= https://github.com/openhwgroup/cv32e40s
CV_CORE_BRANCH ?= master
CV_CORE_HASH   ?= 103056f0deeac8e6cc10244c86bff83d3014f66f
CV_CORE_TAG    ?= none
```

### Additional context
Ubuntu 22.04.1 LTS
Verilator v5.006
LLVM Toolchain (Last)

MikeOpenHWGroup, 2023-02-06:
Hi @weik56100, thanks for your issue.  As of version v4.* Verilator was unable to support CV32E40S, and we do not use Verilator for this core.  It is possible that v5.* of Verilator can support the E40S and we would welcome a pull request to add this capability.

weik56100, 2023-02-06:
Hi @MikeOpenHWGroup, 

What is the procedure to simulate the "hello-world" on the cv32e40s core if Verilator is not on the loop for the moment ?

MikeOpenHWGroup, 2023-02-06:
OpenHW members use one or more of the commerical SystemVerilog simulators.  I believe that most of the Contributors active on the CV32E40S project use Cadence Xcelium, but there is no reason why that core and the CORE-V-VERIF testbench will not work with any other commerical tool.

A bit more background on this topic can be found [here](https://docs.openhwgroup.org/projects/core-v-verif/en/latest/intro.html#a-note-about-eda-tools).

silabs-hfegran, 2024-01-11:
@MikeOpenHWGroup I believe this issue can be closed?

closed 2024-01-11 by silabs-hfegran