openhwfoundation/core-v-verif #1623: Clean-up of Coremark directories for license clarity

christian-herber-nxp, 2023-02-07:
## Task Title
Clean-up of Coremark directories for license clarity

### Task Outcome
Currently, Coremark benchmark is integrated on a per core basis. In addition, the original sources seem to have been modified. In consequence, a Silicon Labs/OpenHW group copyright has been added to the header, in addition to a solderpad license. The header does not explain what the changes are that have been done to the files. 
The issue is twofold:
1. It is not clear what was modified (i.e. what is subject to the respective copyright
2. Expectation would be that the benchmark files are not modified to ensure the integrity of the benchmark

My expectation would be that the coremark files are taken unmodified, straight from the eembc github. Best case, you don't even copy the files but just pull them when building. The respective core directories in the repository should only contain the core_portme{.c/.h} files, of which the .h file should most likely also not be modified.

Provide a brief description of what is expected.  For example, "Implement an
XYZ UVM Agent." If necessary point to a specification document.

### Background information
https://github.com/eembc/coremark

### Location Information
core-v-verif/*core*/tests/programs/custom/coremark/

### Completion Criteria
Clear copyright and license situation. Original Coremark files included.

christian-herber-nxp, 2023-02-14:
@MikeOpenHWGroup could you have a look here?

christian-herber-nxp, 2023-02-17:
i did have a deeper look, and diffed with the eembc upstream repo. It seems that the only changes that OpenHW group did was add an OpenHW group copyright statement and their own license. This is void and should be reverted. The coremark sources seem to correspond to the state of the coremark repo around 2020, with mostly typos fixed since.
So good news is the benchmark had not been tampered with, but still a rather strange thing.

MikeOpenHWGroup, 2023-02-18:
The additional copyright is _not_ void and need not be removed.  It is my understanding that if anyone modifies a permissively  open source file they are obligated to add their own copyright header if they re-publish the files under an open-source license (which is what OpenHW/SiliconLabs is doing here).

So no action is required.

Having said that, if our Coremark files are direct copies of the Embecosm sources, without modification, then we could (should?) remove them from CORE-V-VERIF and clone-as-needed using the same Makefile scheme as is used for the RTL of the core, riscv-dv and Embench.

@silabs-robin, are you aware of any modifications we have made to the Coremark tests for the E40P/S/X?  Is there any reason why we need to keep these in CORE-V-VERIF?

christian-herber-nxp, 2023-02-19:
Mike, if you look at my comment, you will see that NO changes were done in the coremark benchmark files besides adding a copyright notice AND a different than the original license.

silabs-robin, 2023-02-20:
> @silabs-robin, are you aware of any modifications we have made to the Coremark tests for the E40P/S/X?

Some adaptation was necessary, but IIRC it should be contained to some wrapper file or adaptation layer.
I don't think that any tests themselves were modified, but I can't claim that with 100% certainty.

>  Is there any reason why we need to keep these in CORE-V-VERIF?

That depends on which files where changes.
In principle, no, it could be cloned in like the other repos we clone in.
But whatever files are core-specific or cvverif-specific must be handled somehow.

MikeOpenHWGroup, 2023-02-20:
OK, thanks for the input.  I think getting this done is a good idea and a nice-to-have, but not gating anything.  If anyone has a new Contributor to ramp up, this would be a good first task.

@christian-herber-nxp, the License question is a non-issue.  The original Coremark license is Apache2 and OpenHW has added a Solderpad2 license.  This is a valid use-case of the Apache2 license.

christian-herber-nxp, 2023-03-07:
Summary:
- Priority 1: The Silicon Labs and OpenHW group copyrights and solderpad license should be removed from files to which no (or no significant) contribution has been made. This includes in particular the benchmark files, which anyway should not be modified and have not been modified. This should be as easy as copying the latest upstream files over the existing ones.
- Priority 2: coremark should be cleaned up by restructuring in a way that the coremark sources are cloned in make, and only the *port.h/c files are contained here.

PR #1754 (2023-03-31): Remove OpenHW/SiLabs copyright headers from unmodified Coremark files
This PR is the first step to resolve #1623.

It updates all of the Coremark license headers in `core-v-verif/<core>/tests/programs/custom/coremark` to remove the OpenHW/SiLabs copyright from unmodified files.

@christian-herber-nxp, if/when you sign off on this I will make the same changes to the `cv32e20` branch.
PR #1790 (2023-04-11): Remove OpenHW/SiLabs copyright headers from unmodified Coremark files
This PR is the second step to resolve #1623.

It updates all of the Coremark license headers in `core-v-verif/<core>/tests/programs/custom/coremark` to remove the OpenHW/SiLabs copyright from unmodified files.
closed 2023-03-31 by silabs-hfegran