openhwfoundation/core-v-verif #1637: Secureseed0 does not return 0 upon a read

silabs-krdosvik, 2023-02-10:
## Secureseed0 does not return 0 upon a read
**Requirement:**
[image]

**Test:**
[image]

### Type
Indicate whether the type of problem you found:
* Functionally incorrect behavior

### Steps to Reproduce
Please provide:
1. URL to branch that exhibits the issue: https://github.com/silabs-krdosvik/core-v-verif/tree/ISSproblem
(URL using hash: https://github.com/silabs-krdosvik/core-v-verif/tree/3ca6348ec45ba152f3d6fa870ff84979a1121d18)

3. Command line  (e.g. `make xrun-riscv-compliance`):
-	cd cv32e40s/sim/uvmt
- 	run test TEST=xsecure_test cfg=PMP
4. Logfile and/or wave-dump info (screen shots can be useful):
[image]

### Additional context
Add any other context about the problem here:
The core output 0 when reading the secureseed0 CSR, while the ISS does not. According to the spec, read of the secureseed0 CSR should return 0.

@eroom1966

eroom1966, 2023-02-10:
Thanks for the testcase - this looks like an early mistake where the register was defined as RW, but in the description that it always read 0
This later changed to WARL (0x0)
I have a fix I can commit shortly

eroom1966, 2023-02-10:
This issue is fixed in the above PR

eroom1966, 2023-02-13:
@silabs-krdosvik 
Can you please update/close the Issue with respect to the changes made to the modes

closed 2023-02-13 by silabs-krdosvik