openhwfoundation/core-v-verif #1638: ISS output when accessing the secureseed CSRs using the csrr instruction

silabs-krdosvik, 2023-02-10:
## Bug Title
ISS output when accessing the secureseed CSRs using the csrr instruction

**User manual:**
https://docs.openhwgroup.org/projects/cv32e40s-user-manual/en/latest/control_status_registers.html
[image]
[image]
[image]

**Test::**
[image]

### Type
Indicate whether the type of problem you found:

* Functionally incorrect behavior/Confusing or extraneous status or error messages

### Steps to Reproduce
Please provide:
1. URL to branch that exhibits the issue: https://github.com/silabs-krdosvik/core-v-verif/tree/ISSproblem2
(URL using hash: https://github.com/silabs-krdosvik/core-v-verif/tree/b51351b8d093ddcef0d370bdd57862b90c974471)

2. Command line  (e.g. `make xrun-riscv-compliance`):
-	cd cv32e40s/sim/uvmt
-	run test TEST=xsecure_test cfg=PMP

3. Logfile and/or wave-dump info (screen shots can be useful):
[image]

### Additional context
Add any other context about the problem here:
According to the user manual it is only possible to access the secureseed CSRs using the csrrw instruction. Trying to access the CSRs using other instructions should be treated as illegal instructions. The core behaves according to the user manual, while the ISS does not.

@eroom1966

eroom1966, 2023-02-10:
Thanks for the testcase - this looks like an early mistake where the register was defined as RW, but in the description that it always read 0
This later changed to WARL (0x0)
I have a fix I can commit shortly

eroom1966, 2023-02-10:
This issue is fixed in the above PR

eroom1966, 2023-02-13:
@silabs-krdosvik 
Can you please update the Issue with respect to the changes made to the modes

silabs-krdosvik, 2023-02-14:
I have rerun the test with the updated ISS, and core-v-verif and cv32e40s repo. I still see the same problem. Accessing the secureseed0 CSR with the csrr (csrrs) instruction is not treated as an illegal instruction.
[image]

eroom1966, 2023-02-14:
@silabs-krdosvik 

I see the behavior for this changed from 0.7.0 to 0.8.0
https://docs.openhwgroup.org/projects/cv32e40s-user-manual/en/0.7.0/control_status_registers.html#secure-seed-0
https://docs.openhwgroup.org/projects/cv32e40s-user-manual/en/0.8.0/control_status_registers.html#secure-seed-0

As far as I was aware the only changes in 0.8.0 were for the tdata1 WARL behavior as outlined in a previous email from @Silabs-ArjanB 
> I can fix the missing 0xF* resolution value for tdata1.type and then do a new release called 0.8.0 and then we would skip ISS support for 0.8.0, is that okay?

Let me look at this behavior that was also added to 0.8.0

eroom1966, 2023-02-15:
Hi @silabs-krdosvik, can you reassign this issue to me please until resolved

Silabs-ArjanB, 2023-02-15:
@eroom1966 

> As far as I was aware the only changes in 0.8.0 were for the tdata1 WARL behavior as outlined in a previous email from @Silabs-ArjanB

More changes than just tdata1 and secureseed* were done as documented in https://docs.openhwgroup.org/projects/cv32e40s-user-manual/en/latest/preface.html#release-0-8-0.

eroom1966, 2023-02-17:
Fixed by
https://github.com/openhwgroup/core-v-verif/pull/1659
And ImperasDV-OpenHW hash 490e1c0

The test is reported as FAILED, but there are no mismatches - so I am not sure what this means

silabs-krdosvik, 2023-02-17:
> Fixed by #1659 And ImperasDV-OpenHW hash 490e1c0
> 
> The test is reported as FAILED, but there are no mismatches - so I am not sure what this means

Thanks Lee! 

About the teste failing, I suspect that the fail is related to the csrr instruction becoming an illegal instruction.
I believe the csrr instruction will not set the rd variable to 0 (because it is an illegal instruction), and the following check (assert_or_die), make the test fail because rd != 0.
[image]

If this is the case, I can fix it myself 
If this is not the case, and it seems there is another ISS problem, I will let you know. :-)

eroom1966, 2023-03-07:
@silabs-krdosvik any update on this issue - I think it is resolved ?

silabs-krdosvik, 2023-03-08:
> @silabs-krdosvik any update on this issue - I think it is resolved ?

Aa, sorry, forgot to close it. But yess, it is resolved :-)

closed 2023-03-08 by silabs-krdosvik