openhwfoundation/core-v-verif #1648: Make system doesn't update the git branch/commit/tag when rerunning with a different value

christian-herber-nxp, 2023-02-14:
## Improve makefile to update the git repository on incremental builds

### Task Outcome
Make system that properly updates the git repository in incremental runs

### Completion Criteria
in a build using make, the git repository should be updated if it has been cloned before, if
- there is new commits remote
- an environment variable changed to use different branch/commit/tag

frankly, the safest way to do this will likely be to remove and then clone again. this would avoid issues where the remote url changed to a fork but the repo name is the same. shallow clone could be done for performance improvement.

silabs-hfegran, 2023-02-15:
I see pros and cons on both the current and proposed approach - the proposed methodology looses some flexibility if you actually do modifications in-place on cloned repositories (quick tests, riscv-dv development work etc.).  In that case you would have to ensure that the cloned, modified repos are not overwritten; personally I prefer the solution that does not potentially erase development work when one accidentally forgets some flag, rather than the the one that always cleans out everything unless explicitly told otherwise.

pascalgouedo, 2023-02-15:
Hi all,
If I need to make some experiments on Core RTL, I am cloning manually elsewhere than inside core-v-verif/core-v-cores and use CV_CORE_PATH which creates a link in core-v-cores to your updated clone.
No fear that it would be destroyed unintentionally (only the link).
But I agree this mechanism is only available for RTL and not on any other clone core-v-verif is making locally.
To have clean_all destroying all clones would require first to apply same "linking" methodology to all required clones.