openhwfoundation/core-v-verif #1649: ISS - IRQ Behavior Can't Converge

silabs-robin, 2023-02-15:
### Type

* Functionally incorrect behavior

### Steps to Reproduce

1. Branch: `umode_directeds_reglist` - https://github.com/silabs-robin/core-v-verif/tree/umode_directeds_reglist
2. Commit: `e84b41`
3. Command: `make test TEST=interrupt_priv_test CFG=pmp`
4. Log and wave shown below

```
Error (IDV) The reference model is unable to find a legal set of actions to converge to the provided DUT state.
Info (IDV) Pending net changes:
Info (IDV)  - Group 1:
Info (IDV)  -   MExternalInterrupt => 1,  age = 14 cycles, 2 retires
Info (IDV)  -   MExternalInterrupt => 0,  age = 11 cycles, 0 retires
Info (IDV) DUT state change (hartId: 0):
Info (IDV)  - dut.pc               - 0x0000002c vector_table+44
Info (IDV)  - dut.x0               - 0x00000000 
Info (IDV)  - dut.mstatus          - 0x00000000 SD:0 TW:0 MPRV:0 XS:0(Off) FS:0(Off) MPP:0 VS:0(Off) MPIE:0 UBE:0 MIE:0
Info (IDV)  - dut.mepc             - 0x00000660 
Info (IDV)  - dut.mcause           - 0x8000000b Interrupt:1 Code:11(Machine external interrupt)
Info (IDV) Reference state change evaluation tree (HartId: 0):
Info (IDV) No state change found for reference to converge to DUT state.
Warning (IDV) The reference is being forced to converge to the DUT state, results that follow should be treated with caution.
UVM_ERROR @ 4462.300 ns : rvvi-pkg.sv(94) reporter [] uvmt_cv32e40s_tb.imperas_dv.trace2api.state_compare @ 4462.000 ns: MISMATCH
```

[image]

### Additional context

From what I can see, 1) `Machine External Interrupt (MEI) irq_i[11]` is incoming, 2) both the core and ISS sees this, 3) the core correctly takes the MEI.

@eroom1966, could you please have a look and see if you understand what is going wrong?

eroom1966, 2023-02-16:
Hi @silabs-robin , 
I am still investigating, but in the meantime, can you have a look at the value of mstatus.mie, as far as I can tell this is 0, 
hence interrupts are disabled, so I cannot see how this interrupt can be taken ?

eroom1966, 2023-02-16:
I have this fixed
It was a mistake in not passing the interrupts when not in privilege levels below machine mode in the model wrapper code.
I am trying to get a resolution on #1650 before issuing a new release

eroom1966, 2023-02-17:
Fixed by
https://github.com/openhwgroup/core-v-verif/pull/1659
And ImperasDV-OpenHW hash 490e1c0

silabs-robin, 2023-02-19:
> Fixed by #1659 And ImperasDV-OpenHW hash 490e1c0

Test status changed from failing to passing.
Closing this issue as fixed.
Thanks Lee.

closed 2023-02-19 by silabs-robin