openhwfoundation/core-v-verif #1652: Fetch origin/main next to specified branch/commit

szbieg, 2023-02-15:
A user can select a specific branch/commit with CV_* variables. It would be good, and not taking too many resources, to also add the origin/main branch for the reference/debugging/rebase. This could also be done depending on the environment variable setting.
Effectively after:
```git clone -b $(CV_CORE_BRANCH) --single-branch $(CV_CORE_REPO) $(CV_CORE_PKG)```
Perform also:
```git remote set-branches --add origin main```
```git fetch origin```

MikeOpenHWGroup, 2023-02-18:
Hi @szbieg, this is a clever idea.  Some things to consider:
1. The repos that are cloned by the CORE-V-VERIF Makefiles are not intended to be used to submit updates to their origins.
2. Your idea assumes a high level of "git sophistication".
3. There is nothing stopping sophisticated git users from doing this manually.

What use case do you have in mind for this?

szbieg, 2023-03-06:
Thank you for your feedback. Could you give more insight regarding the reason what makes the "makefile-cloned-repository" not suitable for that purpose?
My use case was an automation of workspace preparation. Makefile is intended to automate processes, hence my suggestion to incorporate it conditionally when the selected (cloned) branch cloned is a not a master (main).

MikeOpenHWGroup, 2023-03-06:
> My use case was an automation of workspace preparation.

It is not clear what that means.  When the Makefiles clone the RTL, riscvdv and svlib repositories it is assumed that the user of CORE-V-VERIF will _use_ these resources, but not _change_ them.   That is, the code from those repos is cloned, compiled and forgotten.  The Makefiles already automate that process.

What else is there to do?