openhwfoundation/core-v-verif #1653: ISS: bit[0] of dpc should be hard-coded zero

silabs-hfegran, 2023-02-15:
## bit[0] of dpc should follow the same IALIGN restrictions as mepc
### Context
[image]

This bit is erroneously described as rw in the user manual (presumably the reason for the current reference model implementation), so there is a corresponding issue for updates to the user manual here: https://github.com/openhwgroup/cv32e40s/issues/407

RTL correctly implements this bit as hard-coded zero, and the user manual and ISS needs to be updated accordingly.

silabs-hfegran, 2023-02-20:
Thanks for the update @eroom1966, I will leave this open until I have verified the fix.

eroom1966, 2023-03-07:
@silabs-hfegran any update on this issue, I am pretty sure it is resolved

silabs-hfegran, 2023-03-07:
No update yet, pending checks after PR #1702 gets merged - will update/close this issue accordingly

silabs-hfegran, 2023-03-13:
This has been resolved, Thanks!

closed 2023-03-13 by silabs-hfegran