openhwfoundation/core-v-verif #1657: ISS - PMP/PMA Mismatch on Push Instruction

silabs-robin, 2023-02-16:
### Type

* Functionally incorrect behavior

### Steps to Reproduce

1. Branch: `pmadirectedfix`, https://github.com/silabs-robin/core-v-verif/tree/pmadirectedfix
1. Hash: `414984`
1. Compiler: `corev-openhw-gcc-centos7-20221213`
1. Command: `make test TEST=pma_debug CFG=pma_debug`

```
Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x000000ce b862     cm.push {x1,x8-x9},-16: No access permission for store/atomic memory
 operation (0x3ffffc)
Error (IDV) PC mismatch (HartId:0, PC:0x00000000 vector_table+0):
Info (IDV)  0>
Info (IDV)   . dut:0x000000ce main+2
Info (IDV)   . ref:0x00000000 vector_table+0
Error (IDV) Insn. bit pattern mismatch (HartId:0, PC:0x00000000 vector_table+0):
Info (IDV)  1>
Info (IDV)   . dut:b862     cm.push {x1,x8-x9},-16
Info (IDV)   . ref:16e0006f j       16e
Error (IDV) GPR register value mismatch (HartId:0, PC:0x00000000 vector_table+0):
Info (IDV)  2> GPR x2
Info (IDV)   . dut:0x003ffff0
Info (IDV)   . ref:0x00400000
Error (IDV) CSR register value mismatch (HartId:0, PC:0x00000000 vector_table+0):
Info (IDV)  3> CSR 341 (mepc)
Info (IDV)   . dut:0x00000000
Info (IDV)   . ref:0x000000ce
Info (IDV)  4> CSR 342 (mcause)
Info (IDV)   . dut:0x00000000 Interrupt:0 Code:0(Instruction address misaligned)
Info (IDV)   . ref:0x00000007 Interrupt:0 Code:7(Store/AMO access fault)
UVM_ERROR @ 3328.300 ns : rvvi-pkg.sv(94) reporter [] uvmt_cv32e40s_tb.imperas_dv.trace2api.state_compare @ 3328.000 ns: MISMA
TCH
```

From "showoverrides", it seems the RTL and ISS has the same PMA config, and the push instruction does not seem to be accessing a restricted region.
The default PMP config should have clearance to access anything in M-mode.
So the reason for the mismatch is not clear.

### Additional context

This was not produced with the very latest gcc available, so it is likely that this case is provoked by a botched generation of push/pop
But nevertheless, the ISS and RTL is in disagreement.

silabs-robin, 2023-02-16:
@eroom1966 could you please have a look?

eroom1966, 2023-03-07:
@silabs-robin 
can you please double check this, I did the following

1. took the latest ImperasDV
git clone https://github.com/Imperas/ImperasDV-OpenHW.git

2. took your branch
git clone https://github.com/silabs-robin/core-v-verif -b pmadirectedfix

I had to modify cv32e40s/tb/uvmt/imperas_dv.flist to match the latest cv32e40s/dev
```
+incdir+${IMPERAS_HOME}/ImpProprietary/include/host
-f ${IMPERAS_HOME}/ImpPublic/source/host/rvvi/rvvi.f
-f ${IMPERAS_HOME}/ImpProprietary/source/host/rvvi/imperasDV.f

// ImperasDV test bench wrapper
${TBSRC_HOME}/uvmt/uvmt_cv32e40s_imperas_dv_wrap.sv
```

When I run your test
```
make clean_all
make test TEST=pma_debug CFG=pma_debug
```
I get 
```
hello pma_debug

requested debug mode
Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x1a1107f0 0000     c.illegal: No access permission for fetch (0x1a1107f0)
single-step exception handled
Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x1a1107f4 0000     c.illegal: No access permission for fetch (0x1a1107f4)
dm_exception handled
goodbye pma_debug
```
...
```
Info (IDV) ---------------------------------------------------
Info (IDV) ImperasDV VERIFICATION REPORT
Info (IDV)   Instruction retires   : 120,870
Info (IDV)   Traps                 : 2
Info (IDV)   Interrupt events      : 0
Info (IDV)   Ending cycle count    : 351,136
Info (IDV)                               Sets / Compares
Info (IDV)     PC                  :  120,872 / 120,870
Info (IDV)     Instruction         :  120,872 / 120,870
Info (IDV)     GPR                 :   81,875 / 81,875
Info (IDV)     CSR                 :    2,153 / 4,150
Info (IDV)     FPR                 :        0 / 0
Info (IDV)     VR                  :        0 / 0
Info (IDV)  
Info (IDV)   Total compares        : 327,765
Info (IDV)   Mismatches            : 0
Info (IDV) ---------------------------------------------------

```

silabs-robin, 2023-03-07:
Hi Lee. Thank you. Yes, I plan on testing that, but the last couple of times I tried we have had other issues (gcc, iss, cvverif, etc) that have blocked me, so I am waiting for those to be entirely fixed before I try again. (Which is probably soon, because we have seen some PRs to fix those things, and we have a big PR up right now that fixed a lot more.)

silabs-robin, 2023-03-09:
This issue has now been resolved, I can confirm. Thanks for the fix Lee.

closed 2023-03-09 by silabs-robin