openhwfoundation/core-v-verif #1658: ISS – Off-by-One Haltreq Debug Entry

silabs-robin, 2023-02-16:
### Type

* Functionally incorrect behavior

### Steps to Reproduce

1. Branch: `pmadirectedfix`, https://github.com/silabs-robin/core-v-verif/tree/pmadirectedfix
1. Hash: `414984`
1. Compiler: `corev-openhw-gcc-centos7-20221213`
1. Command: `make test TEST=pma_debug CFG=pma_debug`
1. **Edit:** Remove everything `zc` from `cv32e40s/tests/cfg/pma_debug.yaml`  (except "zca")

The first mismatch is on PC, when the core has entered debug (per haltreq requested by the test program), but the ISS is not executing debug code yet.
On the subsequent PC mismatch errors, the ISS and core is off by one.

```
Error (IDV) PC mismatch (HartId:0, PC:0x000000ec main+32):
Info (IDV)  0>
Info (IDV)   . dut:0x1a110800 debugger_entry+0
Info (IDV)   . ref:0x000000ec main+32
Error (IDV) Insn. bit pattern mismatch (HartId:0, PC:0x000000ec main+32):
Info (IDV)  1>
Info (IDV)   . dut:7b211073 csrw    dscratch0,x2
Info (IDV)   . ref:49050513 addi    x10,x10,1168
Error (IDV) CSR register value mismatch (HartId:0, PC:0x000000ec main+32):
Info (IDV)  2> CSR 7b1 (dpc)
Info (IDV)   . dut:0x000000f0
Info (IDV)   . ref:0x000000ec
UVM_ERROR @ 15262.300 ns : rvvi-pkg.sv(94) reporter [] uvmt_cv32e40s_tb.imperas_dv.trace2api.state_compare @ 15262.000 ns: MIS
MATCH
Error (IDV) PC mismatch (HartId:0, PC:0x1a110800 debugger_entry+0):
Info (IDV)  3>
Info (IDV)   . dut:0x1a110804 debugger_entry+4
Info (IDV)   . ref:0x1a110800 debugger_entry+0
Error (IDV) Insn. bit pattern mismatch (HartId:0, PC:0x1a110800 debugger_entry+0):
Info (IDV)  4>
Info (IDV)   . dut:00001117 auipc   x2,0x1
Info (IDV)   . ref:7b211073 csrw    dscratch0,x2
Error (IDV) GPR register value mismatch (HartId:0, PC:0x1a110800 debugger_entry+0):
Info (IDV)  5> GPR x2
Info (IDV)   . dut:0x1a111804
Info (IDV)   . ref:0x003ffff0
UVM_ERROR @ 15271.300 ns : rvvi-pkg.sv(94) reporter [] uvmt_cv32e40s_tb.imperas_dv.trace2api.state_compare @ 15271.000 ns: MIS
MATCH
```

### Additional context

This is the same test case as https://github.com/openhwgroup/core-v-verif/issues/1657 , but without Zc instructions it will progress further and hit another problem.

silabs-robin, 2023-02-16:
@eroom1966 could you have a look?

eroom1966, 2023-02-17:
@silabs-robin this is a duplicate issue to https://github.com/openhwgroup/core-v-verif/issues/1649
this instruction is actually retiring, not taking an exception, the exception occurs on the following instruction

eroom1966, 2023-02-17:
Fixed by
https://github.com/openhwgroup/core-v-verif/pull/1659
And ImperasDV-OpenHW hash 490e1c0

silabs-robin, 2023-02-19:
> Fixed by #1659 And ImperasDV-OpenHW hash 490e1c0

Hi. The test still fails, so it does not appear to be fixed.

I did `git clean -xdff . && git checkout .` before following the "Steps to Reproduce" above, twice (to be sure), and I got the same error message as in the original report.

Does this test pass for you when following the above instructions?
Are you using the same compiler version?
@eroom1966

eroom1966, 2023-02-21:
@silabs-robin 
Did you merge in the code from #1659 ?
I merged locally (does not appear to be on your branch pmadirectedfix) and now I see a different error
investigating now

eroom1966, 2023-02-21:
@silabs-robin 

This passes for me
This is what I did

1. Checked out your branch
2. merged the change from #1659 (correct detection of retired instruction)
3. removed all refs to zc in cv32e40s/tests/cfg/pma_debug.yaml (except zca)

it now runs to completion

```
hello pma_debug

requested debug mode
Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x1a1107f0 0000     c.illegal: No access permission for fetch (0x1a1107f0)
single-step exception handled
Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x1a1107f4 0000     c.illegal: No access permission for fetch (0x1a1107f4)
dm_exception handled
goodbye pma_debug

UVM_INFO @ 1054119.300 ns : uvmt_cv32e40s_firmware_test.sv(152) uvm_test_top [TEST] Finished RUN: exit status is 0
UVM_INFO @ 1054119.300 ns : uvm_objection.svh(1271) reporter [TEST_DONE] 'run' phase is ready to proceed to the 'extract' phase
UVM_INFO @ 1054119.300 ns : uvme_cv32e40s_buserr_sb.sv(230) uvm_test_top.env.buserr_sb [BUSERRSB] received 0 D-side 'err' transactions
UVM_INFO @ 1054119.300 ns : uvme_cv32e40s_buserr_sb.sv(231) uvm_test_top.env.buserr_sb [BUSERRSB] received 0 D-side 'first err' transactions
UVM_INFO @ 1054119.300 ns : uvme_cv32e40s_buserr_sb.sv(232) uvm_test_top.env.buserr_sb [BUSERRSB] observed 0 rvfi nmi handler entries
UVM_INFO @ 1054119.300 ns : uvme_cv32e40s_buserr_sb.sv(233) uvm_test_top.env.buserr_sb [BUSERRSB] received 0 I-side 'err' transactions
UVM_INFO @ 1054119.300 ns : uvme_cv32e40s_buserr_sb.sv(234) uvm_test_top.env.buserr_sb [BUSERRSB] retired 0 expectedly ifault instructions
UVM_INFO @ 1054119.300 ns : uvme_cv32e40s_buserr_sb.sv(235) uvm_test_top.env.buserr_sb [BUSERRSB] retired 0 expectedly ifault instructions during debug
UVM_INFO @ 1054119.300 ns : uvme_cv32e40s_buserr_sb.sv(236) uvm_test_top.env.buserr_sb [BUSERRSB] observed 0 rvfi ifault handler entries
UVM_INFO @ 1054119.300 ns : uvma_pma_sb.sv(226) uvm_test_top.env.pma_agent.scoreboard [PMASB] checked 143041 OBI I transactions
UVM_INFO @ 1054119.300 ns : uvma_pma_sb.sv(227) uvm_test_top.env.pma_agent.scoreboard [PMASB] checked 82680 OBI D transactions
UVM_INFO @ 1054119.300 ns : uvmt_cv32e40s_base_test.sv(345) uvm_test_top [END_OF_TEST] DUT WRAPPER virtual peripheral signaled exit_value=0.
UVM_INFO @ 1054119.300 ns : uvm_report_catcher.svh(705) reporter [UVM/REPORT/CATCHER] 
--- UVM Report catcher Summary ---

Number of demoted UVM_FATAL reports  :    0
Number of demoted UVM_ERROR reports  :    0
Number of demoted UVM_WARNING reports:    0
Number of caught UVM_FATAL reports   :    0
Number of caught UVM_ERROR reports   :    0
Number of caught UVM_WARNING reports :    0

UVM_INFO @ 1054119.300 ns : uvm_report_server.svh(847) reporter [UVM/REPORT/SERVER] 
--- UVM Report Summary ---

Quit count :     0 of     5
** Report counts by severity
UVM_INFO :   82
UVM_WARNING :    0
UVM_ERROR :    0
UVM_FATAL :    0
** Report counts by id
[BASE TEST]     5
[BUSERRSB]     7
[CFG]     1
[CLKNRST]     3
[CV32E40SCORECTRLAGT]     1
[DEBUGCOVG]     1
[END_OF_TEST]     1
[FENCEI_MON]     2
[FETCHTOGGLE]     1
[INTERRUPTCOVG]     1
[ImperasDV_wrap]     3
[MEMREADMEMH]     1
[OBIVPSEQ]    37
[OBI_MEMORY_MON]     4
[PMASB]     2
[RNTST]     1
[RST_VSEQ]     3
[TEST]     4
[TEST_CFG]     1
[TEST_DONE]     1
[UVM/RELNOTES]     1
[UVM/REPORT/CATCHER]     1

xmsim: *W,SLFINV: Call to process::self() from invalid process; returning null.
Info (IDV) ---------------------------------------------------
Info (IDV) ImperasDV VERIFICATION REPORT
Info (IDV)   Instruction retires   : 129,449
Info (IDV)   Traps                 : 2
Info (IDV)   Interrupt events      : 0
Info (IDV)   Ending cycle count    : 351,367
Info (IDV)                               Sets / Compares
Info (IDV)     PC                  :  129,451 / 129,449
Info (IDV)     Instruction         :  129,451 / 129,449
Info (IDV)     GPR                 :   82,063 / 82,063
Info (IDV)     CSR                 :    2,158 / 6,237
Info (IDV)     FPR                 :        0 / 0
Info (IDV)     VR                  :        0 / 0
Info (IDV)  
Info (IDV)   Total compares        : 347,198
Info (IDV)   Mismatches            : 0
Info (IDV) ---------------------------------------------------
Info 
Info ---------------------------------------------------
Info CPU 'refRoot/cpu' STATISTICS
Info   Type                  : CVE4S (CV32E40S_DEV-B)
Info   Final program counter : 0x25e
Info   Simulated instructions: 129,449
Info ---------------------------------------------------
Info 
Info ---------------------------------------------------
Info SIMULATION TIME STATISTICS
Info   Simulated time        : 0.00 seconds
Info   User time             : 101.03 seconds
Info   System time           : 0.20 seconds
Info   Elapsed time          : 101.99 seconds
Info ---------------------------------------------------

ImperasDVasync finished: Tue Feb 21 09:13:53 2023

ImperasDVasync (64-Bit) v20230217.0 Open Virtual Platform simulator from www.IMPERAS.com.
Visit www.IMPERAS.com for multicore debug, verification and analysis solutions.

Error (CF_UE) '--override cpu/Zcmb=1' did not match anything in the platform
    PPPPPPP    AAAAAA    SSSSSS    SSSSSS   EEEEEEEE  DDDDDDD     
    PP    PP  AA    AA  SS    SS  SS    SS  EE        DD    DD    
    PP    PP  AA    AA  SS        SS        EE        DD    DD    
    PPPPPPP   AAAAAAAA   SSSSSS    SSSSSS   EEEEE     DD    DD    
    PP        AA    AA        SS        SS  EE        DD    DD    
    PP        AA    AA  SS    SS  SS    SS  EE        DD    DD    
    PP        AA    AA   SSSSSS    SSSSSS   EEEEEEEE  DDDDDDD     
    ----------------------------------------------------------
                        SIMULATION PASSED                     
    ----------------------------------------------------------
Simulation complete via $finish(1) at time 1054119300 PS + 71
/home/nda/cadence/installs/XCELIUM2107/tools/methodology/UVM/CDNS-1.2-ML/sv/src/base/uvm_root.svh:551     $finish;
xcelium> exit

```

can you please post the differences you see

silabs-robin, 2023-02-23:
Thanks. I misunderstood. Will give it a re-try in a moment.

silabs-robin, 2023-02-27:
> This passes for me This is what I did
> 
>     1. Checked out your branch
> 
>     2. merged the change from [fix handling of instructions reporting that they have retired #1659](https://github.com/openhwgroup/core-v-verif/pull/1659) (correct detection of retired instruction)
> 
>     3. removed all refs to zc in cv32e40s/tests/cfg/pma_debug.yaml (except zca)

Hi. Since the PR you refer to is merged, I merged in cv32e40s/dev.
Now there is a new problem, but we can resolve that in a different ticket and I'll come back again to this after that is fixed.

Including a short snippet of the output just for reference:
```
UVM_INFO @ 114.300 ns : uvmt_cv32e40s_firmware_test.sv(143) uvm_test_top [TEST] Started RUN
Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x000000ce b862     cm.push {x1,x8-x9},-16: No access permission for store/atomic memory
operation (0x3ffffc)
Error (IDV) PC mismatch (HartId:0, PC:0x00000000 vector_table+0):
Info (IDV)  0>
Info (IDV)   . dut:0x000000ce main+2
Info (IDV)   . ref:0x00000000 vector_table+0
```

eroom1966, 2023-02-27:
@silabs-robin - please tell me how to reproduce and I can investigate
I am still messed up due to the compiler issues - I have lots of failures @silabs-hfegran reproduced, but I have not seen a solution

silabs-hfegran, 2023-02-27:
> UVM_INFO @ 114.300 ns : uvmt_cv32e40s_firmware_test.sv(143) uvm_test_top [TEST] Started RUN
Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x000000ce b862     cm.push {x1,x8-x9},-16: No access permission for store/atomic memory
operation (0x3ffffc)
Error (IDV) PC mismatch (HartId:0, PC:0x00000000 vector_table+0):
Info (IDV)  0>
Info (IDV)   . dut:0x000000ce main+2
Info (IDV)   . ref:0x00000000 vector_table+0

This is due to an issue with the ISS that I was investigating earlier today. 
Somehow the ISS has started to take the pma.atomic bit into account, AND treats cm.push as atomic instructions (which is not how I would interpret the spec, they are not contained in the AMO or LR/SC instructions as far as I know) 

If you set the pma atomic override to the ISS to 1 it will pass the cm.push point of failure. (Was some work on atomic on X enabled lately that potentially caused this issue?)

Silabs-ArjanB, 2023-02-27:
> Somehow the ISS has started to take the pma.atomic bit into account, AND treats cm.push as atomic instructions (which is not how I would interpret the spec, they are not contained in the AMO or LR/SC instructions as far as I know)

pma.atomic should indeed not have ay impact on cm.push (but only on instructions from the A extension).

eroom1966, 2023-02-27:
I am sure this issue does not exist in my local copy, but I have held off committing due to the issue with the compiler
Can I get some instructions on how to reproduce this issue.

eroom1966, 2023-03-07:
@silabs-robin do you think there is still an issue here, I am pretty sure this is resolved

silabs-robin, 2023-03-09:
This issue has also been resolved. Thanks.

closed 2023-03-09 by silabs-robin