openhwfoundation/core-v-verif #1663: ISS: mseccfg writability with deconfigured PMP

silabs-hfegran, 2023-02-20:
User manual specifies the following: 
[image]

CV32E40S cannot disable PMP, the existence of the PMP regions is unconditional, but their writability and legal values are controlled by the number of pmp-regions parameter in the core. Similarly, there needs to be a way to control the writability mask of mseccfg, as specified in the user manual. To my knowledge this register is currently always writeable as long as smepmp is enabled in the ISS, causing mismatches if one attempts to write to this register if there are no configured pmp regions. (In this case, the ISS is configured with 64 regions, but sets all pmpaddr and pmpcfg to always-zero)

eroom1966, 2023-02-20:
Hi @silabs-hfegran 
Can you provide a testcase showing the issue.
To our knowledge, writability of mseccfg is already constrained as in the user manual.

silabs-hfegran, 2023-05-12:
Duplicate of #1874

closed 2023-05-12 by silabs-hfegran