openhwfoundation/core-v-verif #1664: Debug vplan mistake

silabs-krdosvik, 2023-02-20:
Please use this file as a template for filing issues against any of the
code in this repository.

## Bug Title
Verification point in the vplan is wrong

### Type
Indicate whether the type of problem you found:
* Other: vplan outpdated/wrong

### Steps to Reproduce
Please provide:
1. URL to branch that exhibits the issue: https://github.com/openhwgroup/core-v-verif/tree/c57012a2d911a52b2b6c275b0e78b519f0ff020e/cv32e40s/docs/VerifPlans/Simulation/debug-trace

2. Logfile and/or wave-dump info (screen shots can be useful):
vplan mistake:
[image]

I cant find anything in the spec or usermanual that says there should be a trap when writing to tdata when not in debug mode.

I guess, it can be interpreted las described over considering that The DMODE field in the tdata1 register is set to 1, and we can therefore only write to tdata1 in debug mode.
[image]

But the user manual also state that the tdata registers are MRW:
[image]

MikeOpenHWGroup, 2023-02-20:
Good catch @silabs-krdosvik!  @pascalgouedo, would this also apply to the cv32e40p?

silabs-robin, 2023-12-11:
Reportedly fixed.

closed 2023-12-11 by silabs-robin