openhwfoundation/core-v-verif #1666: Environment config : Missing constraint in env_cfg

AyoubJalali, 2023-02-20:
Hello,
While re-using the core control agent, I notice that there's only one constraint for the interface signals in the env cfg, the constraint is simple if there's no **+plusargs** in the cmd line of **vcs** give a default value.

I thing the **boot_addr** should be half-word aligned, so we should add a constraint for the boot_addr to keep it in the green values.
What do you think about that ?

AyoubJalali, 2023-02-20:
We can add boot_addr constraint in the core control cfg file, should be enough I think.

AyoubJalali, 2023-02-21:
In which branch should I create the PR .

AyoubJalali, 2023-02-22:
@MikeOpenHWGroup 
I have a question about this, what is the C extension isn't enabled so the boot_addr should be word aligned

MikeOpenHWGroup, 2023-02-22:
The CV32E4 cores do not have this restriction, so this constraint was not needed for them.

AyoubJalali, 2023-02-23:
Also CVA6 doesn't have this restriction, the idea here is to have a generic constraint like this :
    _**constraint boot_addr_cons { 
      if (ext_c_supported) boot_addr % 2 == 0; //boot addr should be half-word aligned if C extension is enabled
      else                 boot_addr % 4 == 0; //boot addr should be word aligned if C extension isn't enabled
   }**_

AyoubJalali, 2023-12-11:
There a PR to fix this issue

closed 2023-12-11 by AyoubJalali