openhwfoundation/core-v-verif #1672: cv32e40s/dev cannot compile applications

eroom1966, 2023-02-22:
I have updated the branch cv32e40s/dev, and can no longer compile applications
There is an incompatibility between the compiler and the testcases/compiler flags

my updated branch is at the following hash
```
commit b3057428ca247c9a4d0dcac29701bc83f9da5aef (HEAD -> cv32e40s/dev, origin/cv32e40s/dev)
Merge: 7928c947 1b919372
Author: Henrik Fegran <[henrik.fegran@silabs.com](mailto:henrik.fegran@silabs.com)>
Date:   Fri Feb 17 14:08:24 2023 +0100
```

I am running the following ubuntu18 compiler
```
riscv32-corev-elf-gcc ('corev-openhw-gcc-ubuntu1804-20230112') 12.0.1 20220422 (experimental)
Copyright (C) 2022 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Seeing error messages such as
```
/home/moore/riscv/riscv/bin/riscv32-corev-elf-gcc \
\
-Os -g -static -mabi=ilp32 -march=rv32im_zba1p00_zbb1p00_zbc1p00_zbs1p00_zicsr_zca_zcb_zcmp_zifencei -Wall -pedantic  \
\
-I ../../tests/asm \
-I /home/moore/tmp/ttt/cv32e40s_dev/cv32e40s/bsp \
-o /home/moore/tmp/ttt/cv32e40s_dev/cv32e40s/sim/uvmt/xrun_results/default/interrupt_test/0/test_program/interrupt_test.elf \
-nostartfiles \
/home/moore/tmp/ttt/cv32e40s_dev/cv32e40s/tests/programs/custom/interrupt_test/interrupt_test.c \
-T /home/moore/tmp/ttt/cv32e40s_dev/cv32e40s/bsp/link.ld \
-L /home/moore/tmp/ttt/cv32e40s_dev/cv32e40s/sim/uvmt/xrun_results/default/interrupt_test/0/test_program/bsp \
-lcv-verif
/tmp/ccGlIRox.s: Assembler messages:
/tmp/ccGlIRox.s:572: Error: illegal operands `cm.push {ra},-64'
/tmp/ccGlIRox.s:612: Error: illegal operands `cm.pop {ra},64'
/tmp/ccGlIRox.s:658: Error: illegal operands `cm.push {ra},-64'
/tmp/ccGlIRox.s:698: Error: illegal operands `cm.pop {ra},64'
/tmp/ccGlIRox.s:744: Error: illegal operands `cm.push {ra},-64'
```
 
I am blocked on progressing a number of issues until this is resolved

silabs-hfegran, 2023-02-22:
reproduced - investigating

silabs-hfegran, 2023-02-22:
For this test, the issue is .option norvc in inline assembly without option push/pop around the respective sections. 

@eroom1966 I have a fix for this test, do you know of any other tests failing with the same error messages?

silabs-hfegran, 2023-02-22:
The random tests will fail with the zcmp at the moment, as they require the zc-support; I will test to see if the Zcmt-issues I was previously seeing with the compiler have been resolved, and hopefully get the necessary support in later today.

eroom1966, 2023-02-22:
I just ran the ci_check script and only 4 tests compiled

eroom1966, 2023-02-23:
@silabs-hfegran
Hi Henrik, any update on this ?
I am wondering how you are all running any testcases - have you modified the yaml files locally ?

silabs-hfegran, 2023-02-27:
> @silabs-hfegran Hi Henrik, any update on this ? I am wondering how you are all running any testcases - have you modified the yaml files locally ?

Apologies for the late reply, I have been out of office for a few days.
I will try to explain the situation; what has happened here is that RTL enabled support for Zc*-extensions, and verification followed suit by implementing new directed tests (and rewritten old ones) to take advantage of this. This in turn means that we need (in particular Zcmp) support from the compiler and thus enabled this across the board. This has unveiled quite a large amount of problems that I am currently trying to resolve:

- Random illegal instruction tests needs defined/enabled Zc* instructions in riscv-dv; this is done locally, awaiting compiler fixes to merge upstream.
- Without a working compiler we cannot merge in the above, as we would not be able to compile the generated source code
- Directed tests depend on support from extensions above, so disabling the extensions will break new tests.
- I have reported issues with the latest toolchain, currently Zcmt is causing headaches, as there are issues with evaluating the benefit of jump tables, causing table jump instructions to be generated without their corresponding tables etc. 
- Binutils version that was released last week is having issues with correct/wrong immediate values for table jump instructions. 

In short, we are stuck in a loop of circular dependencies at the moment, so there is no quick fix. I am working on a fix to get the environment back up asap, but unfortunately it is going to take a little while. Hopefully I can get this back up within a couple of days, but I cannot promise anything.

silabs-hfegran, 2023-02-27:
@eroom1966: https://github.com/openhwgroup/core-v-verif/pull/1685  
Please try out this branch and see if that resolves your compilation issues. This is a draft PR and I will most likely rebase and cleanup this before I make a proper PR, but hopefully this can unblock your work for now.

eroom1966, 2023-02-27:
@silabs-hfegran this also has a dependency on my PR for updating the Manifest file here https://github.com/openhwgroup/core-v-verif/pull/1676
once I had merged those changes in I was able to run ci_check

I am no longer seeing compilation errors
I also do not see any mismatches between the RTL and the REF model

However, a number of tests are reported as failing - but not mismatching.

```
CI Check results:
                 xrun-interrupt_test.log :  FAILED
                     xrun-debug_test.log :  FAILED
                    xrun-hello-world.log :  PASSED
           xrun-corev_rand_interrupt.log :  PASSED
           xrun-corev_rand_interrupt.log :  PASSED
                        xrun-illegal.log :  PASSED
               xrun-csr_instructions.log :  PASSED
  xrun-riscv_arithmetic_basic_test_0.log :  PASSED
xrun-corev_rand_arithmetic_base_test.log :  PASSED
          xrun-corev_rand_instr_test.log :  PASSED
    xrun-corev_rand_jump_stress_test.log :  PASSED
    xrun-corev_rand_jump_stress_test.log :  PASSED
```

Thx
Lee

MikeOpenHWGroup, 2023-02-27:
> However, a number of tests are reported as failing - but not mismatching.

A test can fail without mismatches.  For example, there could be a bus protocol error that throws a `uvm_error()`.   If you grep for UVM_ERROR in the logfile you can see if that is the case.

eroom1966, 2023-02-27:
@MikeOpenHWGroup agreed, I am more concerned with mismatches rather than tests FAILED to be hones - so I am actually happy with the above that this either resolves, or works around the compiler issue
If this is the case could someone look at merging the two PR's I submitted to correct the manifest file lists ?

MikeOpenHWGroup, 2023-02-27:
> so I am actually happy with the above that this either resolves, or works around the compiler issue

All good!

> could someone look at merging the two PR's I submitted to correct the manifest file lists ?

I assume you speak of #1676 (40s) and #1677 (40x)x is that correct? If so, they both look fine to me, but I will defer to @silabs-hfegran unless you are stuck.

silabs-hfegran, 2023-02-28:
I am fine with bringing in the #1676 and #1677 prior to these changes, if that is OK.  

As for this draft PR, I would like to do some more testing before pushing this to dev; this is a rather large change, and also introduces my riscv-dv fork with support for the Zc instructions. 
I will not however immediately introduce the changes in this PR to 40x; that will be done at a later point in time as a larger merge effort.

eroom1966, 2023-03-14:
I think this is resolved now so closing this issue - I am no longer seeing compilation errors

closed 2023-03-14 by eroom1966