openhwfoundation/core-v-verif #1680: How can I test hello-world with vsim?

Lucas-Wye, 2023-02-24:
Hi, I met a problem when trying to use vsim.

I run the hello-world program successfully with verilator, but when I switch to vsim, the hello-world program didn't have any output and the simulation didn't stop. I also generated the waveform but the variable `instr_rdata_i` is always `x`, which means there is not instructions in memory.

All above tests are based on CV32E40P processor.

MikeOpenHWGroup, 2023-02-24:
Hi @Lucas-Wye, thanks for your issue.  Please provide me the _exact_ command lines you use, starting from `git clone...` as well as the compiler logfiles.

Lucas-Wye, 2023-02-25:
Thanks for your support. It works for the latest version. I will transfer to this version.

closed 2023-02-25 by MikeOpenHWGroup