openhwfoundation/core-v-verif #1681: Missing vcs simulation file in cv32e40p/sim/tools/vcs/vcs_batch.tcl

shariethernet, 2023-02-24:
Hello OpenHW Group Team,

I am trying to run workloads on CV32E40P using VCS. As a starting point I tried to compile the hello-world test. It seems like some tcl files are missing for vcs. 

Path: `cv32e40p/sim/core`
Command:  `make vcs-hello-world`
Error Message:  ```do: couldn't open ../tools/vcs/vcs_batch.tcl```

Logfile:
[logfile.txt](https://github.com/openhwgroup/core-v-verif/files/10820589/logfile.txt)

Could you also suggest any other working methods (besides verilator) to run workloads on the simulation?  I would need to use a commercial simulator, as I would then be performing gate-level simulations, for some research on power-energy optimizations of a certain commercial PDK. 

Thanks
Shrihari

MikeOpenHWGroup, 2023-02-24:
Hi @shariethernet.

VCS is rarely used for the "core" testbench so the Makefile rules for running it with VCS are out of date.  Thanks for pointing this out.  I believe I have a fix and will create a pull-request for it shortly.  However, my VCS run-time license has expired (!) and so I will need to get that restored before I can know for sure if the problem is completely resolve.

As for the other questions, please reach out to me at mike@openhwgroup.org and we can discuss.

MikeOpenHWGroup, 2023-02-24:
Hi @shariethernet, please have a look at pull-request #1682.  It _should_ resolve your issue.  I will not get this PR merged in until I can test it locally (once our VCS run-time license is restored).

shariethernet, 2023-02-26:
Hi @MikeOpenHWGroup. Thanks for the update. I got it working, but there were still a few bugs in the Makefile(s), after some tweaking, I got hello-world up. Other tests still require some work on the makefile as the programs are not getting compiled. I will resolve them and raise a PR once it works
Thanks

cheng0560011, 2023-03-08:
Hi @MikeOpenHWGroup. I am recently working on cv32e40p using vcs. Thank you for the #1682 update.
I also find that line 545 in Makefile may need som modification.
Whe I make vcs-hello-world, $(COSTOM) will return black and the elf will be put under /. May it be $(TEST_PROGRAM_PATH)/hello/hello.elf instead of $(COSTOM)/hello.elf ?

MikeOpenHWGroup, 2023-03-08:
I will have a look @cheng0560011.  Tomorrow is already busy for me, so it will be later this week.

abdullahyildiz, 2023-03-08:
@cheng0560011,

Could you please try to run the following command?

`make sanity CUSTOM=. CV_SIMULATOR=vcs`

cheng0560011, 2023-03-08:
@abdullahyildiz 
Thank you for the advice. Testcase hello-world passed.

MikeOpenHWGroup, 2023-03-08:
Ah, _now_ I understand.  That is helpfull @abdullahyildiz, thanks.

HugoWei888, 2023-06-02:
Hello OpenHW Group Team and @MikeOpenHWGroup ,

I am also trying to run workloads on CV32E40P using VCS. But there are some errors as below.
Path:
$CORE_V_VERIF/cv32e40p/sim/core
Command:
make vcs-hello-world CV_SIMULATOR=vcs CV_SW_TOOLCHAIN=$CORE_V_VERIF/cv32e40p/sim/corev_gcc CV_SW_PREFIX=riscv32-corev-elf-
or
make sanity CUSTOM=. CV_SIMULATOR=vcs CV_SW_TOOLCHAIN=$CORE_V_VERIF/cv32e40p/sim/corev_gcc CV_SW_PREFIX=riscv32-corev-elf-

1. There are 2 cv32e40p_tb_wrapper.sv files in "$CORE_V_VERIF/core-v-cores/cv32e40p/bhv" and "$CORE_V_VERIF/cv32e40p/tb/core". Do I need to rename one of cv32e40p_tb_wrapper.sv files ?

Error-[MPD] Module previously declared
  The module was previously declared at: 
  "$CORE_V_VERIF/core-v-cores/cv32e40p/rtl/../bhv/cv32e40p_tb_wrapper.sv",  46
  It is redeclared later at:  "$CORE_V_VERIF/cv32e40p/tb/core/cv32e40p_tb_wrapper.sv",  18: token is 'cv32e40p_tb_wrapper'
  module cv32e40p_tb_wrapper
                            ^
  Please remove one of the declarations and compile again.

2. After I fixed above error by renaming, I found another error. Did it mean there were some problems with GCC ?
I have downloaded CORE-V top-of-tree compilers (GCC.tar.gz) with platform CentOS 7 (04 May 2023) and put it in $CORE_V_VERIF/cv32e40p/sim/.

*******************************************************************************************
* Compiling the BSP
*******************************************************************************************
mkdir -p simulation_results/hello-world/0/test_program/bsp
cp $CORE_V_VERIF/cv32e40p/bsp/Makefile simulation_results/hello-world/0/test_program/bsp
make -C simulation_results/hello-world/0/test_program/bsp \
	VPATH=$CORE_V_VERIF/cv32e40p/bsp \
	RISCV=$CORE_V_VERIF/cv32e40p/sim/corev_gcc \
	RISCV_PREFIX=riscv32-corev-elf- \
	RISCV_EXE_PREFIX=$CORE_V_VERIF/cv32e40p/sim/corev_gcc/bin/riscv32-corev-elf- \
	RISCV_MARCH=rv32imc \
	RISCV_CC=gcc \
	RISCV_CFLAGS="" \
	all
make[2]: Entering directory `$CORE_V_VERIF/cv32e40p/sim/core/simulation_results/hello-world/0/test_program/bsp'
$CORE_V_VERIF/cv32e40p/sim/corev_gcc/bin/riscv32-corev-elf-gcc -Os -g -static -mabi=ilp32 -march= -Wall -pedantic -c $CORE_V_VERIF/cv32e40p/bsp/crt0.S -o crt0.o
riscv32-corev-elf-gcc: error: missing argument to '-march='
make[2]: *** [crt0.o] Error 1
make[2]: Leaving directory `$CORE_V_VERIF/cv32e40p/sim/core/simulation_results/hello-world/0/test_program/bsp'
make[1]: *** [bsp] Error 2
make[1]: Leaving directory `$CORE_V_VERIF/cv32e40p/sim/core'
make: *** [/hello-world.elf] Error 2

Thanks
Hugo

pascalgouedo, 2023-06-02:
Hi @HugoWei888 

For your first sv compilation pb, the file from bhv should only be used for UVM tb and for core you should use the one in tb/core instead.
Don't know how to easily fix that as bhv/cv32e40p_tb_wrapper.sv is directly mentioned in cv32e40p manifest files ...

For gcc compilation march options, it seems your bsp Makefile does not propagate RISCV_MARCH variable value to CFLAGS one.
In master branch I see that:
> CFLAGS ?= -Os -g -static -mabi=ilp32 -march=$(CV_SW_MARCH) -Wall -pedantic

while in cv32e40p/dev it is 

> CFLAGS ?= -Os -g -static -mabi=ilp32 -march=$(RISCV_MARCH) -Wall -pedantic

I think master one is wrong and should use RISCV_MARCH rather than CV_SW_MARCH.
Temporarily you can bypass the pb by setting locally:

> setenv CV_SW_MARCH rv32imc_zicsr

or 

> export CV_SW_MARCH=rv32imc_zicsr

But you will have to remember to change it if you change the RTL configuration (manually) in tb_wrapper.
This RISCV_MARCH variable setting is now automatically managed by test yaml config files.

Best regards,
Pascal.

HugoWei888, 2023-06-02:
Hi @pascalgouedo

I have passed this testcase.

Thanks
Hugo

HugoWei888, 2023-06-05:
Hi @MikeOpenHWGroup
Yes, this issue was resolved.
It can be closed, thanks.

closed 2023-06-05 by MikeOpenHWGroup