openhwfoundation/core-v-verif #1692: CORE-V-Verif verifies a cv32e40s of one year old.

janhoogerbrugge, 2023-03-02:
CORE-V-Verif currently uses 103056f0deeac8e6cc10244c86bff83d3014f66f of cv32e40s which is more than a year old. Are there plans to upgrade CORE-V-Verif to a more recent cv32e40s? I tried to do this myself but I got lots of issues.

MikeOpenHWGroup, 2023-03-02:
Hi @janhoogerbrugge, thanks for your interest in CORE-V-VERIF.

The CV32E40S is in active development and the most recent activity for that core would be on the `cv32e40s/dev` branch of this repo.  As things stablize, updates will be propagated up to the master branch of this repo.

A "production ready" version of CV32E40S is not yet available, although it is expected to be ready by the middle of 2023.

If that answers you question, please close this issue.  Thanks!

janhoogerbrugge, 2023-03-03:
Thanks. cv32e40s/dev gave some problems. Xcelium was complaining about the +incdir option. Therefore I tried cv32e40s/release and that was working fine.

closed 2023-03-03 by janhoogerbrugge

MikeOpenHWGroup, 2023-03-03:
Hi @janhoogerbrugge, if `cv32e40s/dev` does not compile, _that_ is a problem.   Please create an issue for that, or reach out to me at mike@openhwgroup.org.