openhwfoundation/core-v-verif #1695: ISS mismatch for tdata2 (for tdata1.type==5)

silabs-halfdan, 2023-03-03:
# Mismatch between RTL/reference manual and the ISS

In the [reference manual](https://cv32e40s-user-manual.readthedocs.io/en/latest/control_status_registers.html#trigger-data-register-2-tdata2-view-when-tdata1-type-is-0x5) the tdata2 register (for tdata1.type==5) is described with only the bits for valid exception codes as writable. In the ISS however, any value can be written.
This triggers ISS mismatches when writing these bits as the ISS will store the info in the WARL(0x0) bit fields.

silabs-halfdan, 2023-03-03:
@eroom1966

silabs-halfdan, 2023-03-03:
@eroom1966 I'm in the process of making one. If you run debug_trigger_test on [this branch](https://github.com/silabs-halfdan/core-v-verif/tree/debug_trigger_test_iss_tdata2_issue) you should see the issue.

Running:
`makeuvmt test TEST=debug_test_trigger`

Results in:
```
Error (IDV) CSR register value mismatch (HartId:0, PC:0x1a1109b8 _debugger+362):
Info (IDV)  0> CSR 7a2 (tdata2)
Info (IDV)   . dut:0x030009ae
Info (IDV)   . ref:0xffffffff
```

silabs-halfdan, 2023-03-06:
For tdata1 we have a similar issue when tdata1.type==0x2.
In the [Reference manual](https://cv32e40s-user-manual.readthedocs.io/en/latest/control_status_registers.html#match-control-type-2-mcontrol) and in RTL the **ACTION** bit field is tied to 0x1.
Writing `tdata1 = 0x20000000` results in the following mismatch:
```
Info (IDV) Instruction executed prior to mismatch '0x1a110956(_debugger+264): 7a141073 csrw    tdata1,x8'
Error (IDV) CSR register value mismatch (HartId:0, PC:0x1a110956 _debugger+264):
Info (IDV)  0> CSR 7a1 (tdata1)
Info (IDV)   . dut:0x28001000 type:2(mcontrol) dmode:1 maskmax:0 hit:0 select:0 timing:0 sizelo:0(any) action:1(enter debug mode) chain:0 match:0 m:0 u:0 execute:0 store:0 load:0
Info (IDV)   . ref:0x28000000 type:2(mcontrol) dmode:1 maskmax:0 hit:0 select:0 timing:0 sizelo:0(any) action:0(raise breakpoint exception) chain:0 match:0 m:0 u:0 execute:0 store:0 load:0
```

silabs-halfdan, 2023-03-06:
@eroom1966, I found some more issues in the tdata1 register so I made a table with an overview in the ticket descripton.

eroom1966, 2023-03-06:
Hi @silabs-halfdan 

This issue is fixed with the commit to the ImperasDV-OpenHW repository
> commit 51add72a7fa9a4cf6c89a66a3ebe0c5b21a52dfd (HEAD -> main, origin/main, origin/HEAD)

This now produces the following result, please update and let me know if it matches what you see.

```
NUM_TRIGGERS = 1
csr_write: tselect = 0

 --- Testing execption triggers ---

trigger_test():
  Entered debug
    Setting up triggers
      csr_write: tdata1 = 0x50000200
      csr_write: tdata2 = 0x00000000 (0x0 + 0x0)
45758.000 ns: Illegal instruction (core 0) at PC 0x0000028c:
Warning (RISCV_CSR_UNIMP) CPU 'refRoot/cpu' 0x0000028c 7a805073 csrwi   mcontext,0: Unimplemented CSR
  Address match debug entry: 0 (expected: 0)

trigger_test():
  Entered debug
    Setting up triggers
      csr_write: tdata1 = 0x50000200
      csr_write: tdata2 = 0xffffffff (0xffffffff + 0x0)
81002.000 ns: Illegal instruction (core 0) at PC 0x0000028c:
Warning (RISCV_CSR_UNIMP) CPU 'refRoot/cpu' 0x0000028c 7a805073 csrwi   mcontext,0: Unimplemented CSR
  Entered debug
  Address match debug entry: 1 (expected: 1)

trigger_test():
  Entered debug
    Setting up triggers
      csr_write: tdata1 = 0x50000200
      csr_write: tdata2 = 0x0000040c (0x40c + 0x0)
118952.000 ns: Illegal instruction (core 0) at PC 0x0000028c:
Warning (RISCV_CSR_UNIMP) CPU 'refRoot/cpu' 0x0000028c 7a805073 csrwi   mcontext,0: Unimplemented CSR
  Entered debug
  Address match debug entry: 1 (expected: 1)

trigger_test():
  Entered debug
    Setting up triggers
      csr_write: tdata1 = 0x50000200
      csr_write: tdata2 = 0x00000400 (0x400 + 0x0)
156902.000 ns: Illegal instruction (core 0) at PC 0x0000028c:
Warning (RISCV_CSR_UNIMP) CPU 'refRoot/cpu' 0x0000028c 7a805073 csrwi   mcontext,0: Unimplemented CSR
  Address match debug entry: 0 (expected: 0)

trigger_test():
  Entered debug
    Setting up triggers
      csr_write: tdata1 = 0x50000200
      csr_write: tdata2 = 0x00000008 (0x8 + 0x0)
191678.000 ns: Illegal instruction (core 0) at PC 0x0000028c:
Warning (RISCV_CSR_UNIMP) CPU 'refRoot/cpu' 0x0000028c 7a805073 csrwi   mcontext,0: Unimplemented CSR
  Address match debug entry: 0 (expected: 0)

trigger_test():
  Entered debug
    Setting up triggers
      csr_write: tdata1 = 0x50000200
      csr_write: tdata2 = 0x00000004 (0x4 + 0x0)
226454.000 ns: Illegal instruction (core 0) at PC 0x0000028c:
Warning (RISCV_CSR_UNIMP) CPU 'refRoot/cpu' 0x0000028c 7a805073 csrwi   mcontext,0: Unimplemented CSR
  Entered debug
  Address match debug entry: 1 (expected: 1)

trigger_test():
  Entered debug
    Setting up triggers
      csr_write: tdata1 = 0x50000200
      csr_write: tdata2 = 0x00000000 (0x0 + 0x0)
Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x0000029a 9002     c.ebreak: EBREAK instruction executed (0x29a)
CV32E40S BSP: ebreak exception handler entered

  Address match debug entry: 0 (expected: 0)

trigger_test():
  Entered debug
    Setting up triggers
      csr_write: tdata1 = 0x50000200
      csr_write: tdata2 = 0xffffffff (0xffffffff + 0x0)
Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x0000029a 9002     c.ebreak: EBREAK instruction executed (0x29a)
  Entered debug
CV32E40S BSP: ebreak exception handler entered

  Address match debug entry: 1 (expected: 1)

trigger_test():
  Entered debug
    Setting up triggers
      csr_write: tdata1 = 0x50000200
      csr_write: tdata2 = 0x00000404 (0x404 + 0x0)
Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x0000029a 9002     c.ebreak: EBREAK instruction executed (0x29a)
CV32E40S BSP: ebreak exception handler entered

  Address match debug entry: 0 (expected: 0)

trigger_test():
  Entered debug
    Setting up triggers
      csr_write: tdata1 = 0x50000200
      csr_write: tdata2 = 0x00000008 (0x8 + 0x0)
Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x0000029a 9002     c.ebreak: EBREAK instruction executed (0x29a)
  Entered debug
CV32E40S BSP: ebreak exception handler entered

  Address match debug entry: 1 (expected: 1)

trigger_test():
  Entered debug
    Setting up triggers
      csr_write: tdata1 = 0x50000200
      csr_write: tdata2 = 0x0000000c (0xc + 0x0)
Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x0000029a 9002     c.ebreak: EBREAK instruction executed (0x29a)
  Entered debug
CV32E40S BSP: ebreak exception handler entered

  Address match debug entry: 1 (expected: 1)

trigger_test():
  Entered debug
    Setting up triggers
      csr_write: tdata1 = 0x50000200
      csr_write: tdata2 = 0x00000000 (0x0 + 0x0)
Warning (RISCV_IMA) CPU 'refRoot/cpu' 0x0000029a 9002     c.ebreak: EBREAK instruction executed (0x29a)
CV32E40S BSP: ebreak exception handler entered

  Address match debug entry: 0 (expected: 0)

Finished 

```

silabs-halfdan, 2023-03-07:
@eroom1966 , Thanks! The output looks good, I will try it out locally.

eroom1966, 2023-03-07:
You will need to merge some updates from cv32e40s/dev - your branch is a little behind

silabs-halfdan, 2023-03-07:
@eroom1966, Thanks,  I've rebased it to the latest cv32e40s/dev branch. And the ISS updates fix the tdata.type=5 issues.
There are still some issues for tdata.type=2 (see table in the ticket description).

eroom1966, 2023-03-07:
I cannot understand - so are you seeing a different output to what I am seeing as I showed here https://github.com/openhwgroup/core-v-verif/issues/1695#issuecomment-1456600809 ?

We have some incompatibility in our environments somehow
can you explain the exact set of steps to reproduce the mismatch you are seeing ?

silabs-halfdan, 2023-03-07:
I'm currently writing the test and finding issues as I write it, so those checks are not in your branch yet. I can push the updates to the test if you need them.

eroom1966, 2023-03-07:
The RISCV debug spec, and in particular the linkage between tdata1 and tdata2, and in particular the WARL behavior - is a challenge to say the least

silabs-halfdan, 2023-03-07:
I agree, cant believe a register map like that managed to get through ratification...

eroom1966, 2023-03-07:
I see the issue now I have referred to your table, BTW the same will hold true for the E40X, if you can update your test I will pull your changes and try

silabs-halfdan, 2023-03-07:
I've pushed an update to the test now so you can see the other issues.
This also includes a failure for a new issue derived from a vague point in the debug spec:
_"If a debugger writes an unsupported configuration, the register will read back a value that is
supported (which may simply be a disabled trigger)."_
We have interpreted this (i.e. the only solution given the other requirements) to mean that if tdata2 contains an illegal value for the type that is being written, the write will resolve to a disabled trigger.

eroom1966, 2023-03-07:
@silabs-halfdan I am getting many errors after pulling your updates

```
                             .DM_REGION_END        (uvmt_cv32e40s_pkg::CORE_PARAM_DM_REGION_END),
                                                                                              |
xmvlog: *E,NOSYM (/home/moore/tmp/issue-1695/core-v-verif/cv32e40s/tb/uvmt/uvmt_cv32e40s_tb.sv,99|94): CORE_PARAM_DM_REGION_END could not be found in package uvmt_cv32e40s_pkg.
                             .DM_REGION_START      (uvmt_cv32e40s_pkg::CORE_PARAM_DM_REGION_START),
                                                                                                |
xmvlog: *E,NOSYM (/home/moore/tmp/issue-1695/core-v-verif/cv32e40s/tb/uvmt/uvmt_cv32e40s_tb.sv,100|96): CORE_PARAM_DM_REGION_START could not be found in package uvmt_cv32e40s_pkg.
    .DM_REGION_START  (uvmt_cv32e40s_pkg::CORE_PARAM_DM_REGION_START),
                                                                   |
xmvlog: *E,NOSYM (/home/moore/tmp/issue-1695/core-v-verif/cv32e40s/tb/uvmt/uvmt_cv32e40s_pmp_assert.sv,76|67): CORE_PARAM_DM_REGION_START could not be found in package uvmt_cv32e40s_pkg.
    .DM_REGION_END    (uvmt_cv32e40s_pkg::CORE_PARAM_DM_REGION_END)
                                                                 |
....

                                                                |
xmvlog: *E,NOSYM (/home/moore/tmp/issue-1695/core-v-verif/cv32e40s/tb/uvmt/uvmt_cv32e40s_pmprvfi_assert.sv,258|65): CORE_PARAM_DM_REGION_END could not be found in package uvmt_cv32e40s_pkg.
xrun: *E,VLGERR: An error occurred during parsing.  Review the log file for errors with the code *E and fix those identified problems to proceed.  Exiting with code (status 1).

```

silabs-halfdan, 2023-03-07:
Ah yes, I think this is broken on cv32e40s/dev too. Pushed a fix on this branch for now.

eroom1966, 2023-03-07:
Thanks that fixed it
I have a question, why would a write to tdata1 with the value 0x5FFFFFFF cause it to take the value 0xF8000000

silabs-halfdan, 2023-03-07:
Yes, exactly; side-effects of the register mapping.
It is not described in the manual yet, but @silabs-oysteink is on it.

silabs-halfdan, 2023-03-07:
No, that resolves to a legal tdata2 value.
This statement from the debug spec is the reason for not doing the same for both:
_"Writes to one **tdata** register must not modify the contents of other **tdata** registers, nor the configuration
of any trigger besides the one that is currently selected"_

silabs-oysteink, 2023-03-07:
When tdata1 is configured as an exception trigger of type 0x5, any bits written to tdata2 that maps to unimplemented exception codes will be written as zero.

silabs-halfdan, 2023-03-07:
> I presume this is not the case, and that tdata2 cannot be written with an illegal value due to the WARL masking imposed by tdata1=etrigger ?

Exactly

eroom1966, 2023-03-07:
OK, I have implemented the WARL behavior as described here, I am running some tests now - I should have an update available shortly.
of all the RISCV specifications I have worked on the debug specification is my least favorite

eroom1966, 2023-03-07:
Committed
commit 2cc7839fc510bc61c4f46ee2a042b1dc19ad9b45 (HEAD -> main, origin/main, origin/HEAD)

I notice that the interrupt test no longer runs correctly

silabs-halfdan, 2023-03-07:
Thanks! That fixed the remaining tdata issues. I will close this ticket.
@silabs-hfegran is working on the interrupt test issues.

closed 2023-03-07 by silabs-halfdan