openhwfoundation/core-v-verif #1705: cv32e40s/dev : interrupt test failing

eroom1966, 2023-03-07:
The interrupt test is failing with the following output

> TEST 1 - TRIGGER ALL IRQS IN SEQUENCE:
> MCAUSE reported wrong irq, exp = 39, act = 0x80000003

Any idea why this was not caught by ci_check before committing ?

commit 99f8ad0a47551c3b2b5255de887b19850dfa5855 (HEAD -> cv32e40s/dev, origin/cv32e40s/dev)

silabs-hfegran, 2023-03-08:
I am looking into it - the commit you are pointing to only updated the riscv-dv hash which should have no impact on that directed tests.

silabs-hfegran, 2023-03-08:
After some consideration I think we should remove the test from ci_check for now - the recent breakage we are seeing in ci_check is likely due to updated compilers - the code has not changed for quite some time. Reading the source code I have a fairly good idea why it is no longer working and why it might have worked (or more likely, appeared to work) in the past.

eroom1966, 2023-03-08:
This is likely a very bad idea
there are only two tests in ci_check which exercise any asynchronous event behavior.
debug_test and interrupt_test

The debug_test has not been working for a long time probably more than 3 months
If you also remove the interrupt_test, then when I come to commit code using the ci_check
no asynchronous behavior would have been tested

That seems a really bad idea

silabs-hfegran, 2023-03-08:
The problem with the interrupt test that I would like to remove is that it is not doing what it claims to be doing. ( I want to reintroduce it at some point, but not in its current unstable form)

I believe that the failures we are seeing now is due to the compiler optimizing differently compared to previous versions, causing all sorts of undefined behavior due to lack of volatile qualifiers on certain variables (and possibly more errors). Looking at the code, to me it appears that that code previously "worked by accident", or even worse; "Appeared to work by accident", giving false confidence that something actually got tested that did not in fact get tested. 

we do have the random interrupt tests in the ci_check now, (working, though in rare instances it might time out):
[image]

In not too long I will also introduce a clic-directed test to the ci_check, but there are still some minor work remaining.

silabs-hfegran, 2023-03-08:
Edit to the above, I meant that there are some minor work remaining to the debug_test update; that and a clic directed will be added as soon as I get this all up to date, and the old debug-test will be retired.

eroom1966, 2023-03-08:
OK, so you are saying that if I run ci_check - this will still give some coverage of asynchronous activity on either interrupt or debug signals - if that is the case, I am happy for this short term measure.

silabs-hfegran, 2023-03-20:
This has been resolved - the fixed test takes too long to remain in the ci_check regressions and has thus been removed (It is still present in the full regression test)

closed 2023-03-20 by silabs-hfegran