openhwfoundation/core-v-verif #1707: Frequent sporadic ISS crashes

silabs-hfegran, 2023-03-08:
After the latest ISS update  ( 2cc7839fc510bc61c4f46ee2a042b1dc19ad9b45 ) I am seeing frequent ISS crashes, I do not have one specific command that reproduces the issue, it has been observed sporadically on different tests. 

(If the problem can be reproduced consistently I will post how to reproduce it here)

```
xmsim: *F,SIGUSR: Unix Signal SIGSEGV raised from user application code. Stack trace information is captured in file sigusrdump.out. 
```

ISS stracktrace
```
ImperasDVasync (64-Bit) v20230307.0 Open Virtual Platform simulator from www.IMPERAS.com.
Visit www.IMPERAS.com for multicore debug, verification and analysis solutions.

Internal Abort (ASRT) /home/moore/Imperas/SimCommon/source/memory/memRegion.c:10926 : Assertion failure : region contains code

*** backtrace 14 stack frames:
-------------------------------
/work/hefegran/imperas/ImperasDV-OpenHW/Imperas/bin/Linux64/libCpuManager.so(+0xaae6a) [0x7f9f0a712e6a]
/work/hefegran/imperas/ImperasDV-OpenHW/Imperas/bin/Linux64/libCpuManager.so(+0xab3de) [0x7f9f0a7133de]
/work/hefegran/imperas/ImperasDV-OpenHW/Imperas/bin/Linux64/libCpuManager.so(+0xab5f9) [0x7f9f0a7135f9]
/work/hefegran/imperas/ImperasDV-OpenHW/Imperas/bin/Linux64/libCpuManager.so(+0xab687) [0x7f9f0a713687]
/work/hefegran/imperas/ImperasDV-OpenHW/Imperas/bin/Linux64/libCpuManager.so(+0x11584e) [0x7f9f0a77d84e]
/work/hefegran/imperas/ImperasDV-OpenHW/Imperas/bin/Linux64/libCpuManager.so(+0x11ce8e) [0x7f9f0a784e8e]
/work/hefegran/imperas/ImperasDV-OpenHW/Imperas/bin/Linux64/libCpuManager.so(+0x11cf05) [0x7f9f0a784f05]
/work/hefegran/imperas/ImperasDV-OpenHW/Imperas/bin/Linux64/libCpuManager.so(+0x71df6) [0x7f9f0a6d9df6]
/work/hefegran/imperas/ImperasDV-OpenHW/Imperas/bin/Linux64/libCpuManager.so(+0x72b13) [0x7f9f0a6dab13]
/work/hefegran/imperas/ImperasDV-OpenHW/Imperas/bin/Linux64/libCpuManager.so(opiSessionTerminate+0xeb) [0x7f9f0a6ee67b]
/work/hefegran/imperas/ImperasDV-OpenHW/Imperas/lib/Linux64/ImperasLib/imperas.com/verification/riscv/1.0/../../../../../../../bin/Linux64/libRuntimeLoader.so(opSessionTerminate+0x2d) [0x7fa31659c61d]
/work/hefegran/imperas/ImperasDV-OpenHW/Imperas/lib/Linux64/ImperasLib/imperas.com/verification/riscv/1.0/model.so(implRefShutdown+0x45) [0x7fa316811d05]
/work/hefegran/imperas/ImperasDV-OpenHW/Imperas/lib/Linux64/ImperasLib/imperas.com/verification/riscv/1.0/model.so(rvviRefShutdown+0x24) [0x7fa316814814]
[0x7fa31fd929f3]

```

```
[INS] % cat /scratch/hefegran/results/xrun_results/clic_default/debug_test2/0/sigusrdump.out
*** Current stack trace:
--> 0x7fa31681770a hartCsrVolatileMaskGet + a        /work/hefegran/imperas/ImperasDV-OpenHW/Imperas/lib/Linux64/ImperasLib/imperas.com/verification/riscv/1.0/model.so
--> 0x7fa32c8c3630 <don't know>         + f630     /usr/lib64/libpthread.so.0
--> 0x7fa31681770a hartCsrVolatileMaskGet + a        /work/hefegran/imperas/ImperasDV-OpenHW/Imperas/lib/Linux64/ImperasLib/imperas.com/verification/riscv/1.0/model.so
--> 0x7fa3168199f9 <don't know>         + 1f9f9    /work/hefegran/imperas/ImperasDV-OpenHW/Imperas/lib/Linux64/ImperasLib/imperas.com/verification/riscv/1.0/model.so
--> 0x7f9f0a6f7be1 <don't know>         + 8fbe1    /work/hefegran/imperas/ImperasDV-OpenHW/Imperas/bin/Linux64/libCpuManager.so
--> 0x7f9f0a6f81ff opiNetWrite          + 8f       /work/hefegran/imperas/ImperasDV-OpenHW/Imperas/bin/Linux64/libCpuManager.so
--> 0x7f9f0a7d2d1b <don't know>         + 16ad1b   /work/hefegran/imperas/ImperasDV-OpenHW/Imperas/bin/Linux64/libCpuManager.so
--> 0x7f9f05e569ad <don't know>         + fffffffffb7ee9ad /work/hefegran/imperas/ImperasDV-OpenHW/Imperas/bin/Linux64/libCpuManager.so
```

eroom1966, 2023-03-08:
@silabs-hfegran 
Can you give me an idea of what you were running, I can try running in a loop to try to reproduce

silabs-hfegran, 2023-03-13:
Sorry about the late reply, 
I've observed it on this branch: https://github.com/openhwgroup/core-v-verif/pull/1717/commits, on which I have been mostly concerned with debug_test2. I have only seen it in GUI-mode with xcelium/indago. (have not tested other simulators)
e.g. 
```
make test TEST=debug_test2 CFG=clic_default GUI=1 WAVES=1 ADV_DEBUG=1 USE_ISS=1

or

make test TEST=debug_test2 CFG=default GUI=1 WAVES=1 ADV_DEBUG=1 USE_ISS=1
```

Note that there are mismatches in these tests (but that is a topic for another issue)

eroom1966, 2023-03-13:
@silabs-hfegran 
Thanks we were able to reproduce this from your description
fix is on the way.

eroom1966, 2023-03-14:
This is fixed by the following commit for the REF model
commit d762db0c2181c34e835d5a3a00d0da833f772671 (HEAD -> main, origin/main, origin/HEAD)
Author: eroom1966 <moore@imperas.com>
Date:   Tue Mar 14 11:11:51 2023 +0000

    Fix Issues 1717 1719

closed 2023-03-16 by silabs-hfegran