openhwfoundation/core-v-verif #1709: OVPsim failed with CV32E40S

weik56100, 2023-03-08:
Hello everyone, I have a problem when I try to run the hello-world program with OVPsim on the CV32E40S processor.
[image]
(Do not pay attention to the name of the executed program)

It works partially, I get errors after a while of execution and the program does not finish correctly. I don't understand why the program crashes on the debugger at address 0x1a110800. (debug vector address).
There is an illegal instruction that prevents the program from terminating normally.

If I use the link editor of the CV32E40S and build it manually, I have the same problem but no print from the program appears at all.
Any ideas?

[image]

Manual script :

`#!/bin/bash`

`PATH=${PATH}:/opt/riscv_last/bin`
`CORE_V_VERIF=/opt/core-v-verif`

`MARCH=rv32imc_zicsr`

`riscv32-corev-elf-gcc \`
   `-Os -g -static -mabi=ilp32  \`
   `-march=${MARCH} \`
   `-Wall -pedantic    \`
   `-I ${CORE_V_VERIF}/cv32e40s/tests/asm \`
   `-I ${CORE_V_VERIF}/cv32e40s/bsp  \`
   `-o hello-world.elf  \`
   `-nostartfiles ${CORE_V_VERIF}/cv32e40s/tests/programs/custom/hello-world/hello-world.c  \`
   `-T ${CORE_V_VERIF}/cv32e40s/bsp/link.ld   \`
   `#-L ${CORE_V_VERIF}/cv32e40s/sim/core/simulation_results/hello-world/0/test_program/bsp  \`
   `#-lcv-verif`

`echo`
`riscv32-corev-elf-gcc --version`

Any ideas why the execution is not going well?

### Steps to Reproduce
`sudo git clone https://github.com/openhwgroup/core-v-verif.git`

Branch: cv32e40s/
ExternalRepos.mk :
```
CV_CORE_REPO   ?= https://github.com/openhwgroup/cv32e40s
CV_CORE_BRANCH ?= master
CV_CORE_HASH   ?= a9c6c716e4d1a219775b968bafec239f1190ac16
CV_CORE_TAG    ?= none
```

### Additional context
Ubuntu 22.04.1 LTS

silabs-hfegran, 2023-03-09:
Hi @weik56100, I see that you are running simulations on the master branch - It has been quite a while since we merged development to master (cv32e40s/dev-branch). We only actively maintain simulations on the development branch, and this is where you should be able to expect working simulations. As things stand today, master may or may not work. If you are not dependent on the instruction simulator, you may disable its usage by adding USE_ISS=0 to the make invocation that starts the simulation. 

As for the compilation error - it looks like somehow you are triggering debug (1a110800 is the default debug start address), and the hello world test does not initialize this section of memory. Did you try running our make scripts (from core-v-verif/cv32e40s/sim/uvmt?  `make test TEST=hello-world`

@MikeOpenHWGroup Actually, the details and relations between the simulators we used before and now are a bit fuzzy to me - we swapped over to a different interface to the simulator, ImperasDV (requiring a local installation of a new version and license), Though from the messages output from this binary, it appears to be a variant of OVPSim: 
[image]

eroom1966, 2023-03-10:
I should add, I don't think that 'OVPsim failed with CV32E40S'
The failure looks like it is in the application code ?

eroom1966, 2023-03-10:
@weik56100 
Can I ask you to do the following

1. upload the elf file you are running
2. provide the exact command line you are using to run riscvOVPsimCOREV

I can take a look and debug. I tried to reproduce your findings from your instructions but was not able.

eroom1966, 2023-03-10:
I see what the problem is here

Firstly, riscvOVPsimCOREV is doing **exactly the right thing** for the program it is executing, let me explain

The version of the model which you are simulating has a default debug trigger at address 0x00000000

When you execute your code, I am guessing it starts at some non zero address - you do not show the entire trace.
At the end of executing main you perform a return
> c.jr ra

I am assuming the ra has the value 0x0
Now the processor fetches at address 0x0
Address 0x0 has the debug trigger address
So now the processor goes into debug mode.
The (default) debug address is 0x1a110800
Now the processor attempts to fetch an instruction at address 0x1a110800
You do not have any code at address 0x1a110800, so now you get another exception as this is an illegal instruction

So in this example, riscvOVPsimCOREV has simulated your program faithfully, but your program has a number of bugs that have been highlighted.

You could probably get a little further by changing the variant using 
--variant CV32E40S_DEV
This variant for specification 0.8.0 DOES NOT have the same issue with going into debug mode when fetching from address 0x0

Maybe you could change the title of this issue from 
OVPsim failed with CV32E40S, to
**OVPsim helped me diagnose a program error which was very difficult to identify** !

weik56100, 2023-03-13:
@eroom1966 
I tried to see with the variant CV32E40S_DEV and I have exactly the same problem. There are no more instructions that are executed than the previous variant.

here are my files to help:
[hello-world_OVPSim.zip](https://github.com/openhwgroup/core-v-verif/files/10956160/hello-world_OVPSim.zip)

eroom1966, 2023-03-13:
I dont understand - I just downloaded your zip file and ran agains riscvOVPsimCOREV version 20230201 and it ran perfectly
I attach the entire log for you to take a look.

[sim.log](https://github.com/openhwgroup/core-v-verif/files/10956385/sim.log)

What is more interesting is that I am not sure you are running the file you actually provided.
When I load the elf file provided I see the following program headers loaded
```
Info (OR_OF) Target 'riscvOVPsim/cpu' has object file read from 'hello-world.elf'
Info (OR_PH) Program Headers:
Info (OR_PH) Type           Offset     VirtAddr   PhysAddr   FileSiz    MemSiz     Flags Align
Info (OR_PD) PROC           0x000050d7 0x00000000 0x00000000 0x00000033 0x00000000 R--   1
Info (OR_PD) LOAD           0x00001000 0x00000000 0x00000000 0x00003ed8 0x00400000 RWE   1000
Info (OR_PD) LOAD           0x000001e0 0x000041e0 0x000041e0 0x00000000 0x003fbe20 RW-   1000
Info (OR_PD) LOAD           0x00005000 0x00100000 0x00100000 0x0000008a 0x0000008a R-E   1000
Info (OR_PD) LOAD           0x00000800 0x1a110800 0x1a110800 0x00000000 0x00000080 RW-   1000
```
In your earlier post, The headers look totally different, please go back and compare
It appears you are not running the program you think you are running

weik56100, 2023-03-13:
I get the same thing than you when I run the "short version" of hello-world. (like zip files archive)

But why the "Hello world" is not displayed in the simulation interface? even if it's works, I don't see it in your simulation log result file. 

What is strange is that on my side I have to stop the simulation with `CTRL + C`, it does not stop by itself.
[image]

I had to do some tests with and without linker, which explains the different headers from my initial post.

If we take the initial C file hello-world of the CV32E40S 
https://github.com/openhwgroup/core-v-verif/blob/cv32e40s/dev/cv32e40s/tests/programs/custom/hello-world/hello-world.c

with this script:
```
#!/bin/bash

PATH=${PATH}:/opt/riscv_last/bin
#PATH=${PATH}:/opt/riscv_test/lib/Linux32/CrossCompiler/riscv-none-embed/bin
CORE_V_VERIF=/opt/core-v-verif

MARCH=rv32imc_zicsr

riscv32-corev-elf-gcc \
   -Os -g -static -mabi=ilp32  \
   -march=${MARCH} \
   -Wall -pedantic    \
   -I ${CORE_V_VERIF}/cv32e40s/tests/asm \
   -I ${CORE_V_VERIF}/cv32e40s/bsp  \
   -o hello-world.elf  \
   -nostartfiles ${CORE_V_VERIF}/cv32e40s/tests/programs/custom/hello-world/hello-world.c  \
   -T ${CORE_V_VERIF}/cv32e40s/bsp/link.ld   \
   -L ${CORE_V_VERIF}/cv32e40s/sim/core/simulation_results/hello-world/0/test_program/bsp  \
   -lcv-verif

echo
riscv32-corev-elf-gcc --version
```
I get this : (like my first post)

[image]

the full log with all traces:
[hello-world-cv32e40s.log](https://github.com/openhwgroup/core-v-verif/files/10958658/hello-world-cv32e40s.log)

the simulation does not display the desired result and it does not end.
I want to believe that everything is going well, but why does OVPSim stop before the end and does not display the summary information and why are the prints not visible in the terminal?

if I delete the last 3 lines of my building script:

   ```
-T ${CORE_V_VERIF}/cv32e40s/bsp/link.ld   \
   -L ${CORE_V_VERIF}/cv32e40s/sim/core/simulation_results/hello-world/0/test_program/bsp  \
   -lcv-verif
```
when i compile I get :

[image]

_start is not found and the compilation uses the default address **10094**.

[image]

But in the end in these conditions, OVPSim executes some instructions and displays it in the terminal but it stops before the end.
The simulation ends despite the errors.

I guess there is a problem with a linker address at compile time somewhere but I don't see what is going wrong.

eroom1966, 2023-03-14:
Please try the following by attaching a debugger

add this flag to your command line

--trace --tracechange --gdbconsole

when the gdb console pops up, add a breakpoint at the debug address
break * 0x1a110800

now continue till the breakpoint
continue

Now try to determine why your execution has gone into debug mode, looking at 
dcsr, tdata1, tdata2 etc

MikeOpenHWGroup, 2023-03-14:
Hi @weik56100, I think your work is very interesting.  Can you reach out to me at mike@openhwgroup.org?

weik56100, 2023-03-14:
Trace log with full compilation options:
[Full_execution.log](https://github.com/openhwgroup/core-v-verif/files/10970194/Full_execution.log)

Trace log log with less compilation options:
Remove this 3 lines:
```
-T ${CORE_V_VERIF}/cv32e40s/bsp/link.ld   \
-L ${CORE_V_VERIF}/cv32e40s/sim/core/simulation_results/hello-world/0/test_program/bsp  \
-lcv-verif
```
[Full_execution_without_linker.log](https://github.com/openhwgroup/core-v-verif/files/10970206/Full_execution_without_linker.log)

Test with debugger:
[image]

eroom1966, 2023-03-14:
@weik56100 

Why have you removed the startup code ?
If you do this, the code is now broken for the following reasons

Here is the start of your execution log
```
Info (OR_OF) Target 'riscvOVPsim/cpu' has object file read from 'hello-world.elf'
Info (OR_PH) Program Headers:
Info (OR_PH) Type           Offset     VirtAddr   PhysAddr   FileSiz    MemSiz     Flags Align
Info (OR_PD) PROC           0x00012761 0x00000000 0x00000000 0x00000033 0x00000000 R--   1
Info (OR_PD) LOAD           0x00000000 0x00010000 0x00010000 0x00011c30 0x00011c30 R-E   1000
Info (OR_PD) LOAD           0x00012000 0x00022000 0x00022000 0x00000714 0x00000890 RW-   1000
Info 'riscvOVPsim/cpu', 0x0000000000010094(main): 1141     c.addi  sp,sp,-16
Info   sp fffff000 -> ffffeff0
```
The first instruction is at address 0x10094, which means in the elf file you have defined this as the elf entry point
There is no startup code prior to this, normally the first instruction is the C runtime startup code. this then results in the following issue at the end

`Info 'riscvOVPsim/cpu', 0x00000000000100de(main+4a): 8082     c.jr    ra`
This says jump to the return address, but this was never written as there was no startup code, so now the processor jumps to address 0x0
`Info 'riscvOVPsim/cpu', 0x0000000000000000: 00000043 fmadd.s ft0,ft0,ft0,ft0,rne`
This address 0x0 has a debug trigger set, and so the processor goes into debug mode, setting the following registers
```
Info   dcsr 40000013 -> 40000093
Info   DM 00 -> 01
```
The debug vector address is 0x1a110800, so now the processor fetches from this location in debug mode
`Info 'riscvOVPsim/cpu', 0x000000001a110800: 0000     c.illegal`
however this location contains no code (illegal instruction) and hence we have an exception
Processor Exception (PC_PRX) Processor 'riscvOVPsim/cpu' 0x1a110800: 0000     c.illegal
Processor Exception (PC_FIX) Executing at uninitialized address 0x1a110800

The root of your problems is that your elf file has no startup code.
As I mentioned in a previous email, the program you are executing here - is not the same as the program you provided in the zip file. This can be seen by the fact that 

The problems you are seeing are unrelated to the simulator, they are problems with how you are creating the executable
The simulator is executing exactly how it should, given the content of the executable program, the problem is that the program is not what you think it is, as you hove removed the startup code.

weik56100, 2023-03-15:
@eroom1966 
Thank you for the explanations, it is very interesting.

I did several tests and what I found strange is that by removing some compiler options, the simulation could go further.

If I do it right from the beginning
I compile my program with this script: (startup code should be there)
all compiler options are enabled as below:
```
#!/bin/bash

PATH=${PATH}:/opt/riscv_last/bin
#PATH=${PATH}:/opt/riscv_test/lib/Linux32/CrossCompiler/riscv-none-embed/bin
CORE_V_VERIF=/opt/core-v-verif

MARCH=rv32imc_zicsr

riscv32-corev-elf-gcc \
   -Os -g -static -mabi=ilp32  \
   -march=${MARCH} \
   -Wall -pedantic    \
   -I ${CORE_V_VERIF}/cv32e40s/tests/asm \
   -I ${CORE_V_VERIF}/cv32e40s/bsp  \
   -o hello-world.elf  \
   -nostartfiles ${CORE_V_VERIF}/cv32e40s/tests/programs/custom/hello-world/hello-world.c  \
   -T ${CORE_V_VERIF}/cv32e40s/bsp/link.ld   \
   -L ${CORE_V_VERIF}/cv32e40s/sim/core/simulation_results/hello-world/0/test_program/bsp  \
   -lcv-verif

echo
riscv32-corev-elf-gcc --version
```
I use the following cv32e40s/dev linker:
https://github.com/openhwgroup/core-v-verif/blob/cv32e40s/dev/cv32e40s/bsp/link.ld

In this case I don't have anymore the "input symbol not found" problem:
[image]
Everything is good and it makes sense.

Now if I run a simulation with OVPsim with the default options I get this:
[image]
[image]

The simulation does not finish and I have no results on the screen of the CV32E40S hello-world program.
Where is the display of the hello-world and the CPU information but also the summary table of the OVPSim?
I am still stuck in this state.

If I activate  `--trace --tracechange` here are the logs:
[Trace.log](https://github.com/openhwgroup/core-v-verif/files/10978754/Trace.log)

If I do things right I'm here.

For the CV32E40P and the same hello-world program (version CV32E40P), I use VERILATOR and the result of the simulation gives me that:
`sudo make veri-test TEST=hello-world`
[image]
Everything works perfectly well in this case.

As VERILATOR is not compatible with the CV32E40S, I try to find a solution to simulate the processor, OVPSim seems to be the best candidate for that. I didn't manage to make something work with Questa Sim.

One of my previous posts mentions this problem:
https://github.com/openhwgroup/core-v-verif/issues/1709#issuecomment-1463897093

eroom1966, 2023-03-15:
@weik56100 

I now see the confusion

The riscvOVPsimCOREV simulator provides a platform with a processor instance and memory.
In order to handle semihosting (printf/write etc) it will support the PK semihosting calls.

core-v-verif is a platform environment which contains timers, random number generators and importantly **a UART.**
The program flow you are using writes characters to the UART data register in order to print characters through a virtual peripheral to the screen. These same writes simply write to memory in riscvOVPsimCOREV, hence you do not see any characters, as these are simply byte stores to memory

> riscvOVPsimCOREV is restricted to only run RISC-V processor model variants in a fixed platform
> configuration of **one processor instance and one memory sub-system**. Caches and other processor
> microarchitecture features are not included in programmer view models. If you need different
> platform configurations or to extend the platform or models then please contact contact Imperas
> or visit www.OVPworld.org.

weik56100, 2023-03-16:
I understand better now why I don't see any display since the beginning. Thank you for your explanations.

I have two questions that come to mind.

If everything goes well, why doesn't the simulation stop by itself at the end? I have to kill it with `CTRL + C `each time to get the statistical tables from OVPSim.

If I want to display something from my C program on the screen is there a solution?
Can I display the bytes values stored in memory?

eroom1966, 2023-03-16:
> If everything goes well, why doesn't the simulation stop by itself at the end? I have to kill it with `CTRL + C `each time to get the statistical tables from OVPSim.

I think this is because the last instruction in your program is a WFI
So it is 'waiting for an Interrupt'
Please confirm it is a WFI, by enabling instruction trace

> If I want to display something from my C program on the screen is there a solution? Can I display the bytes values stored in memory?
Have you read the user guide ?
https://github.com/openhwgroup/riscv-ovpsim-corev/blob/v20230201/doc/riscvOVPsimCOREV_User_Guide.pdf
please refer to Appendix C Compiling RISC-V programs

MikeOpenHWGroup, 2023-03-16:
@weik56100, @eroom1966, I see that this issue is being used as a "chat room", which is not how we like to use GitHub Issues.  Please keep issues to a single topic and close the issue when the topic is closed.

I strongly recommend moving this discussion to [MatterMost](https://mattermost.openhwgroup.org/).  If you like, I can create a specific channel for discussions about the `riscvOVPsimCOREV` model.

@weik56100, if you do not have a MatterMost ID, please reach out to me and I can set you up.  It is easy!

moyouth, 2023-10-18:
> @weik56100 Can I also request that you provide a full execution trace log so we can see that has happened you can attach to this github thread
> 
> please add these flags and capture all output --trace --tracechange

Run command: python3 run.py --test riscv_arithmetic_basic_test --target rv64gc --iterations 1 --simulator vcs --iss ovpsim

riscv_arithmetic_basic_test_0.log

[image]

iss.yaml
[image]

why ovpsim does not support "--trace --tracechange"

silabs-hfegran, 2024-01-11:
Closing issue as initial issue was resolved - please create new issues if there are further things that needs to be addressed.

closed 2024-01-11 by silabs-hfegran