openhwfoundation/core-v-verif #1719: ISS mismatch at tcontrol register (for 0 implemented triggers)

silabs-halfdan, 2023-03-13:
For DBG_NUM_TRIGGERS==0, the tcontrol register is implemented and tied to 0 while all the other trigger registers trap (not implemented) when no triggers are implemented.
It is not stated explicitly in the reference manual (added a separate issue for it).
## To reproduce

### Branch:
https://github.com/silabs-halfdan/core-v-verif/tree/iss_tcontrol_issue

### Test: 
`makeuvmt test TEST=debug_test_0_triggers CFG=debug_trigger_cfg0`

### Error:
```
Warning (RISCV_UICSR) CPU 'refRoot/cpu' 0x000001fe 7a502473 csrr    x8,tcontrol: Unimplemented CSR (hardwired to zero)
Info (IDV) Instruction executed prior to mismatch '0x1fa(test_register_access+194): d401a223 sw      x0,-700(x3)'
Error (IDV) PC mismatch (HartId:0, PC:0x000001fe test_register_access+198):
Info (IDV)  0>
Info (IDV)   . dut:0x00000000 vector_table+0
Info (IDV)   . ref:0x000001fe test_register_access+198
```

eroom1966, 2023-03-13:
@Silabs-ArjanB (copying Arjan for comment)

I am a bit confused here, as this **is explicitly** in the documentation
https://docs.openhwgroup.org/projects/cv32e40s-user-manual/en/0.8.0/control_status_registers.html#control-and-status-register-map
[image]

Silabs-ArjanB, 2023-03-13:
@eroom1966 It seems that the ISS implements the tcontrol CSR (hardwired to 0) if DBG_NUM_TRIGGERS == 0, which violates the RTL and the User Manual. I.e. the ISS does incorrectly _not_ treat tcontrol as illegal if DBG_NUM_TRIGGERS == 0.

silabs-halfdan, 2023-03-14:
> @Silabs-ArjanB (copying Arjan for comment)
> 
> I am a bit confused here, as this **is explicitly** in the documentation https://docs.openhwgroup.org/projects/cv32e40s-user-manual/en/0.8.0/control_status_registers.html#control-and-status-register-map [image]

Exactly, but it is present and tied to 0 in the ISS.

eroom1966, 2023-03-14:
yes my mistake - I read this the other way round, in your initial comment
I thought this comment was related to the RTL
> For DBG_NUM_TRIGGERS==0, the tcontrol register is implemented and tied to 0 while all the other trigger registers trap (not implemented) when no triggers are implemented.
> It is not stated explicitly in the reference manual (added a separate issue for it).

Whereas I now realize you were referring to the REF model
I have a fix for this which I will push shortly

In E40S/X This register is different to the base debug register (when present
The modificaton for E40S/X had it present unconditionally, rather than sensitized by num_triggers
Fix available shortly

silabs-halfdan, 2023-03-14:
Thanks! That fixed the issue.

closed 2023-03-14 by silabs-halfdan