openhwfoundation/core-v-verif #1730: ISS: Mismatch on GPRs when clearing/setting dcsr bits in debug mode

silabs-hfegran, 2023-03-20:
## DCSR read/write bitfields GPR read value mismatch in debug mode

The following sequence of operations creates ISS mismatches, where the reference model appears to readback the value it wrote, not the value that was present prior to the csrrw instruction: enter debug mode, read dcsr and write all 0s, read dcsr and write all 1s, read dcsr and write back original value.

### Type
* Functionally incorrect behavior

### Steps to Reproduce
Please provide:
1. cv32e40s/dev 04a877e90098d126238caf32069cbfdf1b99acff
2. make test TEST=debug_test2 CFG=default USE_ISS=1  (This is at the end of a larger test, to only run this failing part, change the START_TEST_NUM define in debug_test2.c to 18)

[image]

```
Info (IDV) Instruction executed prior to mismatch '0x3230(cover_known_iss_mismatches_dbg+12): 7b0717f3 csrrw   x15,dcsr,x14'
Error (IDV) GPR register value mismatch (HartId:0, PC:0x00003230 cover_known_iss_mismatches_dbg+12):
Info (IDV)  0> GPR x15
Info (IDV)   . dut:0x400004d3
Info (IDV)   . ref:0x400000d0
UVM_ERROR @ 40542.300 ns : rvvi-pkg.sv(93) reporter [] uvmt_cv32e40s_tb.imperas_dv.idv_trace2api.state_compare @ 40542.000 ns: MISMATCH
Info (IDV) Instruction executed prior to mismatch '0x3234(cover_known_iss_mismatches_dbg+16): 577d     c.li    x14,-1'
Error (IDV) GPR register value mismatch (HartId:0, PC:0x00003234 cover_known_iss_mismatches_dbg+16):
Info (IDV)  1> GPR x15
Info (IDV)   . dut:0x400004d3
Info (IDV)   . ref:0x400000d0
UVM_ERROR @ 40551.300 ns : rvvi-pkg.sv(93) reporter [] uvmt_cv32e40s_tb.imperas_dv.idv_trace2api.state_compare @ 40551.000 ns: MISMATCH
Info (IDV) Instruction executed prior to mismatch '0x3236(cover_known_iss_mismatches_dbg+18): 7b071773 csrrw   x14,dcsr,x14'
Error (IDV) GPR register value mismatch (HartId:0, PC:0x00003236 cover_known_iss_mismatches_dbg+18):
Info (IDV)  2> GPR x14
Info (IDV)   . dut:0x400000d0
Info (IDV)   . ref:0x40009cd7
Info (IDV)  3> GPR x15
Info (IDV)   . dut:0x400004d3
Info (IDV)   . ref:0x400000d0
UVM_ERROR @ 40557.300 ns : rvvi-pkg.sv(93) reporter [] uvmt_cv32e40s_tb.imperas_dv.idv_trace2api.state_compare @ 40557.000 ns: MISMATCH
Info (IDV) Instruction executed prior to mismatch '0x323a(cover_known_iss_mismatches_dbg+22): 7b079773 csrrw   x14,dcsr,x15'
Error (IDV) GPR register value mismatch (HartId:0, PC:0x0000323a cover_known_iss_mismatches_dbg+22):
Info (IDV)  4> GPR x14
Info (IDV)   . dut:0x40009cd7
Info (IDV)   . ref:0x400000d0
Info (IDV)  5> GPR x15
Info (IDV)   . dut:0x400004d3
Info (IDV)   . ref:0x400000d0
UVM_ERROR @ 40566.300 ns : rvvi-pkg.sv(93) reporter [] uvmt_cv32e40s_tb.imperas_dv.idv_trace2api.state_compare @ 40566.000 ns: MISMATCH
Info (IDV) Instruction executed prior to mismatch '0x323e(cover_known_iss_mismatches_dbg+26): c43a     c.swsp  x14,8(x2)'
Error (IDV) GPR register value mismatch (HartId:0, PC:0x0000323e cover_known_iss_mismatches_dbg+26):
Info (IDV)  6> GPR x14
Info (IDV)   . dut:0x40009cd7
Info (IDV)   . ref:0x400000d0
Info (IDV)  7> GPR x15
Info (IDV)   . dut:0x400004d3
Info (IDV)   . ref:0x400000d0
UVM_ERROR @ 40575.300 ns : rvvi-pkg.sv(93) reporter [] uvmt_cv32e40s_tb.imperas_dv.idv_trace2api.state_compare @ 40575.000 ns: MISMATCH
```

[image]

silabs-hfegran, 2023-03-20:
Also occurs in clic mode, with CFG=clic_default

silabs-hfegran, 2023-03-20:
Note, with clic_default there is a mismatch in the test prior to the testcase that was specifically crafted to reproduce this one, thus I expect both configurations to run all tests (with START_TEST_NUM 1) without any mismatches when this is resolved.

eroom1966, 2023-03-20:
Hi Henrik,
I think this issue was already resolved, but maybe I had not released it back to you

The first failure I see is here
```
Info (IDV) Instruction executed prior to mismatch '0x1dc0(has_pmp_configured+12): 3b079773 csrrw   x14,pmpaddr0,x15'
Error (IDV) CSR register value mismatch (HartId:0, PC:0x00001dc0 has_pmp_configured+12):
Info (IDV)  0> CSR 3b0 (pmpaddr0)
Info (IDV)   . dut:0x00000000 address:0x0
Info (IDV)   . ref:0xffffffff address:0x3fffffffc
UVM_ERROR @ 948849.300 ns : rvvi-pkg.sv(93) reporter [] uvmt_cv32e40s_tb.imperas_dv.idv_trace2api.state_compare @ 948849.000 ns: MISMATCH
```
Which looks much later than you report, and unrelated to the DCSR

I will try and get a release out today for you to try

silabs-hfegran, 2023-03-21:
The message you are seeing should not happen if you changed the starting test to 18 (only running the case that recreates this problem) - The pmp-configuration is only set for test 17. I am seeing the same error as you if I start execution with test 17 (change line 38 to `#define START_TEST_NUM 17`) on `CFG=default`, and dcsr-error if I start with test 17 and `CFG=clic_default`

eroom1966, 2023-03-21:
OK, I misunderstood, I thought you were saying that it just took longer to run

eroom1966, 2023-03-21:
I have found the issue - this is related to the fact that we have defined one of the bits (interrupt pending) in DCSR as volatile. We should have a fix out later today.

eroom1966, 2023-03-22:
@silabs-hfegran 
I was about to commit my modifications but run ci_check first
I am now getting a lockup in ci_check when running the clic_test clic.elf

I am not seeing mismatches, I am seeing the RTL & REF agree, but making no progress, it seems locked up here

```
Warning (RISCV_CSR_UNIMP) CPU 'refRoot/cpu' 0x000024ce 30701773 csrrw   x14,mtvt,x0: Unimplemented CSR
Test: "w_mie_notrap_r_zero" FAIL!unInvalid instruction fault at 0x000024ce
134666.000 ns: Illegal instruction (core 0) at PC 0x000024ce:
Warning (RISCV_CSR_UNIMP) CPU 'refRoot/cpu' 0x000024ce 30701773 csrrw   x14,mtvt,x0: Unimplemented CSR
Invalid instruction fault at 0x000024ce
149270.000 ns: Illegal instruction (core 0) at PC 0x000024ce:
Warning (RISCV_CSR_UNIMP) CPU 'refRoot/cpu' 0x000024ce 30701773 csrrw   x14,mtvt,x0: Unimplemented CSR
Invalid instruction fault at 0x000024ce
164012.000 ns: Illegal instruction (core 0) at PC 0x000024ce:
Warning (RISCV_CSR_UNIMP) CPU 'refRoot/cpu' 0x000024ce 30701773 csrrw   x14,mtvt,x0: Unimplemented CSR
Invalid instruction fault at 0x000024ce
```

eroom1966, 2023-03-22:
Hi @silabs-hfegran 
That trace was an output from running ci_check
Were you running ci_check ?
I have no changes in my sandbox

Thx
Lee

silabs-hfegran, 2023-03-22:
We are generally not using the ci_check command line tool, but rather the generated vsif from the cv_regress script in /bin. I had a look at the ci_check script and it does not currently appear to take other configurations into account, which explains why your ci_check is failing with this test (it cannot be run with the default configuration, as that is lacking clic-support). I will see if I can get this fixed asap.

silabs-hfegran, 2023-03-23:
@eroom1966 You should be able to run ci_check with the correct configs with this PR: https://github.com/openhwgroup/core-v-verif/pull/1736/files

eroom1966, 2023-03-23:
@silabs-hfegran I have pushed the fix can you try it ?

silabs-hfegran, 2023-03-23:
@eroom1966 I am still seeing dcsr-related mismatches in random tests - I will debug a bit and update here when I have something tangible to report

eroom1966, 2023-03-29:
@silabs-hfegran 
I am getting a linker error
```
/home/moore/riscv/riscv/bin/riscv32-corev-elf-gcc \
	 \
	-Os -g -static -mabi=ilp32 -march=rv32im_zba1p00_zbb1p00_zbc1p00_zbs1p00_zicsr_zca_zcb_zcmp_zcmt_zifencei -Wall -pedantic  \
	 \
	-I ../../tests/asm \
	-I /home/moore/tmp/issue-1730/core-v-verif/cv32e40s/bsp \
	-o /home/moore/tmp/issue-1730/core-v-verif/cv32e40s/sim/uvmt/xrun_results/pma_test_cfg_1/corev_rand_debug/66/test_program/corev_rand_debug_66.elf \
	-nostartfiles \
	 \
	-T /home/moore/tmp/issue-1730/core-v-verif/cv32e40s/sim/uvmt/xrun_results/pma_test_cfg_1/corev_rand_debug/66/test_program/link.ld \
	-L /home/moore/tmp/issue-1730/core-v-verif/cv32e40s/sim/uvmt/xrun_results/pma_test_cfg_1/corev_rand_debug/66/test_program -L /home/moore/tmp/issue-1730/core-v-verif/cv32e40s/sim/uvmt/xrun_results/pma_test_cfg_1/corev_rand_debug/66/test_program/bsp \
	-lcv-verif
/home/moore/riscv/corev-openhw-gcc-ubuntu1804-20230310/bin/../lib/gcc/riscv32-corev-elf/12.0.1/../../../../riscv32-corev-elf/bin/ld: cannot open linker script file linkcmds.memory: No such file or directory
collect2: error: ld returned 1 exit status
```

The version of compiler I have is
```
/home/moore/riscv/riscv/bin/riscv32-corev-elf-gcc --version
riscv32-corev-elf-gcc ('corev-openhw-gcc-ubuntu1804-20230310') 12.0.1 20220422 (experimental)
```

You pointed me at the ubuntu 2204 version, but I needed 1804, I am pretty sure this is correct ?

eroom1966, 2023-03-29:
Ignore this - I think I have a setup error ....

eroom1966, 2023-03-29:
There is a hangover from the changes to RTL in the DV flow

For a while the RTL was producing the wrong value in DCSR (with respect to the specification), the DCSR was set to ignore comparisons in this file/line
uvmt_cv32e40s_imperas_dv_wrap.sv:941
     rvviRefCsrCompareEnable(hart_id, `CSR_DCSR_ADDR, RVVI_FALSE);

As the RTL now seems to be producing the correct values in DCSR, this line should now be removed, I removed this and all tests ran to completion without mismatch.

The reason why the failure occurred is that the REF model is trying to evaluate which of the legal possibilities have been produced by the RTL, but this register was masked from comparison, hence this could not be evaluated.

silabs-hfegran, 2023-03-29:
Thanks for investigating this @eroom1966, it is interesting, but raises some questions;

I would expect the ISS to maintain a legal set of next states based on input signals (async) and instructions, not based on whether or not a CSR is being compared. How will this behavior translate to the situation with the special CLIC-registers, which in fact do not behave as actual registers, but rather reflect the state of the system at the point in time which they are read? (mnxti, mscratchcsw, mscratchcswl). Will this also then lead to situations where the core filters out a set of legal states, as comparison of these CSRs are disabled?

closed 2023-04-18 by silabs-hfegran