openhwfoundation/core-v-verif #1731: Core-v-verif installation kills terminal

JeanRochCoulon, 2023-03-20:
To install the cva6 env, we have to:
- clone core-v-verif
- setup variables: RISCV, VERILATOR,...
- run cva6/regress/smoke_tests.sh

Up to ce33f87beebab878eade4c31788344c7f1a4890e, installation works fine
For following hashes, the execution of cva6/regress/smoke_test.sh kills the terminal

MikeOpenHWGroup, 2023-03-20:
Is this an interactive terminal?  CORE-V-VERIF is not intended to work interactively and therefore should not use an interactive terminal.

JeanRochCoulon, 2023-03-20:
To debug CI or to develop, we need to run tests in terminals ?!?   I maybe misunderstand something...

MikeOpenHWGroup, 2023-03-20:
No, I am saying we should _not_ have terminals.  Sorry I was not clear.

zchamski, 2023-03-21:
CVA6 tests are run as shell scripts `source`-ed within the current shell, without requiring the presence of an actual terminal for interactive I/O etc..   It seems that in this specific issue, the "killing of the terminal" is in fact an unexpected exit from the enclosing shell via `exit`, `return` or `Ctrl-D` (EOF) on an input line.

This seems confirmed by the current workaround that consists in wrapping the commands in a subshell: instead of
```
. cva6/regress/smoke-tests.sh
```
one should use
```
sh cva6/regress/smoke-tests.sh
```

The root cause needs to be investigated and may be linked to changes in PR #1720.  These changes make sure that the exit code of the test is duly written into variable `tohost` at the end of test execution and could trigger an error condition in `cva6/sim/cva6.py` if a test returns a non-zero exit code.

zchamski, 2023-03-23:
This failure is not reproducible on cleanroom checkouts of `core-v-verif` from PR #1720 onwards if combined with CVA6 commits used in CI since PR #1720.

If anyone encounters this behavior again, please attach here a log of your `smoke-tests` session.  To capture it, you can use the following shell syntax to run the tests from your `core-v-verif` directory:
```
{ . cva6/regress/smoke-tests.sh ; } |& tee build.log
```
Even if your current shell session gets killed, the output of the regression test will be captured in file `build.log` inside the `core-v-verif` directory.

MikeOpenHWGroup, 2023-09-14:
Closing as @zchamski reports that the issue is not reproducible and there has been no comments for more than five months.

closed 2023-09-14 by MikeOpenHWGroup