openhwfoundation/core-v-verif #1767: ISS: dcsr reset value incorrect

silabs-hfegran, 2023-03-30:
## ISS DCSR reset value incorrect
ISS implementation of dcsr does not have the correct reset value for all configurations

### Type
* Functionally incorrect behavior

### Steps to Reproduce
Please provide:
1. core-v-verif, cv32e40s/dev : e9b88e72cbbe348dc1ce913ffed48eedf04291b7
2. `make test TEST=debug_test_0_triggers CFG=debug_trigger_cfg0 USE_ISS=YES COV=YES RUN_INDEX=17 GEN_START_INDEX=17 RNDSEED=-1485419968`

[image]

```
UVM_INFO @ 16.800 ns : uvmt_cv32e40s_base_test.sv(294) uvm_test_top [BASE TEST] set load_instr_mem
Info (IDV) Instruction executed prior to mismatch '0x80(_start+0): 00004197 auipc   x3,0x4'
Error (IDV) CSR register value mismatch (HartId:0, PC:0x00000080 _start+0):
Info (IDV)  0> CSR 7b0 (dcsr)
Info (IDV)   . dut:0x40000413 debugver:4 ebreakm:0 ebreaku:0 stepie:0 stopcount:1 stoptime:0 cause:0 mprven:1 nmip:0 prv:3(M-mode)
Info (IDV)   . ref:0x40000003 debugver:4 ebreakm:0 ebreaku:0 stepie:0 stopcount:0 stoptime:0 cause:0 mprven:0 nmip:0 prv:3(M-mode)
UVM_ERROR @ 39.300 ns : rvvi-pkg.sv(94) reporter [] uvmt_cv32e40s_tb.imperas_dv.idv_trace2api.state_compare @ 39.000 ns: MISMATCH

```

eroom1966, 2023-03-30:
@silabs-hfegran 
Apologies that was a very silly mistake for the scenario of num_trigger=0
I think this was an overhang of another issue found by halfden for the case of num_trigger=0
Fix on way

eroom1966, 2023-03-30:
@silabs-hfegran 
I have pushed a change to ImperasDV in order to fix the initial assignment of a dcsr reset when no triggers are present
commit 3a8b506da71e119b34bfc94e3f76158d0d67a6a7 (HEAD -> main, origin/main, origin/HEAD)
Author: eroom1966 <moore@imperas.com>
Date:   Thu Mar 30 10:36:12 2023 +0100

    Fix issue with reset condition of dcsr when no triggers are present

silabs-hfegran, 2023-03-30:
Thank you @eroom1966 I will test out your changes

silabs-hfegran, 2023-04-18:
Resolved, closing

closed 2023-04-18 by silabs-hfegran