openhwfoundation/core-v-verif #1772: ISS: clic, nmi handler address incorrect

silabs-hfegran, 2023-03-30:
## ISS: clic, nmi handler address incorrect

The ISS appears to attempt to jump to address 0x3c in clic mode when taking an nmi, 

User manual 0.8.0:
[image]

### Type
* Functionally incorrect behavior

### Steps to Reproduce
Please provide:
1. core-v-verif, cv32e40s/dev, 5d60f103406778ae2aaaccc3add20cf36ff30262
2. reproduced by the following tests:
``` 
make comp_corev-dv gen_corev-dv test TEST=corev_rand_data_obi_err_debug CFG=clic_default USE_ISS=YES COV=YES RUN_INDEX=464 GEN_START_INDEX=464 RNDSEED=-822216260
make comp_corev-dv gen_corev-dv test TEST=corev_rand_data_obi_err_debug CFG=clic_default USE_ISS=YES COV=YES RUN_INDEX=124 GEN_START_INDEX=124 RNDSEED=10307558 
make comp_corev-dv gen_corev-dv test TEST=corev_rand_data_obi_err_debug CFG=clic_default USE_ISS=YES COV=YES RUN_INDEX=602 GEN_START_INDEX=602 RNDSEED=1920325064 
make comp_corev-dv gen_corev-dv test TEST=corev_rand_data_obi_err_debug CFG=clic_default USE_ISS=YES COV=YES RUN_INDEX=322 GEN_START_INDEX=322 RNDSEED=625585626   
make comp_corev-dv gen_corev-dv test TEST=corev_rand_data_obi_err CFG=clic_default USE_ISS=YES COV=YES RUN_INDEX=316 GEN_START_INDEX=316 RNDSEED=832925814 
make comp_corev-dv gen_corev-dv test TEST=corev_rand_data_obi_err CFG=clic_default  USE_ISS=YES COV=YES RUN_INDEX=596 GEN_START_INDEX=596 RNDSEED=2082100806
```
3. Log (first command above):
```
Error (IDV) The reference model is unable to find a legal set of actions to converge to the provided DUT state.
Info (IDV) Reference model mode (HartId: 0):
Info (IDV)   Machine - Machine mode
Info (IDV) Currently applied nets:
Info (IDV)  - nmi => 1
Info (IDV)  - irq_id_i => 13
Info (IDV)  - irq_lev_i => cb
Info (IDV)  - irq_sec_i => 3
Info (IDV)  - irq_i => 1
Info (IDV) Pending net changes:
Info (IDV)  - Group 3:
Info (IDV)  -          nmi_cause => 1024,  age = 0 cycles, 0 retires
Info (IDV) DUT state change (hartId: 0):
Info (IDV)  - dut.pc               - 0x00000000 __jvt_base$+0
Info (IDV)  - dut.x0               - 0x00000000
Info (IDV)  - dut.mepc             - 0x00000080
Info (IDV)  - dut.mcause           - 0xb0000400 Interrupt:1 minhv:0 mpp:3 mpie:0 mpil:0 Code:1024
Info (IDV)  - dut.dcsr             - 0x400004d3 debugver:4 ebreakm:0 ebreaku:0 stepie:0 stopcount:1 stoptime:0 cause:3(haltreq) mprven:1 nmip:0 prv:3(M-mode)
Info (IDV) Reference state change evaluation tree (HartId: 0):
Info (IDV)  +  Net nmi_cause => 0x400 { when:960 }
Info (IDV)     +  Exception { pc:0x80(_start+0), mcause:0xb0000400, mstatus:0x1800 }
Info (IDV)        +  Exception { pc:0x3c((null)), mcause:0x30000002, mstatus:0x1800 }
Info (IDV)           +  Retire { pc:0x0(__jvt_base$+0) }
Info (IDV)              > dut.mepc         - 0x00000080
Info (IDV)                ref.mepc         - 0x0000003c
Info (IDV)              > dut.mcause       - 0xb0000400 Interrupt:1 minhv:0 mpp:3 mpie:0 mpil:0 Code:1024
Info (IDV)                ref.mcause       - 0x30000002 Interrupt:0 minhv:0 mpp:3 mpie:0 mpil:0 Code:2(Virtual supervisor software interrupt)
Info (IDV)             
Info (IDV)  +  Exception { pc:0x80(_start+0), mcause:0xb0000fff, mstatus:0x1800 }
Info (IDV)     +  Net nmi_cause => 0x400 { when:960 }
Info (IDV)        +  Exception { pc:0x3c((null)), mcause:0x30000002, mstatus:0x1800 }
Info (IDV)           +  Retire { pc:0x0(__jvt_base$+0) }
Info (IDV)              > dut.mepc         - 0x00000080
Info (IDV)                ref.mepc         - 0x0000003c
Info (IDV)              > dut.mcause       - 0xb0000400 Interrupt:1 minhv:0 mpp:3 mpie:0 mpil:0 Code:1024
Info (IDV)                ref.mcause       - 0x30000002 Interrupt:0 minhv:0 mpp:3 mpie:0 mpil:0 Code:2(Virtual supervisor software interrupt)
Info (IDV)             
Info (IDV)  +  Exception { pc:0x80(_start+0), mcause:0xb0000fff, mstatus:0x1800 }
Info (IDV)     +  Exception { pc:0x3c((null)), mcause:0x30000002, mstatus:0x1800 }
Info (IDV)        +  Net nmi_cause => 0x400 { when:960 }
Info (IDV)           +  Retire { pc:0x0(__jvt_base$+0) }
Info (IDV)              > dut.mepc         - 0x00000080
Info (IDV)                ref.mepc         - 0x0000003c
Info (IDV)              > dut.mcause       - 0xb0000400 Interrupt:1 minhv:0 mpp:3 mpie:0 mpil:0 Code:1024
Info (IDV)                ref.mcause       - 0x30000002 Interrupt:0 minhv:0 mpp:3 mpie:0 mpil:0 Code:2(Virtual supervisor software interrupt)
Info (IDV)             
Info (IDV)  +  Exception { pc:0x80(_start+0), mcause:0xb0000fff, mstatus:0x1800 }
Info (IDV)     +  Exception { pc:0x3c((null)), mcause:0x30000002, mstatus:0x1800 }
Info (IDV)        +  Retire { pc:0x0(__jvt_base$+0) }
Info (IDV)           > dut.mepc         - 0x00000080
Info (IDV)             ref.mepc         - 0x0000003c
Info (IDV)           > dut.mcause       - 0xb0000400 Interrupt:1 minhv:0 mpp:3 mpie:0 mpil:0 Code:1024
Info (IDV)             ref.mcause       - 0x30000002 Interrupt:0 minhv:0 mpp:3 mpie:0 mpil:0 Code:2(Virtual supervisor software interrupt)
Info (IDV)          
Info (IDV) No state change found for reference to converge to DUT state.

```

eroom1966, 2023-04-06:
@silabs-hfegran 
Did I forget to communicate this was fixed - apologies

commit 957fedd1c8fe1ba6fec7b995346e74130a5de0e8 (HEAD -> main, origin/main, origin/HEAD)
Author: eroom1966 <moore@imperas.com>
Date:   Thu Mar 30 16:45:44 2023 +0100
    fix issue for CLIC mode NMI address setting

silabs-hfegran, 2023-05-12:
Thank you for the fix, closing