openhwfoundation/core-v-verif #1777: make fails on CV32E40X

aanujdu, 2023-03-30:
I am trying to run make for the CV32E40X core, however, the command fails with the following error.

```rm Vtb_top_verilator__ALL.verilator_deplist.tmp
make[2]: Leaving directory '/home/aanujdu/core_v_verif/cv32e40x/sim/core/cobj_dir'
cp cobj_dir/Vtb_top_verilator testbench_verilator
mkdir -p
mkdir: missing operand
Try 'mkdir --help' for more information.
/home/aanujdu/core_v_verif/mk/Common.mk:485: recipe for target '/hello-world/hello-world.elf' failed
make[1]: *** [/hello-world/hello-world.elf] Error 1
make[1]: Leaving directory '/home/aanujdu/core_v_verif/cv32e40x/sim/core'
Makefile:454: recipe for target 'sanity-veri-run' failed
make: *** [sanity-veri-run] Error 2
```

### Steps to Reproduce
```
git clone https://github.com/openhwgroup/core-v-verif.git 
cd cv32e40x/sim/core
make
```
The commit hash is 760ecd3057d0de3998407ebd1e9cced9b3e7b074
The codebase should be quite similar (or even the same) to the very recent master branch because I cloned it yesterday.

### Additional context
I was able to get past the issue by adopting the simulation directory structure of the Makefile used for CV32E40P, but I am not sure why the originally written Makefile for the X core fails at the mkdir command. The variable VERI_LOG_DIR stays unresolved during Makefile execution which leads to this issue.

However, even after getting past this issue, the simulation hangs at the following message.
```
[tb_top_verilator] finished dumping memory
```
I can get past that too by specifying the maxcycles to simulate for in the testbench but I would like to know what is the right approach here.

I do not suspect any issues with my environment (packages, tools, etc.) because the simulation succeeds on the CV32E40P core.

MikeOpenHWGroup, 2023-03-31:
Hi @aanujdu, thanks for your interest in CORE-V-VERIF.  As explained [here](https://docs.openhwgroup.org/projects/core-v-verif/en/latest/quick_start.html#uvm), the "core" testbench is for demonstration purposes only.  Sadly, we have never been successful getting Verilator to properly compile and simulate the CV32E40X RTL.

AFAIK, noone has check to see if Verilator v5 is now sufficiently capable to handle the E40X.  If would be a nice contribution if you would be willing to investigate this.

aanujdu, 2023-03-31:
Thank you for the response, Mike. In that case, what do you think is the best way to test the X core which is tested and known to work right now? I need a working environment for now. 

Of course, I'd be happy to help. 
As I mentioned, I can simulate the X core using Verilator (Verilator 5.008 2023-03-04 rev v5.008-32-gaaaf8e75a, specifically). Because I am already using v5, the simulation probably proceeds without any compilation issues, but I don't know why it hangs--that would require looking at the RTL signals and finding the point of failure. I will update this thread on the debug soon.

MikeOpenHWGroup, 2023-04-03:
The CV32E40X is currently in active development and the verification is on-going.  The verification environment is written in SystemVerilog using the UVM so you will need a commerical SV simulator such as Cadence Xcelium, Siemens Questasim or Synopsys VCS.  There are at least two other commerical SV simulators (Aldec Riviera Pro and Metrics DSim), but to the best of my knowledge these have not been used in some time so the Makefiles will be suspect.  You will also need a license for ImperasDV, which is the Reference Model used for most of the E40 cores.

leemango1998, 2023-04-13:
Hi @aanujdu，PR #1787 has restored the core testbench of cv32e40x using Verilator or Vsim. You could have a try to see whether it works for you.

aanujdu, 2023-04-14:
Thanks @leemango1998 , I have tested it and it works! Thanks for the fix.