openhwfoundation/core-v-verif #1780: core-only cva6 simulator with verilator

adwait, 2023-03-31:
## Verilator target for cva6 tb is missing

### Reproduce/reference
Makefile in `cva6/tb/core` does not have a verilator target, however the README mentions that verilator based simulation is possible.

### Question?
Is it possible to run a core-only simulation with verilator for cva6?

MikeOpenHWGroup, 2023-04-03:
Hi @adwait, thanks for your question.  The short answer to your question is that there is a simple testbench in the cva6 repo that does run under Verilator.  The verification environments in this repo use SystemVerilog and the UVM and they are unlikely to run under Verilator.

Having said that, it may still be possible to run the CVA6 environment here with Verilator.  @ASintzoff, can you comment?

adwait, 2023-04-03:
Hi @MikeOpenHWGroup, thanks for your response. 

I tried to understand the files, in particular, the testbench `cva6_tb_verilator.cpp`, and the `bootrom` subdir. However, I couldn't put it together into a working core-only simulator. I was wondering whether there is an easy Makefile-based setup. Eventually, I hope to have a simulator where I can hook up a simple scratchpad (SystemVerilog) memory to the core. 

I am unfamiliar with how manifests work, so maybe that is the issue. Any help would be much appreciated.

MikeOpenHWGroup, 2023-04-03:
Hi again @adwait.  We are not in a situation where we can offer tutorials on how to build new testbenches for our cores.  If you need or want to do that, you'll need to figure it out on your own.  We are happy to help you get the testbenches we have provided up and running.

I think we can close this issue now.  If you have any issues with cva6 and/or core-v-verif, feel free to open another one.

closed 2023-04-03 by MikeOpenHWGroup

adwait, 2023-04-03:
Hi @MikeOpenHWGroup , I understand. In fact, I want to just run the provided verilator based testbench. But, I am not able to do that.

MikeOpenHWGroup, 2023-04-03:
What and where is `the provided verilator based testbench`?  Do you mean in this repo?  That is not supported.   The testbench in the CVA6 repo _can_ be run with Verilator, but it is not a core-only testbench.