openhwfoundation/core-v-verif #1802: CV32E40S does not immediately take pending interrupt in CLIC mode

eroom1966, 2023-04-13:
@Silabs-ArjanB @silabs-hfegran 

Whilst adding in the pipeline synchronization for E40S in CLIC mode, we discovered that when an mret is executed, and there is a pending interrupt, the interrupt is not immediately taken, instead the first instruction pointed to by the mepc is taken

In this particular case the instruction at mepc is an illegal instruction, and hence the RTL takes an illegal instruction trap, rather than taking the pending interrupt

Steps to reproduce
make comp_corev-dv gen_corev-dv test TEST=corev_rand_data_obi_err CV_CORE=cv32e40s CFG=clic_default COV=YES RUN_INDEX=458 GEN_START_INDEX=458 RNDSEED=-828202144

core-v-verif : 
commit 7525bd5d8052949d1b1e2716f3f32fc25ae8acd1 (HEAD -> cv32e40s/dev, origin/cv32e40s/dev)
Merge: e89b8f0e ecf2c2f2
Author: Henrik Fegran <henrik.fegran@silabs.com>
Date:   Thu Apr 13 10:57:05 2023 +0200

ImperasDV-OpenHW:
commit 8350e029623ca3644df3d3b8ff019af3ab2cc243 (HEAD -> main, origin/main, origin/HEAD)
Author: eroom1966 <moore@imperas.com>
Date:   Thu Apr 13 13:07:52 2023 +0100

silabs-hfegran, 2023-04-14:
Thanks for reporting this @eroom1966  - I have been investigating this today, and the sequence in question is quite convoluted. 

The sequence of events that I observe should be something along these lines:
```
non-shv irq 11 -> mepc 0x15c
mret                
(the above should not be relevant to the issues we are seeing)
---
non-shv irq 14 -> mepc 0x60c
nmi -> mepc -> 0x1b126 (inside handler)
mret -> mpp = 0, u-mode
pmp-trap due to access fault in u-mode interrupted by non-shv irq 5 -> m-mode, mepc -> 0x1b126, mpp = u-mode
mret -> u-mode, pc = 0x1b126
shv-irq id 18 -> mepc -> 0x1b126
mret -> u-mode, pmp-error (no pmp configured for U-mode) -> mismatch (no pending and enabled irq, so to me it looks like the trap is correct)
```
[image]

eroom1966, 2023-04-14:
@silabs-hfegran 
Thanks for the clarification.
I had a minor error in the fan-in code for the clic interrupt not updating state when negated
I am running tests and will commit this shortly if all pass OK

eroom1966, 2023-04-14:
Fix pushed 
commit 807ad380ced688ac82bf9183c1e4394f7035f54c (HEAD -> main, origin/main, origin/HEAD)
Author: eroom1966 <moore@imperas.com>
Date:   Fri Apr 14 17:56:09 2023 +0100

silabs-robin, 2023-12-11:
> Fix pushed
> commit 807ad380ced688ac82bf9183c1e4394f7035f54c (HEAD -> main, origin/main, origin/HEAD)

Hi @eroom1966 , does this mean I can safely close this ticket?

closed 2023-12-11 by eroom1966