openhwfoundation/core-v-verif #1807: Jump to no aligned address

AyoubJalali, 2023-04-14:
Actually this not a bug, but it's a question about what should the CVA6 core do in this situation : 

- If the core does not support the C extension, and we try to jump to an address not aligned, because we can't provide a test case when the C extension is enabled, because of the definition of the jump instruction JALR that clean the LSB of the address target, so how should the core react  in this situation.

MikeOpenHWGroup, 2023-04-16:
I could be wrong, but I do not see support of for C ISA and support for non-aligned addressing as necessarily being directly related.

For loads and stores, the alignment should have nothing to do with C ISA.

For instruction fetches, if _should_ be possible to start fetching from a non-aligned address.

AyoubJalali, 2023-04-17:
hello @MikeOpenHWGroup, I'm talking here about fetching from  a non-aligned address, so if the C extension is enabled the address should be aligned 2-byte boundary, and that's is always true because the jump instruction are  define to jump to an aligned address (2 byte boundary), like c.jr rs1 (clear the LSB for example), that's why I'm talking about disabling the C extension, in that case the address sould be 4-byte aligned boundary, and this a case to verify or at least know what should do.

MikeOpenHWGroup, 2023-04-18:
If I understand what you are saying then there are two situations:

- **C extension is not enabled**, therefore addresses must be aligned to 4-byte boundaries:
   - jumps to 2-byte aligned addresses are exceptions. 
- **C extension is enabled**, therefore addresses must be aligned to 2-byte boundaries.
   - jumps to 2-byte aligned address are legal.

We need to test all of these conditions, so we will need to be able to generate jumps to 2-byte aligned  addresses even when the C extension is not enabled.

AyoubJalali, 2023-04-18:
> **C extension is not enabled**, therefore addresses must be aligned to 4-byte boundaries:
> * jumps to 2-byte aligned addresses are exceptions.

we're not sure if the core is going to generate an exception in this situation, that's what this issue for, just to revel this case in the repository of github

MikeOpenHWGroup, 2023-09-14:
@AyoubJalali, we have allowed this issue to become stale, but it is still important, so let's revive it.

@zarubaf and/or @davideschiavone, do you have an opinion about this ISA question?