openhwfoundation/core-v-verif #1864: Post Synthesis Simulation

alpkoyun, 2023-05-04:
Hi,

This is a question not an issue.

Does the verification strategy and verification plans of openhwgroup include post synthesis simulation? 

I noticed that when gathering data for RVFI interface, the data is directly fetched from the registers like below. Please correct me if I am wrong but this type of statement can not be used in a post synthesis simulation right?
[image]

pascalgouedo, 2023-05-04:
Hi @alpkoyun,

To my knowledge, there is nothing to do post-synthesis simulation in verification environment.
The UVM test-bench couldn't be used until large updates would be made to disable those deep design binding.
Not sure we are going to do that for CV32E40Pv2 verification.

Maybe the much more simple non-UVM core test-bench could be tailored (together with its specific makefile) a bit to allow post-synthesis simulation (even timing back-annotated ones).
Still not decided up to now how we will manage those back-annotated post-synthesis simulations for our power consumption evaluations.

Cheers,
Pascal.

alpkoyun, 2023-05-04:
Hi @pascalgouedo 

Thank you for the response. I thought maybe the register data could be fetched using the CSR instructions to handle the problem I pointed out above, but now I realize there are probably more bindings than just these registers.

MikeOpenHWGroup, 2023-05-04:
Thanks for the question @alpkoyun.   As @pascalgouedo stated, the CORE-V-VERIF UVM environment was not designed to support gate-level simulations.  As you noticed, the RVFI is used to extract the core's archiitectual state (PC, CSRs and GPRs) each time an instruction is retired.  This is done by binding in cross-module-references into the RTL which do not work with a gate netlist.

We have done gate-level simulations with the 1.0.0 release of CV32E40P in the context of the [CORE-V-MCU](https://github.com/openhwgroup/core-v-mcu) project.  We cannot publish the gate-level netlist as it requires access to a proprietary PDK.

closed 2023-05-04 by MikeOpenHWGroup

alpkoyun, 2023-05-11:
Hi @MikeOpenHWGroup,

I have been researching the verification methodology of the riscv community. I have looked at many papers and projects, most point to the openhwgroup project and step and compare verification methodology. Even commercial firms such as SiFive and Andres seem to use this methodology for RTL verification but I could not find much information related to post-synthesis verification. Thus, I wanted to hear your opinion as I think this project greatly affects the direction of the risc-v community. Where do you think the community will head regarding post-synthesis verification?  Do you think some altered form of step and compare methodology will be used in post-synthesis simulation or will Logical Equivalence checkers handle it? I saw some open-source logical equivalence checkers online but it seems that they still have some flaws.

pascalgouedo, 2023-05-11:
Hi @alpkoyun 

Frankly speaking I haven't done "post-synthesis **verification**" for decades since commercial LEC tools became standard method to check netlist versus RTL.
I am only using "post-synthesis simulation" to make power consumption measurement of different kind of applications.

Can't talk about using open-source RTL2GDSII EDA tools because personally I think that when you want to do real chips (industrial ones to sell), you need commercial tools for simulation and implementation.

alpkoyun, 2023-05-15:
Hi @pascalgouedo , @MikeOpenHWGroup 

Thank you for the response. I guess I had some misconceptions. It makes sense to go with commercial tools. Thank you for your time.

Best regards,