openhwfoundation/core-v-verif #1872: ISACOV : No rs1' in cg_cb_andi and cg_cb_shift

AyoubJalali, 2023-05-05:
So I noticed that in the cover group [cg_cb_andi](https://github.com/openhwgroup/core-v-verif/blob/8f2df5e04448900baa2c811b2218085fdd45dc71/lib/uvm_agents/uvma_isacov/cov/uvma_isacov_cov_model.sv#L1301), and [cg_cb_shift](https://github.com/openhwgroup/core-v-verif/blob/8f2df5e04448900baa2c811b2218085fdd45dc71/lib/uvm_agents/uvma_isacov/cov/uvma_isacov_cov_model.sv#L1330), trying to toggle and cover all the values of the rs1', and not rd', I know the in the spec we have (rs1'/rd') but i don't think we should try with rs1, rd for me the the best to aim for.

silabs-robin, 2023-05-09:
> in the cover group [cg_cb_andi](https://github.com/openhwgroup/core-v-verif/blob/8f2df5e04448900baa2c811b2218085fdd45dc71/lib/uvm_agents/uvma_isacov/cov/uvma_isacov_cov_model.sv#L1301), and [...]

Here is the spec, for reference:
[image]

AyoubJalali, 2023-05-09:
> Here is the spec, for reference:
> [image]

Yes agree, but this for branching instructions not arithmetic , but what about this : 
[image]

silabs-robin, 2023-05-09:
> Yes agree, but this for branching instructions not arithmetic , but what about this : [...]

Oh, C.ANDI is said to be "CB" format, but it defines it's own new format.

_(Thanks for the screenshot of C.ANDI's format btw.)_

To me it seems arbitrary to choose either `rd'` or `rs1'` when the spec has the field `rd'/rs1'`.
So I think `rd'/rs1'` is the best choice, to just follow the spec 1-1.

--------

Another question:
Isn't it a bit weird that the covergroups for cb_andi and cb_shift cover the entire `cp_rs1: coverpoint instr.rs1`?
Aren't they by definition limited to x8-x15, and so it can never ever be fully covered?

AyoubJalali, 2023-05-09:
> To me it seems arbitrary to choose either `rd'` or `rs1'` when the spec has the field `rd'/rs1'`.

Yes it's a bit confusing !!

> Isn't it a bit weird that the covergroups for cb_andi and cb_shift cover the entire `cp_rs1: coverpoint instr.rs1`?

Yes I open an issue about it before https://github.com/openhwgroup/core-v-verif/issues/1425, and that's why we were discussing using our own decoder to fix this

closed 2024-07-10 by AyoubJalali